dmslabsbr/hoymiles #2: MQQT host does not work

K2M7L, 2021-03-21:
Hi, first of all - great thank you for releasing that add on. Would be good to check MQQT host option, even if I put my own, seems that default ones are forced.

2021-03-21 10:27:52,761 - root - INFO - Starting MQTT 192.168.50.20
2021-03-21 10:27:52,761 - root - DEBUG - mqttStart MQTT_PASSWORD: sen.... [I keep it hidden - presume that's your own password]
MQTT Start / Error! Code: timeout, Message, timed out
2021-03-21 10:28:52,822 - root - WARNING - MQTT Start / Error! Code: timeout, Message, timed out

Thanks

dmslabsbr, 2021-03-21:
Thank you. I'm going to look it.

dmslabsbr, 2021-03-21:
Hi, @K2M7L , I guess I fixed.  Could you try again?

K2M7L, 2021-03-22:
Hi @dmslabsbr, thank you, it seems that it does work :)
However, I do not see any new entity in my MQTT integration. How can I make the data visible?
Thanks

dmslabsbr, 2021-03-22:
Hi,
Do you use MQTT Mosquito server?
Do you set MQTT autodiscovery to true?
If you like, send me the add-on log. (1st ,  remove the passwords).

K2M7L, 2021-03-24:
`[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
Exporting config data
/data
PATH: 
[09:07:24] INFO: secrets.ini exists!
[09:07:24] INFO: /data/secrets.ini exists!
dmslabs - Home Assistant HoyMiles Solar Data Gateway Add-on
********** dmslabs Hoymiles Gateway v.0.09
Starting up... 2021-03-24 09:07:25     
info / Error! Code: OSError, Message, unable to determine login name
2021-03-24 09:07:25,597 - root - DEBUG - info / Error! Code: OSError, Message, unable to determine login name
IP: 172.30.33.8
Getting Config Parser.
2021-03-24 09:07:25,598 - root - DEBUG - Existe secrets.ini
Existe secrets.ini
Reading secrets.ini
2021-03-24 09:07:25,601 - root - DEBUG - Config: HOYMILES_PLANT_ID use default: 0
 IN HASS.IO 
2021-03-24 09:07:25,603 - root - DEBUG - Loading env data....
2021-03-24 09:07:25,603 - root - DEBUG - Env data loaded.
 DEVELOPERS_MODE 
HOYMILES_USER: ### 
HOYMILES_PASSWORD: ### 
HOYMILES_PLANT_ID: 30109
MQTT_HOST: 192.168.0.123
MQTT_PASSWORD: ### 
MQTT_USERNAME: ### 
INTERVALO_MQTT: 240
INTERVALO_HASS: 1200
INTERVALO_GETDATA: 480
### e10adc3949ba59abbe56e0
Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2021-03-24 09:07:25,619 - urllib3.connectionpool - DEBUG - Starting new HTTPS connection (1): global.hoymiles.com:443
2021-03-24 09:07:26,247 - urllib3.connectionpool - DEBUG - https://global.hoymiles.com:443 "POST /platform/api/gateway/iam/auth_login HTTP/1.1" 200 199
Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
{'Content-Type': 'application/json;charset=UTF-8', 'Cache-Control': 'no-cache', 'User-Agent': 'Mozilla/5.0 (Macintosh; Intel Mac OS X 11_2_3) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/89.0.4389.90 Safari/537.36', 'Host': 'global.hoymiles.com', 'Connection': 'keep-alive', 'Accept': 'application/json, text/plain, */*', 'Accept-Encoding': 'gzip, deflate, br', 'Accept-Language': 'pt-BR,pt;q=0.9,it-IT;q=0.8,it;q=0.7,es-ES;q=0.6,es;q=0.5,en-US;q=0.4,en;q=0.3', 'Cookie': "'uid=fff9c382-389f-4a47-8dc9-c5###; hm_token=eyJhbGciOiJIUzI1NiJ9.eyJYLVVJRCI6MzIyOTYsIlgtR0lEIjo0NDc4LCJleHAiOjE2MTY2MTE0MDZ9.g3h8khCKEryQGQIpU3J###; Path=/; Domain=.global.hoymiles.com; Expires=Sat, 19 Mar 2022 22:11:48 GMT;'", 'Content-Length': '68'}
2021-03-24 09:07:26,265 - urllib3.connectionpool - DEBUG - Starting new HTTPS connection (1): global.hoymiles.com:443
2021-03-24 09:07:26,838 - urllib3.connectionpool - DEBUG - https://global.hoymiles.com:443 "POST /platform/api/gateway/pvm-data/data_count_station_real_data HTTP/1.1" 200 326
content: b'{"status":"0","message":"","data":{"today_eq":"2030.0","month_eq":"383615","total_eq":"1424423","real_power":"645.9000091552734","co2_emission_reduction":"1420149.731","plant_tree":"788.973","last_data_time":"2021-03-24 10:01:18","capacitor":"6.16","is_balance":0,"is_reflux":0,"reflux_station_data":null},"systemNotice":null}'
dados_solar: {'status': '0', 'message': '', 'data': {'today_eq': '2030.0', 'month_eq': '383615', 'total_eq': '1424423', 'real_power': '645.9000091552734', 'co2_emission_reduction': '1420149.731', 'plant_tree': '788.973', 'last_data_time': '2021-03-24 10:01:18', 'capacitor': '6.16', 'is_balance': 0, 'is_reflux': 0, 'reflux_station_data': None}, 'systemNotice': None}
2021-03-24 09:07:26,848 - root - INFO - Starting MQTT 192.168.0.123
Starting MQTT 192.168.0.123
2021-03-24 09:07:26,849 - root - DEBUG - mqttStart MQTT_USERNAME: ###
2021-03-24 09:07:26,849 - root - DEBUG - mqttStart MQTT_PASSWORD: ###
MQTT connected with result code 0
2021-03-24 09:07:26,853 - root - DEBUG - MQTT connected with result code 0
Connected to 192.168.0.123
2021-03-24 09:07:27,854 - root - DEBUG - Sensor_dic: 0
2021-03-24 09:07:27,868 - root - DEBUG - Hass Sended`

user interface does not work for me on default port (I saw that you changed it) 10543, nothing discovered in MQTT, no new entity. I use MQQT Mosquito and it autodiscovers new devices

K2M7L, 2021-03-25:
FYI, the only entity which has been added is: 

> title: Insecure secrets in 44349f66_hoymiles_solardata
> friendly_name: Insecure secrets in 44349f66_hoymiles_solardata
> message: The add-on 44349f66_hoymiles_solardata uses secrets which are detected as not secure, see https://www.home-assistant.io/more-info/pwned-passwords for more information.

dmslabsbr, 2021-03-26:
Hi, where you get this message? I've never seen it.

> FYI, the only entity which has been added is:

dmslabsbr, 2021-03-26:
> `[s6-init] mak10:01:18",None} 2021-03-24 09:07:26,848 - root - INFO - Starting MQTT 192.168.0.123 Starting MQTT 192.168.0.123 2021-03-24 09:07:26,849 - root - DEBUG - mqttStart MQTT_USERNAME: ### 2021-03-24 09:07:26,849 - root -23 2021-03-24 09:07:27,854 - root - DEBUG - Sensor_dic: 0 2021-03-24 09:07:27,868 - root - DEBUG - Hass Sended`
> 
> user interface does not work for me on default port (I saw that you changed it) 10543, nothing discovered in MQTT, no new entity. I use MQQT Mosquito and it autodiscovers new devices

I found the error, "Sensor_dic: 0", it must like Sensor_dic: 1.  But now I try to solve it.

dmslabsbr, 2021-03-26:
Hi, I just upload a new version. it should work now, or at least give to us more information.

K2M7L, 2021-03-30:
it started to work as of version 0,15 for me. Thank you. The missing part for me is evolution curve instead of colourful strip ;)

dmslabsbr, 2021-03-31:
You could use a HASS chart to show this information.
[image]

closed 2021-03-31 by dmslabsbr