dmslabsbr/hoymiles #17: Problem with install add-on

Chris125913, 2021-05-04:
HI. muito obrigado @dmslabsbr 
I done all steps and install git clone https://github.com/dmslabsbr/hoymiles.git using commands
cd hoymiles
python3 -m venv ./hoymiles/
source ./bin/activate
pip3 install paho-mqtt
pip3 install requests

After that I add path do add-on repository i try to install your add-on
And nothing happens. Add-on not install and there isn't a configuration menu. What I do wrong?

dmslabsbr, 2021-05-14:
I'm trying to solve it. I guess It was either an update from HASS.io or Alpine Linux.

dmslabsbr, 2021-05-15:
@Chris125913 , I guess it now ok. Could you test it.

The Alpine Linux 3.13 appears to have a bug. That cause a ping error "clock_gettime(MONOTONIC) failed" . So, I forced the container to use the Alpine Linux 3.12 (homeassistant/amd64-base-python:3.9-alpine3.12)

Chris125913, 2021-05-16:
Hi dmslabsbr. It works perfectly. Thx for your work. Is it possible collect data from microinverters?

dmslabsbr, 2021-05-18:
> Hi dmslabsbr. It works perfectly. Thx for your work. Is it possible collect data from microinverters?

Hi, I didN´t understand. 

You would like, get data from each microinverter? Or direct from microinverter gateway without using the hoymiles site?

Chris125913, 2021-05-21:
Hi, 
I think the most convenient solution will be to read parameters directly from microinverters. But as I know they used some specific and closed protocols and there probably will be a problem connecting to these devices due to the features of that protocol. On the other hand, stats from the site are useful too, but they have been sending only 4 times per min. To summarizing, reading data from local microinverters is the best but it can be impossible. Reading data from the page is an acceptable solution,

dmslabsbr, 2021-05-21:
Hi, i guess that is possible to make a website/webservice that receive the data from DTU-W100 gateway.

closed 2021-05-21 by dmslabsbr

Chris125913, 2021-05-23:
That will be great.