dmslabsbr/hoymiles #18: Problem with install

giacomitti, 2021-07-02:
Hello, I am getting the following error when trying to installing it:

21-07-02 10:06:00 INFO (SyncWorker_3) [supervisor.docker.addon] Starting build for 44349f66/amd64-addon-hoymiles_solardata:0.19
21-07-02 10:06:03 ERROR (SyncWorker_3) [supervisor.docker.addon] Can't build 44349f66/amd64-addon-hoymiles_solardata:0.19: manifest for homeassistant/amd64-base-python:3.9-alpine3.122 not found: manifest unknown: manifest unknown
21-07-02 10:06:03 ERROR (SyncWorker_3) [supervisor.docker.addon] Build log: 
Step 1/21 : ARG BUILD_FROM=homeassistant/amd64-base-python:3.7-alpine3.12
Step 2/21 : FROM $BUILD_FROM

dmslabsbr, 2021-07-06:
Sorry, but i did everything based on the raspberry pi ARM architecture. That could be it.

I'll look later.

hstern, 2021-08-06:
I, too, have the same issue installing on hassio/linux amd64.

hstern, 2021-08-06:
I think it's just a typo. See #20

dmslabsbr, 2021-08-13:
I just update. Thank you!

closed 2021-08-13 by dmslabsbr