dmslabsbr/hoymiles #19: HoyMiles Solar Data no visible as data source for the new Energy dashboard (HA 2021.8.1)

Papi77, 2021-08-05:
Hi,
As Home Assistant 2021.8.1 just released their support for Energy management I am trying to use the Hoymiles solar data gateway as source for the "Return to Grid" part of the setup. However, there are "No matching statistics" found message.

Is it possible to use this Add-On as an Integration to allow to be used with Energy dashboard?

Regards,
Pawel

dmslabsbr, 2021-08-06:
Hi, I didn't try it yet.   I soon look it. 

Thank you.

brzozow, 2021-08-08:
Hi @dmslabsbr . Thanks for plugin. I'm using it and it would be great to add support for Energy management.

To use HoyMiles add-in in Energy management you should  add following attributes:
state_class: measurement
last_reset: (time or last reset).
It should be added to month_eq, today_eq and total_eq
It can be tricky to calculate last_reset for month_eq and today_eq (should it be reset at midnight or when values drop to zero)?

BTW: I noticed that for today_eq sensor unit_of_measurement attribute is set to kWh. In my case in today_eq I have value in Wh.
Best regards,
Michal

Papi77, 2021-08-09:
To add to above requirements: would it be possible to provide total_eq in kWh (so that we don’t have to reset it at all and yet use it foe Energy dashboard)?

Krzysztonek, 2021-09-11:
> Hi @dmslabsbr . Thanks for plugin. I'm using it and it would be great to add support for Energy management.
>
> To use HoyMiles add-in in Energy management you should add following attributes: [...]

Hi there,

To convert any measurements from one unit to another, you can do some recalculation on your own.
Please see how I did this for my own hoymiles solar farm:

[image]

Krzysztonek, 2021-09-11:
> Hi @dmslabsbr . Thanks for plugin. I'm using it and it would be great to add support for Energy management.
>
> To use HoyMiles add-in in Energy management you should add following attributes: [...]

Yes, adding full support to the HA Energy Management would be really great and very useful.

@dmslabsbr , I would also kindly suggest adding long-term statistics if possible, to integrate your solution with the Energy Management add-on.

**Anyway, you did a great job so far! Thank you so much!** 😄  👍

dmslabsbr, 2021-09-14:
Thank everybody for your comments.

I'm a bit busy in these days.  But I will put these suggestions in my add-on.

dmslabsbr, 2021-09-29:
Hi, I just upload a new add-on version. 

If anyone could try, it will be good!

I know that it need some adjustments.

maakuth, 2021-09-30:
I've tried it and it does not seem to be quite correct, even though the new values show up in the energy configuration. Now the real_power value is duplicated to three fields, some of which show up in the energy settings, but they are all getting the same value. That value was earlier in watts, which was correct - now it is in kilowatt hours. I think correct thing to do would be to add the last_reset information to total_eq and change the unit there to MWh, And then the new real_power fields could be removed and the remaining one reverted to W instead of kWh.

dmslabsbr, 2021-09-30:
@maakuth , Hi!  Thank you for your comments.

I replicate real_power to test some different configurations. I'm look closer yours suggestions.

maakuth, 2021-09-30:
Thank you for the effort! I will test it once you push something. It is possible that I can contribute some changes myself.

maakuth, 2021-11-02:
I've been running the new version for a few weeks. The solar production variables can be selected in the "Configure solar panels" dialog, but nothing appears in the statistics, unfortunately.

maakuth, 2021-11-04:
Actually it does work now! I don't quite understand why, but it's nice :+1: 

[image]

dmslabsbr, 2021-12-28:
Ok!

closed 2021-12-28 by dmslabsbr