dmslabsbr/hoymiles #22: Read data from Layout and Devices tab.

giacomitti, 2021-08-18:
Will be very nice if the data on layout (panels) and devices (microinverters) can be readed also.

dmslabsbr, 2021-09-29:
Hi, thank you for your suggestion! 

Would you explain better?

giacomitti, 2021-09-30:
In layout tab, you have the last status o panels, like this:

[image]

and the historic data also.

MrMusk, 2023-11-05:
So, I tried to decode demofile2.txt with protobuf-decoder 0.3.0, which can be found at https://pypi.org/project/protobuf-decoder/. However, I'm unable to extract any meaningful structure from the output. 
Here you can see the output that i got from my code below: [output2.txt](https://github.com/dmslabsbr/hoymiles/files/13259752/output2.txt)

# decoder.py
```from protobuf_decoder.protobuf_decoder import Parser
# Definition of a recursive function for printing the data
def format_data(parsed_result, indent=0):
    result_str = ""
    result_str += "  " * indent
    result_str += "Field {0}: {1}\n".format(parsed_result.field, parsed_result.data)

    if parsed_result.wire_type == 'length_delimited' and hasattr(parsed_result.data, 'results'):
        for sub_result in parsed_result.data.results:
            result_str += format_data(sub_result, indent + 1)

    return result_str

# Open the binary file
with open('demofile2.txt', 'rb') as file:
    binary_data = file.read()

# Convert the binary data to a hexadecimal string
hex_data = ' '.join(f'{byte:02X}' for byte in binary_data)

# Use the Parser to parse the data
parsed_data = Parser().parse(hex_data)

# Open the output file
with open('output2.txt', 'w') as output_file:
    for result in parsed_data:
        output_file.write(format_data(result))```

Cosik, 2023-11-06:
@MrMusk good job. I will need much more time to analyze it. Also I will need to find some some real data and compare it and figure out how to interpret it.

MrMusk, 2023-11-09:
How do I retrieve the protobuf data from my Hoymiles using a console or some Python code? I attempted to capture it in the network tab in Chrome, but for some reason, I'm unable to copy the protobuf message. So, I tried Firefox, where I obtained a base64 string. After decoding it, it seems to have a structure like this:
```
SID: 103281
DATE: 2022-02-13
PVID: 581752
Time ID: 4
Time: 07:00
Placehholder?: 1
All data Field 2: 10
```
(These are just guesses)

```
#Solarpanel-Data?!?
Field 1: 1
      Field 2: 
        Field 1: 
          Field 1: Fixed32Value(int:1110966272, float:46.0)
          Field 2: Fixed32Value(int:1008981770, float:0.009999999776482582)
          Field 3: Fixed32Value(int:1058642330, float:0.6000000238418579)
          Field 5: 1
```
So far it looks like that Field 1 is PV-Voltage, Field 2 Amper, Field 3 Watt, Field 5 no idea. But it would really help if I can get some protobuf data from my Solarpanels.

fcarlier, 2024-08-01:
Update 25 February 2024 : protobuf-decoder 0.4.0

New to integrate device value in HA ?