dmslabsbr/hoymiles #24: Addon restart is required after restarting Core

gralin, 2021-09-11:
If we restart Core, this addon also need to be restarted to continue working. Would be great if this addon could recover from such situation on it's own without having to be restarted.

dmslabsbr, 2021-09-14:
This is new to me!

I'm going to look it.  Thank you for your comment.

dmslabsbr, 2021-09-29:
Hi, have you tried to enable the add-on plug-in Watchdog ?

gralin, 2021-09-29:
Unfortunately I tried that and doesn't help. The add-on doesn't crash, I can still access the web ui, but the sensors become unavailable until I restart the add-on. 
[image]

kucinskimarek, 2022-03-03:
In v. 0.24.1 it keeps happening. 
After restart HA addon starts itself (log's are OK) but sensors are unavailable until restart the addon again in Supervisor.

Cosik, 2022-03-03:
Due to specific working of addon I don't know how to detect reset of HA. Also sending data in shorter period is imo out of sense because values on cloud are updated in 15 min period.

kucinskimarek, 2022-03-03:
Hmm...
Until today I was sure that restarting HA causes all addons to restart at the same time, so I thought it was a bug in the add-on

closed 2022-03-08 by Cosik