dmslabsbr/hoymiles #27: Failed to update to v. 0.20d BETA in HA

kucinskimarek, 2021-09-30:
Hi,
When I try to update add-on in HA I get error:

[image]

tonwelle, 2021-09-30:
Same here, even when reinstalling the adding from scratch. Same error message.

dmslabsbr, 2021-09-30:
I'm receiving the same message.  I'm trying to fix it.  Thank you for your comments.

dmslabsbr, 2021-10-01:
Hi, I made some adjustments and I think the bug is gone.

Krzysztonek, 2021-10-01:
Confirmed, the bug has gone. :)
I'd love also to confirm that DTU Mi model also fully works with this add-on.
[image]
.

closed 2021-10-08 by dmslabsbr