dmslabsbr/hoymiles #30: Units of measurement issue

Krzysztonek, 2021-10-06:
Hi, @dmslabsbr ,

After the last add-on update I noticed the following unit measurement discrepancies (please compare both dashboards): 
[image]
[image]
On the HA dashboard, I see 3,509.0 kWh (by default this value [_three thousand five hundred and nine_] has been wrongly multiplied by 1000. It should be 3.509 kWh) while the original hoymiles dashboard displays this value with the correct unit (3.51kWh). The same applies to the total energy: on the HA dashboard I see 8.73 **kWh**, while the original dashboard displays 8.73 **MWh**, So the value has to be multiplied by 1000, but it is not. I don't use any unit recalculations. Could you please resolve this issue?

dmslabsbr, 2021-10-07:
Goodnight! As I also have this equipment, so I am interested in correcting all the problems.
I'm already working on it.
Thank you!

dmslabsbr, 2021-10-08:
hi, I just upload version "**0.20h**".

I implementated in html code, dark mode detection and change some units.

 Home assistant statistics don't like MWh, so I used kWh for _total_eq_.

I guess that best configuration for  "_Solar energy production_"   is "_solarH_?????_today_eq_". 

Could you try it?

Krzysztonek, 2021-10-09:
Hi @gralin, 
I'd keep the units as they are since on some days in Europe it's so cloudy and you won't produce much energy on that day so I wouldn't expect that kW will be useful. To be decided by the add-on's author @dmslabsbr , who lives in a really sunny country and who probably sees clouds from time to time only...  ;)

gralin, 2021-10-09:
Hi @Krzysztonek. Sure, I get your point. From my point of view, when it comes to PV capacity, it's always mentioned in `kW` as usually 3 panels already give you 1000 W. As to current power readings, it's true that in some regions in winter we get a small fraction of the nominal capacity, so we would end up with e.g. 0.4 kW at peak. But I think monitoring in such cases is a waste of time anyway, as whole month may be worth the same as 2 days in summer. I would focus on how the readings look when we actually have production, and in such cases 3.5 kW looks better than 3500 W, as it's easier to compare to the capacity, which everyone knows in kW.

For now I have template sensors that convert the value/unit to how I think looks better, but looking forward to removing them in case the author agrees with my arguments 😄

dmslabsbr, 2021-10-10:
> Hi @gralin, I'd keep the units as they are since on some days in Europe it's so cloudy and you won't produce much energy on that day so I wouldn't expect that kW will be useful. To be decided by the [...]

Hi @Krzysztonek, Clouds, sometimes there are many. But along with heat and no rain. In a few months the humidity here is the same as in the desert. Brazil is very big and has different climates.  :-)

dmslabsbr, 2021-10-10:
@gralin and @Krzysztonek , 

I don't know if I got it right.
One argues that sensors realpower and capacitor  should be in kW instead of W and the other that W instead of kW?

I could put in the add-on configuration for the user to choose which measure he wants. What do you think?

Are these two sensors (real_power & capacitor) or some other one?

gralin, 2021-10-12:
I can see new entity `capacitor_kw` which gives me system capacity in kW. Since we go this way, could you also add `real_power_kw` giving us the current power in kW?

closed 2023-10-28 by Cosik