dmslabsbr/hoymiles #40: Continuous errors in Home Assistant Core log 'dict object' has no attribute

Papi77, 2021-11-23:
Hi,
When checking the log entries for HA (Core) - I see a lot of lines with something similar to:

2021-11-23 11:38:47 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'reset_real_power_measurement' when rendering '{{ value_json.reset_real_power_measurement }}'
2021-11-23 11:38:47 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'reset_real_power_total_increasing' when rendering '{{ value_json.reset_real_power_total_increasing }}'
2021-11-23 11:38:47 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'reset_today_eq_Wh' when rendering '{{ value_json.reset_today_eq_Wh }}'
2021-11-23 11:38:47 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'reset_total_eq' when rendering '{{ value_json.reset_total_eq }}'
2021-11-23 11:38:48 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'reset_real_power_measurement' when rendering '{{ value_json.reset_real_power_measurement }}'
2021-11-23 11:38:48 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'reset_real_power_total_increasing' when rendering '{{ value_json.reset_real_power_total_increasing }}'
2021-11-23 11:38:48 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'reset_today_eq_Wh' when rendering '{{ value_json.reset_today_eq_Wh }}'
2021-11-23 11:38:48 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'reset_total_eq' when rendering '{{ value_json.reset_total_eq }}'

Could we get rid of this WARNING?

dmslabsbr, 2021-11-25:
At first this does not affect the system. But I'll check it out anyway.

Thank you for your help!

Papi77, 2021-12-03:
Any chance of getting this minor detail fixed? Looks like an easy one yet it pollutes the log files making it easy to miss other warnings :(

dmslabsbr, 2021-12-03:
Hi @Papi77 , I will do it. But I´m bit busy now.  But, I think that i could do it before 2021 end.

Thanks for your interest.

Papi77, 2022-01-09:
Hi, this looks like it isn’t fixed yet. Unfortunately I’m also getting the Errors and the application often disconnects and You can only see the last read. The only workaround is to restart the add on - then it work for a day or two and stops again :(

dmslabsbr, 2022-01-10:
I will look these bugs as soon as possible. But I very busy these days.

dmslabsbr, 2022-01-21:
I guess it now working without warning messages.  Could you check it?

Papi77, 2022-01-21:
Hi,
Unfortunately, the errors still show up: (around every 8-9 minutes):
2022-01-21 11:46:45 WARNING (MainThread) [homeassistant.helpers.template]
Template variable warning: 'dict object' has no attribute
'reset_power_ratio' when rendering '{{ value_json.reset_power_ratio }}'
2022-01-21 11:54:56 WARNING (MainThread) [homeassistant.helpers.template]
Template variable warning: 'dict object' has no attribute 'reset_capacitor'
when rendering '{{ value_json.reset_capacitor }}'
2022-01-21 11:54:56 WARNING (MainThread) [homeassistant.helpers.template]
Template variable warning: 'dict object' has no attribute
'reset_capacitor_kW' when rendering '{{ value_json.reset_capacitor_kW }}'
2022-01-21 11:54:56 WARNING (MainThread) [homeassistant.helpers.template]
Template variable warning: 'dict object' has no attribute
'reset_co2_emission_reduction' when rendering '{{
value_json.reset_co2_emission_reduction }}'
2022-01-21 11:54:56 WARNING (MainThread) [homeassistant.helpers.template]
Template variable warning: 'dict object' has no attribute
'reset_is_balance' when rendering '{{ value_json.reset_is_balance }}'
2022-01-21 11:54:56 WARNING (MainThread) [homeassistant.helpers.template]
Template variable warning: 'dict object' has no attribute 'reset_is_reflux'
when rendering '{{ value_json.reset_is_reflux }}'
2022-01-21 11:54:56 WARNING (MainThread) [homeassistant.helpers.template]
Template variable warning: 'dict object' has no attribute
'reset_last_data_time' when rendering '{{ value_json.reset_last_data_time
}}'
2022-01-21 11:54:56 WARNING (MainThread) [homeassistant.helpers.template]
Template variable warning: 'dict object' has no attribute
'reset_plant_tree' when rendering '{{ value_json.reset_plant_tree }}'
2022-01-21 11:54:56 WARNING (MainThread) [homeassistant.helpers.template]
Template variable warning: 'dict object' has no attribute
'reset_real_power' when rendering '{{ value_json.reset_real_power }}'
2022-01-21 11:54:56 WARNING (MainThread) [homeassistant.helpers.template]
Template variable warning: 'dict object' has no attribute
'reset_reflux_station_data' when rendering '{{
value_json.reset_reflux_station_data }}'
2022-01-21 11:54:56 WARNING (MainThread) [homeassistant.helpers.template]
Template variable warning: 'dict object' has no attribute
'reset_power_ratio' when rendering '{{ value_json.reset_power_ratio }}'
2022-01-21 11:54:56 WARNING (MainThread) [homeassistant.helpers.template]
Template variable warning: 'dict object' has no attribute 'reset_capacitor'
when rendering '{{ value_json.reset_capacitor }}'
2022-01-21 11:54:56 WARNING (MainThread) [homeassistant.helpers.template]
Template variable warning: 'dict object' has no attribute
'reset_capacitor_kW' when rendering '{{ value_json.reset_capacitor_kW }}'
2022-01-21 11:54:56 WARNING (MainThread) [homeassistant.helpers.template]
Template variable warning: 'dict object' has no attribute
'reset_co2_emission_reduction' when rendering '{{
value_json.reset_co2_emission_reduction }}'
2022-01-21 11:54:56 WARNING (MainThread) [homeassistant.helpers.template]
Template variable warning: 'dict object' has no attribute
'reset_is_balance' when rendering '{{ value_json.reset_is_balance }}'
2022-01-21 11:54:56 WARNING (MainThread) [homeassistant.helpers.template]
Template variable warning: 'dict object' has no attribute 'reset_is_reflux'
when rendering '{{ value_json.reset_is_reflux }}'
2022-01-21 11:54:56 WARNING (MainThread) [homeassistant.helpers.template]
Template variable warning: 'dict object' has no attribute
'reset_last_data_time' when rendering '{{ value_json.reset_last_data_time
}}'
2022-01-21 11:54:56 WARNING (MainThread) [homeassistant.helpers.template]
Template variable warning: 'dict object' has no attribute
'reset_plant_tree' when rendering '{{ value_json.reset_plant_tree }}'
2022-01-21 11:54:56 WARNING (MainThread) [homeassistant.helpers.template]
Template variable warning: 'dict object' has no attribute
'reset_real_power' when rendering '{{ value_json.reset_real_power }}'
2022-01-21 11:54:56 WARNING (MainThread) [homeassistant.helpers.template]
Template variable warning: 'dict object' has no attribute
'reset_reflux_station_data' when rendering '{{
value_json.reset_reflux_station_data }}'
2022-01-21 11:54:56 WARNING (MainThread) [homeassistant.helpers.template]
Template variable warning: 'dict object' has no attribute
'reset_power_ratio' when rendering '{{ value_json.reset_power_ratio }}'

Regards,
Pawel.

jgrobelny-dev, 2022-01-21:
I can confirm. Warning is still there in 0.23:

Logger: homeassistant.helpers.template
Source: helpers/template.py:1779 
First occurred: January 20, 2022, 21:56:54 (2258 occurrences) 
Last logged: 18:45:25

Template variable warning: 'dict object' has no attribute 'reset_last_data_time' when rendering '{{ value_json.reset_last_data_time }}'
Template variable warning: 'dict object' has no attribute 'reset_plant_tree' when rendering '{{ value_json.reset_plant_tree }}'
Template variable warning: 'dict object' has no attribute 'reset_real_power' when rendering '{{ value_json.reset_real_power }}'
Template variable warning: 'dict object' has no attribute 'reset_reflux_station_data' when rendering '{{ value_json.reset_reflux_station_data }}'
Template variable warning: 'dict object' has no attribute 'reset_power_ratio' when rendering '{{ value_json.reset_power_ratio }}'

Papi77, 2022-01-24:
Additionally, I now see other types of warnings: 
[image]

When do You think You'll have time to fix it? It's causing quite a bit of logs...

Papi77, 2022-01-31:
Since, it's taking a while to get the above issues fixed - I found a different way to get data from my Hoymiles solar system.
Using a simple nodered flow I can get the data directly from the local DTU-100 unit (without having to go to the web).
The benefits are that it's:
- it's much simpler (no need to install any add-on)
- local (you only need to query Your local DTU unit), no need for account/password/token, etc.
- based on a simple nodered flow using curl command in the background
- easy to modify which entities You want
- easy to specify how frequently you'd like to poll for data

NO WARNING MESSAGES in the Log!!!!!!!!!! :) :) :)

I have published my flow in the HA facebook pages: 
https://www.facebook.com/groups/HomeAssistant/permalink/3144220299182652/

I will check how it runs for a few days and if all is good - I'll be removing Your add-on as I don't see the benefit any longer :( sorry.

jgrobelny-dev, 2022-01-31:
As it might work with DTU-100, as far as I know, there is no solution that works with DTU-PRO. So this addon is a way to go for many users!

Papi77, 2022-01-31:
Can You access DTU-PRO in any way? I don’t have it so cannot test… but
You’re absolutely right - this add-on still offers a great value. It’s just
my needs can be accommodated in a different way…

Cosik, 2022-02-19:
@Papi77 and @jgrobelny-dev  do you still see this problem? Currently I'm working on rewriting this addon, also it is working with DTU-PRO, but I will need some tester to check it with DTU-100

dmslabsbr, 2022-02-19:
Very good @Papi77 !

But, could you explain more how I could use your code in nodered?

It will be a good option to get data without internet connection.

A screen shot will be welcome.

> Since, it's taking a while to get the above issues fixed - I found a different way to get data from my Hoymiles solar system. Using a simple nodered flow I can get the data directly from the local [...]

dmslabsbr, 2022-02-19:
Sorry @Papi77  and other users, but I still not enough time for improve add-on neither fix some bugs.

But now I going to have some help from @Cosik  .
And by the way, anyone that like home assistant and know python are invited to help too.

Cosik, 2022-02-19:
@dmslabsbr and other, I have a bad news, Hoymiles has blocked local access to at least DTU-PRO so read data locally is not possible.

Cosik, 2022-02-26:
Fix is already on master.

closed 2022-03-08 by Cosik

eduspina, 2024-11-12:
Please disregard, for some reason Read_meter_data was enabled and I don't have it, so all data was Unknown.