dmslabsbr/hoymiles #41: Adding device_class & state_class for power sensors

januszwoj, 2021-12-11:
Could you add:

       device_class: power
       state_class: measurement

for power measurement sensors?

   - sensor.solarh _ #### _ real_power
   - sensor.solarh _ #### _ real_power_measuremen
   - sensor.solarh _ #### _ real_power_total_increasing

They could be automatically registered by the "statistics" module and used in the "Statistics Graph Card".
Could you add:

       device_class: power
       state_class: measurement

for power measurement sensors?

   - sensor.solarh _ #### _ real_power
   - sensor.solarh _ #### _ real_power_measuremen
   - sensor.solarh _ #### _ real_power_total_increasing

They could be automatically registered by the "statistics" module and used in the "Statistics Graph Card".

Thank you for this great plugin :+1::+1::+1:.
Janusz

dmslabsbr, 2021-12-13:
Yes, i will try in few days.

januszwoj, 2021-12-13:
Thanks
Janusz

dmslabsbr, 2021-12-28:
Hi @januszwoj ,

I'm looking it now. I just add device_class: power and state_class: measurement to real_power template.  (v.22, not uploaded yet).

But, real_power_measuremend and real_power_total_increasing are in `Wh` that should be use with "device_class":"energy". 

Could you see https://www.home-assistant.io/integrations/sensor/#device-class. 

Any help will be great.

januszwoj, 2021-12-30:
Yes you are right, I wrote faster than I thought.
I guess it should look like this:

device_class: power
state_class: measurement

_capacitor
_real_power
_capacitor_kw

device_class: energy
state_class: measurement          `<-  I think that "state_class" for energy should be "total" [`](https://developers.home-assistant.io/docs/core/entity/sensor/#available-state-classes)

_real_power_measurement
_real_power_total_increasing
_today_eq
_today_eq_Wh
_total_eq
_month_eq

I have one more question / request. In my installation, I have a meter plugged in on the network side connected to the Hoymiles gateway.
While looking at the data that your plugin receives,  I would be interested in: grid_power. I was wondering if it would be possible to extend it to the next entities? Below I am enclosing the most recently received message.
Temporarily after connecting meter to the Hoymiles gateway, I cannot download data from it at the same time.

Received 10:32:57

    QoS: 0
    Payload:

    today_eq: '0.34'
    month_eq: '77 .07 '
    year_eq: '77073'
    total_eq: '77 .07 '
    real_power: '439'
    co2_emission_reduction: '0.08'
    plant_tree: '43'
    last_data_time: '2021-12-30 10:31:18'
    capacitor: '5070'
    is_balance: 0
    is_reflux: 1
    reflux_station_data:
      start_date: null
      end_date: null
      pv_power: '439.4'
      grid_power: '1160'                 <------------------- device_class: power  state_class: measurement
      load_power: '0'
      bms_power: null
      bms_soc: null
      meter_location: 2
      pv_to_load_eq: null
      load_from_pv_eq: null
      meter_b_in_eq: '0'
      meter_b_out_eq: '35960'
      bms_in_eq: null
      bms_out_eq: null
      self_eq: '41113'
      pv_eq_total: null
      use_eq_total: null
      flows:
        - out: 2
          in: 1
        - out: 4
          in: 1
        - out: 4
          in: 2
    real_power_measurement: '439'
    real_power_total_increasing: '439'
    power_ratio: '8.7'
    capacitor_kW: '5.07'
    today_eq_Wh: '338.0'
    reset_today_eq: '2021-12-30T00: 00: 00 + 00: 00'
    reset_month_eq: '2021-12-01T00: 00: 00 + 00: 00'

Thank you again for this great plugin 👍👍👍.
Janusz

Cosik, 2022-02-21:
@januszwoj it will be not so easy. Do you still want this feature? If yes please contact me via discord. I'm on HA channel, because this will required more details to implement.

januszwoj, 2022-02-26:
Hi.
Not particularly anymore, I have invested in two energy meters with modbus interfaces so I will have an excess of data :-). 
I just need to figure out what to do with them.
Thank you for this great plugin 👍👍👍.
Regards Janusz

Cosik, 2022-02-26:
@januszwoj ok, no problem, I was asking because there is possibility to get this value by adding custom sensor in config. Also there was big upade which could required removing mqtt device due to big changes in addon