dmslabsbr/hoymiles #42: Missing timezone information - core-2021.12.8

kom101, 2022-01-05:
2022-01-05 13:28:54 ERROR (MainThread) [homeassistant.util.logging] Exception in message_received when handling msg on 'home/solar/json': '{"today_eq": "1.36", "month_eq": "16.97", "year_eq": "16972", "total_eq": "7124.51", "real_power": "113", "co2_emission_reduction": "7.1", "plant_tree": "3946", "last_data_time": "2022-01-05 13:16:23", "capacitor": "6900", "is_balance": 0, "is_reflux": 0, "reflux_station_data": null, "real_power_measurement": "113", "real_power_total_increasing": "113", "power_ratio": "1.6", "capacitor_kW": "16.9", "today_eq_Wh": "1357.0", "reset_today_eq": "2022-01-05T00:00:00+00:00", "reset_month_eq": "2022-01-01T00:00:00+00:00"}'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 37, in wrapper
    msg_callback(msg)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 249, in message_received
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 505, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 538, in _async_write_ha_state
    state = self._stringify_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 511, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 411, in state
    raise ValueError(
ValueError: Invalid datetime: sensor.solarh_31140_last_data_time provides state '2022-01-05 13:16:23', which is missing timezone information

Papi77, 2022-01-15:
Same issue:

ValueError: Invalid datetime: sensor.solarh_22502_last_data_time provides state '2022-01-15 11:56:03', which is missing timezone information

dmslabsbr, 2022-01-19:
Hi @kom101 and @Papi77,
I think this issue has been fixed - v0.22a7
Could you test and give me feedback?

Papi77, 2022-01-19:
This update caused the plug-in to fail to start completely (I opened a new
issue for it) so I cannot confirm if this is fixed.

Regards,
Paweł

jgrobelny-dev, 2022-01-19:
After running for an hour the log is clear in terms of the error mentioned here.
(While it's still full of warnings from issue #40 )

Papi77, 2022-01-19:
Exactly same behavior here - errors are gone but Core log file is full of
warnings as in #40 <https://github.com/dmslabsbr/hoymiles/issues/40>

dmslabsbr, 2022-01-19:
@Papi77 ,

Is it working again? You said 'Exactly same behavior here - errors are gone but Core log file is full of
warnings'
Is it working again? You said 'Exactly same behavior here - errors are gone but Core log file is full of
warnings'

> This update caused the plug-in to fail to start completely (I opened a new issue for it) so I cannot confirm if this is fixed. Regards, Paweł
> […](#)
> On Wed, 19 Jan 2022 at 03:07, d

Papi77, 2022-01-20:
Yes, after adding a value to External_MQTT_TLS_PORT (w.g. 1) - the add-on
is able to start and working again.

So, the error related to ‘expected int’ is gone.
However, all the warnings as mentioned in issue #44 are still showing up in
Supervisor Core logs

kom101, 2022-01-20:
Time zone fix is ok.

dmslabsbr, 2022-01-20:
it is ok.

closed 2022-01-20 by dmslabsbr