dmslabsbr/hoymiles #48: UI problem and crash after few seconds

Arek1990, 2022-01-27:
Hello,
I installed this addon. In config I write my hoymiles account (I tried Installer account and end user account) and MQTT host, login and password. Addon is starting but when I click on UI i have error: 502: Bad Gateway and after few second addon is dead. In log I have:

on red:
MQTT Server: core-mosquitto

last entry:
2022-01-27 20:34:34,241 - urllib3.connectionpool - DEBUG - https://global.hoymiles.com:443 "POST /platform/api/gateway/pvm-data/data_count_station_real_data HTTP/1.1" 200 283

Any idea what i'm going wrong?

Cosik, 2022-02-16:
@Arek1990 do you still have problems? If yes please select debug mode and send full log, please remember to blure pass and user

dmslabsbr, 2022-02-16:
@Arek1990 , Is your mqtt mosquitto server installed in a different machine than HA?

The mqtt host, user and password is used only if you use an external mqtt server. Otherwise the add-on can get login data by itself.

Coud you send more information?

Arek1990, 2022-02-27:
Ok. I found problem. When I tried to connect to the S-Miles account on the farm that is not yet operational, the application crashes because the Data Update field is empty.

Cosik, 2022-02-27:
@Arek1990 so could we close issue?

Arek1990, 2022-02-27:
`        last_data_time = datetime.strptime(last_data_time, '%Y-%m-%d %H:%M:%S')`
line 153 in hoymilesapi will give an error and crash if the farm has not worked before. It should either be noted that the add-on works only with already operating PV farms or correct this place.

Cosik, 2022-02-27:
@Arek1990 so I don't understand you, You wart to observe PV farm which was not setup and running? last_data_time is taken from API response.

Arek1990, 2022-02-27:
Right now my farm is working so I can't log anymore. Previously, I integrated the add-on in parallel with the construction of the farm. I guess someone might try that too, and there should at least be a warning somewhere that the add-on doesn't work that way. 
If last_data_time is empty add-on will crash because  datetime.strptime cannot have an empty parameter.

Cosik, 2022-02-27:
This API is using unofficial API without documentation, so it is more harder to handle all cases. Currently I'm working for some additional features. One could be that if DTU is not connected to cloud there will be no reading from cloud.

closed 2022-02-27 by Cosik