dmslabsbr/hoymiles #51: Problem with default mqtt authorisation after update to 0.24

jgrobelny-dev, 2022-02-27:
There is some issue with mqtt deafult authorization after update from 0.23a to 0.24 (HA 2022.2.9). 
0.23a used to work fine with built in mosquito broker (log below). I suppose "nda" is not an expected user. How can I make it use internal HA addon mqtt user out of the box?

> 2022-02-27 09:52:34,359 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - last_data_time 2022-02-27T09:46:12+01:00
2022-02-27 09:52:34,360 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Starting MQTT 127.0.0.1
2022-02-27 09:52:34,360 - HoymilesAdd-on.mqttapi.Mqtt - DEBUG - SSL: OpenSSL 1.1.1l  24 Aug 2021
2022-02-27 09:52:34,360 - HoymilesAdd-on.mqttapi.Mqtt - DEBUG - mqttStart TLS: False
2022-02-27 09:52:34,360 - HoymilesAdd-on.mqttapi.Mqtt - DEBUG - mqttStart MQTT_USERNAME: nda
2022-02-27 09:52:34,361 - HoymilesAdd-on.mqttapi.Mqtt - DEBUG - mqttStart MQTT_PASSWORD: nda
2022-02-27 09:52:34,361 - HoymilesAdd-on.mqttapi.Mqtt - DEBUG - mqttStart MQTT_PORT: 1883
2022-02-27 09:52:34,363 - HoymilesAdd-on.mqttapi.Mqtt - ERROR - MQTT connected with result code 5
2022-02-27 09:52:34,364 - HoymilesAdd-on.mqttapi.Mqtt - ERROR - 5Connection refused – not authorised
2022-02-27 09:52:34,364 - HoymilesAdd-on.mqttapi.Mqtt - ERROR - APP EXIT 5

Cosik, 2022-02-27:
@jgrobelny-dev you need to configure mqtt, you didn't set username, password, and ip of mqtt server.

Cosik, 2022-02-27:
@jgrobelny-dev but you have instaled mqtt broker in HA? I think yes, so you need to add configuration of it. It was working out of the box because there was some nasty workarounds which extract some variables, and it was not safe.

jgrobelny-dev, 2022-02-27:
Issue solved! Adding mqtt login credentials to the config worked.

Cosik, 2022-02-27:
But please remember that in instruction https://github.com/dmslabsbr/hoymiles/blob/master/DOCS.md there is information that adding information about user and pass is necessary.

closed 2022-02-27 by Cosik