dmslabsbr/hoymiles #56: stop working after last update

Kwiatjk, 2022-02-28:
SSL MQTT stop working 
2022-02-28 16:21:10,937 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - last_data_time 2022-02-28T16:16:24+01:00
Traceback (most recent call last):
  File "/data/../hoymiles.py", line 249, in <module>
    sys.exit(main())  # next section explains the use of sys.exit
  File "/data/../hoymiles.py", line 209, in main
    mqtt.start()
  File "/mqttapi.py", line 57, in start
    port = self._config['MQTT_TLSPORT']
KeyError: 'MQTT_TLSPORT'
[cmd] /run.sh exited 1
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.
without also 
2022-02-28 16:18:56,937 - HoymilesAdd-on - INFO - Solar data publication...2022-02-28 16:18:56.937428
2022-02-28 16:18:56,938 - HoymilesAdd-on - INFO - Main loop start!

Cosik, 2022-02-28:
There was big update which required reconfiguration of plugin.

Kwiatjk, 2022-02-28:
what is wrong ?
Withot ssl stopo working
HOYMILES_USER: xxxx
HOYMILES_PASSWORD: xxx
HOYMILES_PLANT_ID: 'xxx'
DEVELOPERS_MODE: false
MQTT_Host: core-mosquitto
MQTT_User: xxx
MQTT_Pass: xxx
MQTT_TLS: false
MQTT_TLS_PORT: '8883'
External_MQTT_Server: true
External_MQTT_Host: core-mosquitto
External_MQTT_User: xxxx
External_MQTT_Pass: xxxx
External_MQTT_TLS: false
External_MQTT_TLS_PORT: 8883

Cosik, 2022-02-28:
Now there is no External_* and from which place you have copy those values? Now they are read from config.json or options.json depending where are executed

Kwiatjk, 2022-02-28:
Log from : 
dmslabs - Home Assistant HoyMiles Solar Data Gateway Add-on
2022-02-28 18:06:34,349 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway v.0.24
2022-02-28 18:06:34,350 - HoymilesAdd-on - INFO - Starting up... 2022-02-28 18:06:34
2022-02-28 18:06:34,351 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2022-02-28 18:06:35,221 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - I got the token!!  :-)
2022-02-28 18:06:35,222 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2022-02-28 18:06:35,604 - HoymilesAdd-on.hoymilesapi.Hoymiles - WARNING - real_power = 0
2022-02-28 18:06:35,612 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - last_data_time 2022-02-28T18:01:27+01:00
2022-02-28 18:06:35,615 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Starting MQTT core-mosquitto
2022-02-28 18:06:35,619 - HoymilesAdd-on.mqttapi.Mqtt - INFO - MQTT connected with result code 0
2022-02-28 18:06:35,624 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Connected to core-mosquitto
2022-02-28 18:06:36,631 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2022-02-28 18:06:36,999 - HoymilesAdd-on.hoymilesapi.Hoymiles - WARNING - real_power = 0
2022-02-28 18:06:36,999 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - last_data_time 2022-02-28T18:01:27+01:00
2022-02-28 18:06:37,000 - HoymilesAdd-on - INFO - Solar data publication...2022-02-28 18:06:36.999973
2022-02-28 18:06:37,014 - HoymilesAdd-on - INFO - Main loop start!

Cosik, 2022-02-28:
Regarding to log it is working.

Kwiatjk, 2022-02-28:
No 
502: Bad Gateway

Cosik, 2022-02-28:
So please add local host or your HA ip as I sugest previously in cofiguration, what is more please add full log from begining. In previous one I can see Connected to core-mosquitto

Kwiatjk, 2022-02-28:
WITH SSL enabled
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
Starting add-on container...
Mon Feb 28 19:41:52 CET 2022
Exporting config data...
Getting mqqt data...
total 16
d---------    2 root     root          4096 Feb 28 19:41 .
d---------    4 root     root          4096 Feb 28 19:41 ..
----------    1 root     root           718 Feb 28 19:41 base.html
----------    1 root     root           221 Feb 28 19:41 index.html
[19:41:55] INFO: secrets.ini exists!
[19:41:55] INFO: /data/secrets.ini exists!
[19:41:56] INFO: null
[19:41:56] INFO: {"supervisor":"2022.01.1","homeassistant":"2022.2.9","hassos":null,"docker":"20.10.3","hostname":null,"operating_system":null,"features":[],"machine":"qemux86-64","arch":"amd64","state":"running","supported_arch":["amd64","i386"],"supported":false,"channel":"stable","logging":"info","timezone":"Europe/Warsaw"}
[19:41:56] INFO: {"supervisor":"2022.01.1","homeassistant":"2022.2.9","hassos":null,"docker":"20.10.3","hostname":null,"operating_system":null,"features":[],"machine":"qemux86-64","arch":"amd64","state":"running","supported_arch":["amd64","i386"],"supported":false,"channel":"stable","logging":"info","timezone":"Europe/Warsaw"}
dmslabs - Home Assistant HoyMiles Solar Data Gateway Add-on
2022-02-28 19:41:57,106 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway v.0.24
2022-02-28 19:41:57,107 - HoymilesAdd-on - INFO - Starting up... 2022-02-28 19:41:57
2022-02-28 19:41:57,108 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2022-02-28 19:41:57,885 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - I got the token!!  :-)
2022-02-28 19:41:57,886 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2022-02-28 19:41:58,247 - HoymilesAdd-on.hoymilesapi.Hoymiles - WARNING - real_power = 0
2022-02-28 19:41:58,259 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - last_data_time 2022-02-28T18:01:27+01:00
Traceback (most recent call last):
  File "/data/../hoymiles.py", line 249, in <module>
    sys.exit(main())  # next section explains the use of sys.exit
  File "/data/../hoymiles.py", line 209, in main
    mqtt.start()
  File "/mqttapi.py", line 57, in start
    port = self._config['MQTT_TLSPORT']
KeyError: 'MQTT_TLSPORT'
[cmd] /run.sh exited 1
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.

ANd without 
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
Starting add-on container...
Mon Feb 28 19:42:35 CET 2022
Exporting config data...
Getting mqqt data...
total 16
d---------    2 root     root          4096 Feb 28 19:42 .
d---------    4 root     root          4096 Feb 28 19:42 ..
----------    1 root     root           718 Feb 28 19:42 base.html
----------    1 root     root           221 Feb 28 19:42 index.html
[19:42:38] INFO: secrets.ini exists!
[19:42:38] INFO: /data/secrets.ini exists!
[19:42:39] INFO: null
[19:42:39] INFO: {"supervisor":"2022.01.1","homeassistant":"2022.2.9","hassos":null,"docker":"20.10.3","hostname":null,"operating_system":null,"features":[],"machine":"qemux86-64","arch":"amd64","state":"running","supported_arch":["amd64","i386"],"supported":false,"channel":"stable","logging":"info","timezone":"Europe/Warsaw"}
[19:42:39] INFO: {"supervisor":"2022.01.1","homeassistant":"2022.2.9","hassos":null,"docker":"20.10.3","hostname":null,"operating_system":null,"features":[],"machine":"qemux86-64","arch":"amd64","state":"running","supported_arch":["amd64","i386"],"supported":false,"channel":"stable","logging":"info","timezone":"Europe/Warsaw"}
dmslabs - Home Assistant HoyMiles Solar Data Gateway Add-on
2022-02-28 19:42:40,610 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway v.0.24
2022-02-28 19:42:40,611 - HoymilesAdd-on - INFO - Starting up... 2022-02-28 19:42:40
2022-02-28 19:42:40,612 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2022-02-28 19:42:40,961 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - I got the token!!  :-)
2022-02-28 19:42:40,962 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2022-02-28 19:42:41,348 - HoymilesAdd-on.hoymilesapi.Hoymiles - WARNING - real_power = 0
2022-02-28 19:42:41,353 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - last_data_time 2022-02-28T18:01:27+01:00
2022-02-28 19:42:41,358 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Starting MQTT 192.168.1.90
2022-02-28 19:42:41,362 - HoymilesAdd-on.mqttapi.Mqtt - INFO - MQTT connected with result code 0
2022-02-28 19:42:41,362 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Connected to 192.168.1.90
[... cut ...] oymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2022-02-28 19:42:42,793 - HoymilesAdd-on.hoymilesapi.Hoymiles - WARNING - real_power = 0
2022-02-28 19:42:42,797 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - last_data_time 2022-02-28T18:01:27+01:00
2022-02-28 19:42:42,798 - HoymilesAdd-on - INFO - Solar data publication...2022-02-28 19:42:42.798039
2022-02-28 19:42:42,803 - HoymilesAdd-on - INFO - Main loop start!

502: Bad Gateway

Cosik, 2022-02-28:
I fixed typo for SSL

Kwiatjk, 2022-02-28:
[image]

Cosik, 2022-02-28:
https://github.com/dmslabsbr/hoymiles/blob/master/CHANGELOG.md it was disabled....

Kwiatjk, 2022-02-28:
thx
waiting for ssl update

Cosik, 2022-02-28:
Fix for ssl is already on main branch.

Kwiatjk, 2022-02-28:
Tx

closed 2022-03-01 by Cosik