dmslabsbr/hoymiles #57: Wrong data in sensor.solarH_xxxxx_total_eq at 0,24 rev

Chris125913, 2022-03-01:
My data is in the pic below. Do you have the same issue?

[image]

it is more than 4 GWh :-)

As I see it is a problem with a unit of measure. 

PS 
To all contributors - many thanks, good job,

Cosik, 2022-03-01:
Strange on my side is correct, I also checked code and all values are default in Wh and then divided by 1000. 

[image]

When you perform update?

Chris125913, 2022-03-01:
As I remember it was yesterday after 5 PM. Today I'm reinstalling the Add-on and no chances.

Cosik, 2022-03-01:
Please check now.

Chris125913, 2022-03-02:
HI. I was updating hoymiles.py but no changes. After that, I deinstall and install the add-on and now total_eq is show data ok. 
BTW Should I expect value year_eq but I can't find this
Why, becouse I find line in hoymiles.py
solar_data['total_eq'] = str( round(float(solar_data['total_eq']) / 1000,2) )
and this line was changed to
solar_data['year_eq'] = str( round(float(solar_data['total_eq']) / 1000,2) )

Cosik, 2022-03-02:
Did you update it after my change? How you updated it? Maybe try uninstall it and then install it? To be honest I don't know how to provide hot fixes. I will try bump version.

Chris125913, 2022-03-02:
version 0.24.1 works good

closed 2022-03-03 by Cosik