dmslabsbr/hoymiles #58: 502: Bad Gateway

bridgewatermedia, 2022-03-02:
This could just be me as I'm very new to Home Assistant and MQTT but the Dashboard link on the left has never worked for me.  It constantly gives an error 502:Bad Gateway.  What is this supposed to look like?  I can get data out and display it in LoveLace, but that sidebar menu doesn't do anything.  Also - what is the 'simple web interface' all about? I'd be happy to help with the documentation about how to install if I knew myself how to do so.  I've provided a bunch of screenshots to show my settings.  

[image]

[image]

Cosik, 2022-03-02:
#56 was disabled due too pour quality and need re implementation in python. Probably will be not enabled before version 1.x

dmslabsbr, 2022-03-04:
improvement @ #47

closed 2022-03-04 by dmslabsbr