dmslabsbr/hoymiles #59: Warnings in Log

bridgewatermedia, 2022-03-03:
Just a warning, but I thought I'd post this here in case it is related to something else.

[image]

Cosik, 2022-03-03:
If I good remember watch dog is not implemented, correct @dmslabsbr

kucinskimarek, 2022-03-03:
Hi,
In addition to the ones listed above which also occur to me, I have these warnings:

[image]

Cosik, 2022-03-03:
@kucinskimarek please reinstall plugin and restart HA. There were changes in main json file.

Cosik, 2022-03-03:
@kucinskimarek not implemented watchdog will cause restart of addon even if it not hangup

kucinskimarek, 2022-03-03:
@Cosik I got it, thanks.
In that case, I will update the original issue on this.

dmslabsbr, 2022-03-03:
> If I good remember watch dog is not implemented, correct @dmslabsbr ?

Yes. It not implemented yet. 

Thank you!

closed 2022-03-08 by Cosik