dmslabsbr/hoymiles #60: Connection refused. App EXIT

Krzysztonek, 2022-03-03:
Dear @dmslabsbr 

After last add-on update I faced following issue:

[image]

Could you please, advise?

Cosik, 2022-03-03:
Did you fill mqtt IP, user and password?

dmslabsbr, 2022-03-03:
Do you use a internal mqtt or a external mqtt server?

> Dear @dmslabsbr
> 
> After last add-on update I faced following issue:
> 
> [image]
> 
> Could you please, advise?

Krzysztonek, 2022-03-03:
@dmslabsbr ,
I'm using MQTT Mosquitto Broker only:

[image]

with the following credentials (dedicated `hoymiles` user already created):

[image]

dmslabsbr, 2022-03-03:
There is a bug when add-on is getting internal mqtt host, user and password.
We will fix it.
Thank you.

dmslabsbr, 2022-03-04:
@Cosik , I just upload a correction for bug of issue #60 . https://github.com/dmslabsbr/hoymiles/tree/issue60. - Could you check it?

Cosik, 2022-03-04:
@dmslabsbr looks good, better than previous solution ;)

Cosik, 2022-03-08:
Yes, please do it :)

Cosik, 2022-03-08:
@dmslabsbr when you will implement this fix, please also fix version of stable. I srew up somehow preparing edge version and stable verison is now 1.0 instead of 2.4.x

dmslabsbr, 2022-03-08:
@Cosik , 0.24 wouldn't just be the oldStable version?

Maybe I didn't understand well.

closed 2022-03-26 by Cosik