dmslabsbr/hoymiles #61: Split code - stable + dev

dmslabsbr, 2022-03-04:
Split the add-on code into two folders like stable and dev.

Create a new repository.json for each folder.
or make something like https://github.com/dmslabsbr/hass-addons

Cosik, 2022-03-04:
@dmslabsbr currently developement I'm doing on dev branch. Also I was looking for some solution how to introduce for example odlstable. But due to information in email from you, I need to change release 1.x and there will be no change in entities naming. Only new will be added.

Cosik, 2022-03-04:
Ok, I found how we could do it , like here https://github.com/home-assistant/addons when you think that we could do it? Before release 1.0?

dmslabsbr, 2022-03-04:
@Cosik, I think that it be a good idea.

Because some people already use the beta versions. And they may not like having new bugs. Although this is good for testing

I suggest for now something like that:

[image]

and when the 1st stable new version, the folders goes something like that.

[image]

I tried to add on home assistant add-on a different github branch than master, but i could'nt. 

so, when user add https://github.com/dmslabsbr/hoymiles in your add-ons painnel, they will see two options to instal... the old-stable and last-edge-version.

> Ok, I found how we could do it , like here https://github.com/home-assistant/addons when you think that we could do it? Before release 1.0?

Cosik, 2022-03-04:
Will you have time to prepare it?

dmslabsbr, 2022-03-04:
In this weekend.

dmslabsbr, 2022-03-06:
Ok. OldStable folder is working.

[image]

closed 2022-03-06 by dmslabsbr

Cosik, 2022-03-06:
@dmslabsbr I think that also files in root repo folder should be removed, or moved to stable dir. Or we let it as it is?