dmslabsbr/hoymiles #65: Stop working after back to 0.23a

Kwiatjk, 2022-03-08:
dmslabs - Home Assistant HoyMiles Solar Data Gateway Add-on
2022-03-08 13:31:55,949 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway v.0.24.2d
2022-03-08 13:31:55,949 - HoymilesAdd-on - INFO - Starting up... 2022-03-08 13:31:55
2022-03-08 13:31:55,952 - HoymilesAdd-on - INFO - Using Internal MQTT Server: core-mosquitto
2022-03-08 13:31:55,953 - HoymilesAdd-on - INFO - Using Internal MQTT User: addons
2022-03-08 13:31:55,953 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2022-03-08 13:31:56,306 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - I got the token!!  :-)
2022-03-08 13:31:56,309 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2022-03-08 13:31:56,708 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - last_data_time 2022-03-08T13:31:19+01:00
Traceback (most recent call last):
  File "/data/../hoymiles.py", line 268, in <module>
    sys.exit(main())  # next section explains the use of sys.exit
  File "/data/../hoymiles.py", line 228, in main
    mqtt.start()
  File "/mqttapi.py", line 56, in start
    if self._config['MQTT_TLS']:
KeyError: 'MQTT_TLS'
[cmd] /run.sh exited 1
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.

HOYMILES_USER: xxxx
HOYMILES_PASSWORD:xxxxx
HOYMILES_PLANT_ID: 'xxxx'
DEVELOPERS_MODE: false
External_MQTT_Server: false
External_MQTT_Host: 127.0.0.1
External_MQTT_User: xxxt
External_MQTT_Pass: xxxx
External_MQTT_TLS: false
External_MQTT_TLS_PORT: '8883'
MQTT_Host: 127.0.0.1
MQTT_User: xxxx
MQTT_Pass: xxxx
MQTT_TLS: false
MQTT_TLS_PORT: '8883'

Cosik, 2022-03-08:
You've downgraded addon, so you have old stable. Please install stable version

bridgewatermedia, 2022-03-08:
I have this exact error as well...for both the stable and the old stable.  I thought it was just me.    Yesterday I couldn't log in to my DTU because too many login attempts were recorded.  Again - I thought that was just my setup, but it was working fine before the updates.

Cosik, 2022-03-08:
Easiest way is to uninstall, refresh addon list, install one which is not marked as old-stable, please refer to #61

Kwiatjk, 2022-03-08:
Thx is working, after uninstall and use stable

gralin, 2022-03-08:
Anyone getting `user login times limited per day` error when trying to login via web or app after using HA integration? 

[image]

bridgewatermedia, 2022-03-08:
Yes - I had that as well (user login times limited per day).  I didn't have screenshots etc though - I believe it is related to not being able to connect to your MQTT server.  My MQTT user and password was cleared out somehow in the config.

closed 2022-03-08 by dmslabsbr