dmslabsbr/hoymiles #67: Error: user login times limited per day

dmslabsbr, 2022-03-08:
We need to investigate it.
Maybe, we need to increase the download data time.

Anyone getting `user login times limited per day` error when trying to login via web or app after using HA integration? 

Anyone else having this error?

[image]

_Originally posted by @gralin in https://github.com/dmslabsbr/hoymiles/issues/65#issuecomment-1061817301_

gralin, 2022-03-08:
I've sent an email to hoymiles support about this. Let's see what they reply.

Cosik, 2022-03-08:
I think problem also could be tokens with so long expired period.

brunolinsalves, 2022-03-10:
> Anyone getting `user login times limited per day` error when trying to login via web or app after using HA integration?

Yes, I got this error message when trying to login to the site (https://global.hoymiles.com/) after using this home assistant integration.

gralin, 2022-03-10:
Hoymiles support replied to my email, where I asked why do I have a limit for login times and how can I reset it. They didn't answer my questions, just wrote me that my accounts password has been changed to 123456 and that now I can now login and change it... It did unlock my account but be careful as this procedure is not very safe...

dmslabsbr, 2022-03-10:
I´ve never had this problem. But some people reported this issue too.

> hey didn't answer my questions, just wrote me that my accounts password has been changed to 123456 and that now I can now login and change it... It did unlock my account but be careful as this procedure is not very safe...

Cosik, 2022-03-10:
This is very strange. I'm using this plugin over month and also making debugging and developing on my account and didn't see such situation.

gralin, 2022-03-10:
@Cosik I'm using this plugin for some time and this was the first time I saw this. But I also don't login there often, so maybe this happened in the past and just the plugin lost connectivity and I blamed it on the issue with Core restart impacting it

acdcnow, 2022-07-23:
@Cosik I got it as well. but just for the first day I played with the hardware and DTU.
I connected the DTU by having the micro 1500 just connected to the live line and not to any panel. Hence the status was red on the micro and if I started the addon I could see that the addon end with the message that no energy was found. I restarted the addon I guess 4 to 5 times and then somewhere at the end of the day I could not logon any more with this message.
That had just effected my user account but not my installed account.
I waited until the next day and all was fine again, I connected now my panels as well so I got on the cloud active data and in the  addon the error message is not there any more and the "main loop started" Since then I could logon as many time I link and I do not get any more that message. So I can not reproduce that right now but the observations may help.

closed 2024-05-07 by Cosik