dmslabsbr/hoymiles #69: Version 1.0 error during install add-on

Chris125913, 2022-03-08:
Hi, 

I have an error during try to install version 1.0

The command '/bin/bash -o pipefail -c pip3 install paho-mqtt' returned a non-zero code: 1

before I execute source ./bin/activate I have to change the path to ./hoymiles/hoymiles.
I'm not sure maybe it was wrong but without this command "source ./bin/activate" didn't work
Also, I must use --user option with all "pip3 install ...." commands

What have I done wrong?

Cosik, 2022-03-08:
This is strange. You install it on HA? I installed it 13 min ago without problems. Also I run it as standalone. Please check updated and is it 1.0.2 version?

dmslabsbr, 2022-03-08:
@Chris125913 , I just tested the version 1.0.2 stable and 1.0.2 edge, and both run well.

Are you using the app as a Home assistant add-on, or trying to use it without Home Assistant?

Please, check this DOC - https://github.com/dmslabsbr/hoymiles/blob/master/stable/DOCS.md and let me know.

Chris125913, 2022-03-08:
I'm using an add-on as a part of HA. Of course I know this DOC. Below what I have done

git clone https://github.com/dmslabsbr/hoymiles.git   - add sudo at the beginning
cd hoymiles   - all OK
python3 -m venv ./hoymiles/ - add sudo at the beginning
source ./bin/activate - and there is a problem. 

pi@xxx:~/hoymiles $ source ./bin/activate
-bash: ./bin/activate: There is no such file or directory
command source ./bin/activate works whem I call it from /home/pi/hoymiles/hoymiles directory

also all pip3 install command I have to use with --user option

(hoymiles) pi@xxxx:~/hoymiles $ pip3 install paho-mqtt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting paho-mqtt
  Using cached https://www.piwheels.org/simple/paho-mqtt/paho_mqtt-1.6.1-py3-none-any.whl
Installing collected packages: paho-mqtt
Could not install packages due to an EnvironmentError: [Errno 13] no access: '/home/pi/hoymiles/hoymiles/lib/python3.7/site-packages/paho_mqtt-1.6.1.dist-info'
Consider using the `--user` option or check the permissions.

Cosik, 2022-03-09:
@Chris125913 you mess access rights. Some commands you executive as user some as root, so this is why it is not working correctly. I propose to install it as regular user without sudo.

Chris125913, 2022-03-09:
Hi,
I did this procedure once more on the clean home directory. (after rm -r /hoymiles)
I didn't use any sudo and all commands were made on the pi user
All commands were successful except one

source ./bin/activate
I must change it to 
source ./hoymiles/bin/activate

and the result was the same as previously

The command '/bin/ash -o pipefail -c pip3 install paho-mqtt' returned a non-zero code: 1

Cosik, 2022-03-09:
Yes, because you made venv called hoymiles in catalog hoymiles so the path should be /hoymiles/bin/activate. Next one maybe there is problem with your shell, i'm not familiar with ash, and it looks like you have problems with installing python packages. Please send full output from console, and before install package please execute 'ls -la' in cloned repo dir.

Chris125913, 2022-03-09:
pi@RPi4:~ $ git clone https://github.com/dmslabsbr/hoymiles.git
Cloning into 'hoymiles'...
remote: Enumerating objects: 585, done.
remote: Counting objects: 100% (585/585), done.
remote: Compressing objects: 100% (404/404), done.
remote: Total 585 (delta 377), reused 346 (delta 174), pack-reused 0
Receiving objects: 100% (585/585), 893.34 KiB | 2.14 MiB/s, done.
Resolving deltas: 100% (377/377), done.
pi@RPi4:~ $ cd hoymiles/
pi@RPi4:/hoymiles $ ls -la
razem 48
drwxr-xr-x 8 pi pi 4096 mar 9 08:15 .
drwxr-xr-x 22 pi pi 4096 mar 9 08:15 ..
drwxr-xr-x 5 pi pi 4096 mar 9 08:15 edge
drwxr-xr-x 8 pi pi 4096 mar 9 08:15 .git
drwxr-xr-x 3 pi pi 4096 mar 9 08:15 .github
-rw-r--r-- 1 pi pi 1960 mar 9 08:15 .gitignore
drwxr-xr-x 2 pi pi 4096 mar 9 08:15 img
drwxr-xr-x 4 pi pi 4096 mar 9 08:15 oldStable
-rw-r--r-- 1 pi pi 4084 mar 9 08:15 README.md
-rw-r--r-- 1 pi pi 154 mar 9 08:15 repository.json
drwxr-xr-x 5 pi pi 4096 mar 9 08:15 stable
-rw-r--r-- 1 pi pi 43 mar 9 08:15 workspace.code-workspace
pi@RPi4:/hoymiles $ python3 -m venv ./hoymiles/
pi@RPi4:/hoymiles $ ls -la
razem 52
drwxr-xr-x 9 pi pi 4096 mar 9 08:16 .
drwxr-xr-x 22 pi pi 4096 mar 9 08:15 ..
drwxr-xr-x 5 pi pi 4096 mar 9 08:15 edge
drwxr-xr-x 8 pi pi 4096 mar 9 08:15 .git
drwxr-xr-x 3 pi pi 4096 mar 9 08:15 .github
-rw-r--r-- 1 pi pi 1960 mar 9 08:15 .gitignore
drwxr-xr-x 6 pi pi 4096 mar 9 08:16 hoymiles
drwxr-xr-x 2 pi pi 4096 mar 9 08:15 img
drwxr-xr-x 4 pi pi 4096 mar 9 08:15 oldStable
-rw-r--r-- 1 pi pi 4084 mar 9 08:15 README.md
-rw-r--r-- 1 pi pi 154 mar 9 08:15 repository.json
drwxr-xr-x 5 pi pi 4096 mar 9 08:15 stable
-rw-r--r-- 1 pi pi 43 mar 9 08:15 workspace.code-workspace
pi@RPi4:/hoymiles $ source ./bin/activate
-bash: ./bin/activate: Nie ma takiego pliku ani katalogu
pi@RPi4:/hoymiles $ source ./hoymiles/bin/activate
(hoymiles) pi@RPi4:/hoymiles $ pip3 install paho-mqtt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting paho-mqtt
Using cached https://www.piwheels.org/simple/paho-mqtt/paho_mqtt-1.6.1-py3-none-any.whl
Installing collected packages: paho-mqtt
Successfully installed paho-mqtt-1.6.1
(hoymiles) pi@RPi4:/hoymiles $ pip3 install requests
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting requests
Using cached https://files.pythonhosted.org/packages/2d/61/08076519c80041bc0ffa1a8af0cbd3bf3e2b62af10435d269a9d0f40564d/requests-2.27.1-py2.py3-none-any.whl
Collecting idna<4,>=2.5; python_version >= "3" (from requests)
Using cached https://files.pythonhosted.org/packages/04/a2/d918dcd22354d8958fe113e1a3630137e0fc8b44859ade3063982eacd2a4/idna-3.3-py3-none-any.whl
Collecting certifi>=2017.4.17 (from requests)
Using cached https://files.pythonhosted.org/packages/37/45/946c02767aabb873146011e665728b680884cd8fe70dde973c640e45b775/certifi-2021.10.8-py2.py3-none-any.whl
Collecting charset-normalizer=2.0.0; python_version >= "3" (from requests)
Using cached https://files.pythonhosted.org/packages/06/b3/24afc8868eba069a7f03650ac750a778862dc34941a4bebeb58706715726/charset_normalizer-2.0.12-py3-none-any.whl
Collecting urllib3<1.27,>=1.21.1 (from requests)
Using cached https://files.pythonhosted.org/packages/4e/b8/f5a25b22e803f0578e668daa33ba3701bb37858ec80e08a150bd7d2cf1b1/urllib3-1.26.8-py2.py3-none-any.whl
Installing collected packages: idna, certifi, charset-normalizer, urllib3, requests
Successfully installed certifi-2021.10.8 charset-normalizer-2.0.12 idna-3.3 requests-2.27.1 urllib3-1.26.8
(hoymiles) pi@RPi4:/hoymiles $ pip3 install python-dateutil
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting python-dateutil
Using cached https://files.pythonhosted.org/packages/36/7a/87837f39d0296e723bb9b62bbb257d0355c7f6128853c78955f57342a56d/python_dateutil-2.8.2-py2.py3-none-any.whl
Collecting six>=1.5 (from python-dateutil)
Using cached https://files.pythonhosted.org/packages/d9/5a/e7c31adbe875f2abbb91bd84cf2dc52d792b5a01506781dbcf25c91daf11/six-1.16.0-py2.py3-none-any.whl
Installing collected packages: six, python-dateutil
Successfully installed python-dateutil-2.8.2 six-1.16.0
(hoymiles) pi@RPi4:/hoymiles $ pwd
/home/pi/hoymiles
(hoymiles) pi@RPi4:~/hoymiles $

the final error was the same. version 1.0.2
trying to install stable version
The command '/bin/ash -o pipefail -c pip3 install paho-mqtt' returned a non-zero code: 1

Cosik, 2022-03-09:
I see that installing packages is working correctly. I don't see any errors from your console output.

Also I propose to do it in one of folder stable or edge. If you see problem with that you have adding hoymiles in venv activate please change way how you create it.

Chris125913, 2022-03-09:
All my attempts ended identically as below. Packages installed correctly

Logger: homeassistant.components.hassio
Source: components/hassio/websocket_api.py:120
Integration: Home Assistant Supervisor (documentation, issues)
First occurred: 20:47:15 (5 occurrences)
Last logged: 21:39:41

Failed to to call /addons/44349f66_hoymiles_solardata/install - The command '/bin/ash -o pipefail -c pip3 install paho-mqtt' returned a non-zero code: 1

Cosik, 2022-03-09:
Sorry, I don't know how to help you. I don't have idea how to reproduce your error. I was testing it by install over addon store and running it manually. If you are downloading it manually you could use tags to get previous version.

Chris125913, 2022-03-10:
Yes, version 0.24 worked fine.
OK no problem thx for your help

Cosik, 2022-03-10:
Difference between 0.24 and v1.0 is only in file structure. Now we store different versions on one repo as it is on main addon repo for HA.

Cosik, 2022-03-10:
@Chris125913 please share full instruction how you setup your HA. Maybe when I will have time I will try to check it.

Chris125913, 2022-03-11:
I decided not to try to install add-on on my main HA.  I install new HA on RPI zero and it works

[08:07:09] INFO: {"supervisor":"2022.03.3","homeassistant":"2022.3.3","hassos":"4.12","docker":"19.03.11","hostname":"homeassistant","operating_system":"HassOS 4.12","features":["reboot","shutdown","services","network","hostname","timedate","haos"],"machine":"raspberrypi","arch":"armhf","state":"running","supported_arch":["armhf"],"supported":false,"channel":"stable","logging":"info","timezone":"Europe/Warsaw"}
dmslabs - Home Assistant HoyMiles Solar Data Gateway Add-on
2022-03-11 08:07:16,177 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway  v.1.0.2
2022-03-11 08:07:16,188 - HoymilesAdd-on - INFO - Starting up... 2022-03-11 08:07:16

I send all data from RPIzero to my main HA .
I can believe but is works

Cosik, 2022-03-11:
But @Chris125913 please share information how you setup your main HA, maybe some small changes are required. Also you could run this plugin standalone and send data to main HA.

Chris125913, 2022-03-11:
OK, I will try to describe how my main HA was built. But to be honest I didn't know how to set up this plugin without HA. Can you give some info on how to do this?

Cosik, 2022-03-11:
This is python code. When I test it locally I run it from command line in python venv. For example

source /home/cosik/.venvs/hoymiles/bin/activate 
cd stable - in hoymiles repo
python hoymiles.py

You have also remember to change values in config.json (users, passwords, ips, etc.) and that's all. Code will be running. You could put it in autostart and it will be starting when system will boot up.

dmslabsbr, 2022-03-11:
And they still say that computing is an exact science.  :-)

> I decided not to try to install add-on on my main HA. I install new HA on RPI zero and it works
> 
> [08:07:09] INFO: {"supervisor":"2022.03.3","homeassistant":"2022.3.3","hassos":"4.12","docker":"19.03.11","hostname":"homeassistant","operating_system":"HassOS 4.12","features":["reboot","shutdown","services","network","hostname","timedate","haos"],"machine":"raspberrypi","arch":"armhf","state":"running","supported_arch":["armhf"],"supported":false,"channel":"stable","logging":"info","timezone":"Europe/Warsaw"} dmslabs - Home Assistant HoyMiles Solar Data Gateway Add-on 2022-03-11 08:07:16,177 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway v.1.0.2 2022-03-11 08:07:16,188 - HoymilesAdd-on - INFO - Starting up... 2022-03-11 08:07:16
> 
> I send all data from RPIzero to my main HA via MQTT. I can believe but is works

Chris125913, 2022-03-14:
> But @Chris125913 please share information how you setup your main HA, maybe some small changes are required. Also you could run this plugin standalone and send data to main HA. HA is not required to get it running.

OK, below you can find short info about my configuration

<html>
<body>
<h1 class="card-header" style="color: rgb(33, 33, 33); font-family: Roboto, Noto, sans-serif; font-size: var(--ha-card-header-font-size, 24px); letter-spacing: -0.012em; line-height: 48px; padding: 12px 16px 16px; display: flex; margin-block: 0px; font-weight: normal; justify-content: space-between; align-items: center; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><div class="card-header-text">System Health</div><ha-button-menu corner="BOTTOM_START" slot="toolbar-icon" style="display: inline-block; position: relative;"><div><slot name="trigger"><ha-icon-button slot="trigger" style="display: inline-block; outline: none;"><mwc-icon-button title="Copy menu" style="display: inline-block; outline: none; --mdc-ripple-color: currentcolor; -webkit-tap-highlight-color: transparent; vertical-align: top; --mdc-theme-on-primary: currentColor; --mdc-theme-text-disabled-on-light: var(--disabled-text-color);"><button class="mdc-icon-button mdc-icon-button--display-flex" aria-label="Copy menu" style="font-size: 24px; width: var(--mdc-icon-button-size, 48px); height: var(--mdc-icon-button-size, 48px); padding: calc( (var(--mdc-icon-button-size, 48px) - var(--mdc-icon-size, 24px)) / 2 ); display: inline-flex; position: relative; box-sizing: border-box; border: none; outline: none; background-color: transparent; fill: currentcolor; color: inherit; text-decoration: none; cursor: pointer; user-select: none; z-index: 0; overflow: visible; align-items: center; justify-content: center; vertical-align: top;"><span><slot><ha-svg-icon style="display: block; align-items: center; justify-content: center; position: relative; vertical-align: middle; fill: currentcolor; width: var(--mdc-icon-size, 24px); height: var(--mdc-icon-size, 24px);"><svg preserveAspectRatio="xMidYMid meet" focusable="false" role="img" aria-hidden="true" viewBox="0 0 24 24"><g><path d="M19,21H8V7H19M19,5H8A2,2 0 0,0 6,7V21A2,2 0 0,0 8,23H19A2,2 0 0,0 21,21V7A2,2 0 0,0 19,5M16,1H4A2,2 0 0,0 2,3V17H4V3H16V1Z"></path></g></svg></ha-svg-icon></slot></span></button></mwc-icon-button></ha-icon-button></slot></div><mwc-menu></mwc-menu></ha-button-menu></h1><div class="card-content" style="padding: 0px 16px 16px; margin-top: -8px; color: rgb(33, 33, 33); font-family: Roboto, Noto, sans-serif; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">

Version | core-2022.3.3
-- | --
Installation Type | Home Assistant Supervised
Development | false
Supervisor | true
Docker | true
User | root
Virtual Environment | false
Python Version | 3.9.9
Operating System Family | Linux
Operating System Version | 5.10.63-v7l+
CPU Architecture | armv7l
Timezone | Europe/Warsaw

</div>
</body>
</html>

<html>
<body>
<div class="card-header" style="justify-content: space-between; display: flex; align-items: center; color: rgb(33, 33, 33); font-family: Roboto, Noto, sans-serif; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><h3>Home Assistant Supervisor</h3><a class="manage" href="https://chris125.duckdns.org:8123/hassio" style="color: var(--primary-color); text-decoration: none;"><mwc-button style="display: inline-flex; outline: none; -webkit-tap-highlight-color: transparent; vertical-align: top;"><button id="button" class="mdc-button   " aria-label="" style="-webkit-font-smoothing: antialiased; font-family: var(--mdc-typography-button-font-family, var(--mdc-typography-font-family, Roboto, sans-serif)); font-size: var(--mdc-typography-button-font-size, 0.875rem); line-height: inherit; font-weight: var(--mdc-typography-button-font-weight, 500); letter-spacing: var(--mdc-typography-button-letter-spacing, 0.0892857em); text-decoration: var(--mdc-typography-button-text-decoration, none); text-transform: var(--mdc-typography-button-text-transform, uppercase); position: relative; display: inline-flex; align-items: center; justify-content: center; box-sizing: border-box; min-width: 64px; border: none; outline: none; user-select: none; appearance: none; overflow: hidden; vertical-align: middle; background: transparent; padding-top: 0px; padding-right: var(--mdc-button-horizontal-padding, 8px); padding-bottom: 0px; padding-left: var(--mdc-button-horizontal-padding, 8px); height: 36px; border-radius: var(--mdc-shape-s [... cut ...]
Supported	Unsupported – [more info](https://chris125.duckdns.org:8123/hassio/system)
Supervisor API	ok
Version API	ok
Installed Add-ons	ESPHome (2022.2.6), Glances (0.14.2), Grafana (7.4.1), Home Panel (2.3.3), SSH & Web Terminal (10.0.2), File editor (5.3.3), Duck DNS (1.14.0), eWeLink Smart Home (1.2.3), Mosquitto broker (6.0.1), Samba share (9.5.1), Home Assistant Google Drive Backup (0.105.2), TasmoAdmin (0.16.0), Portainer (2.0.0), InfluxDB (4.3.0), Old Stable HoyMiles Gateway (0.23a)

Thanks for your interest

closed 2022-03-23 by Chris125913