dmslabsbr/hoymiles #72: Stop working after update core-2022.3.5

Kwiatjk, 2022-03-15:
Starting add-on container...
Tue Mar 15 12:38:57 CET 2022
Exporting config data...
Getting mqqt data...
[12:38:58] ERROR: Got unexpected response from the API: Service not enabled
total 16
d---------    2 root     root          4096 Mar 15 12:39 .
d---------    4 root     root          4096 Mar 15 12:38 ..
----------    1 root     root           718 Mar 15 12:39 base.html
----------    1 root     root           221 Mar 15 12:39 index.html
[12:39:00] INFO: secrets.ini exists!
[12:39:00] INFO: /data/secrets.ini exists!
[12:39:01] INFO: null
[12:39:01] INFO: {"supervisor":"2022.03.4","homeassistant":"2022.3.5","hassos":null,"docker":"20.10.3","hostname":null,"operating_system":null,"features":[],"machine":"qemux86-64","arch":"amd64","state":"running","supported_arch":["amd64","i386"],"supported":false,"channel":"stable","logging":"info","timezone":"Europe/Warsaw"}
[12:39:01] INFO: {"supervisor":"2022.03.4","homeassistant":"2022.3.5","hassos":null,"docker":"20.10.3","hostname":null,"operating_system":null,"features":[],"machine":"qemux86-64","arch":"amd64","state":"running","supported_arch":["amd64","i386"],"supported":false,"channel":"stable","logging":"info","timezone":"Europe/Warsaw"}
dmslabs - Home Assistant HoyMiles Solar Data Gateway Add-on
2022-03-15 12:39:02,045 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway v.0.24.2d
2022-03-15 12:39:02,045 - HoymilesAdd-on - INFO - Starting up... 2022-03-15 12:39:02
2022-03-15 12:39:02,046 - HoymilesAdd-on - INFO - Using Internal MQTT Server: 
2022-03-15 12:39:02,046 - HoymilesAdd-on - INFO - Using Internal MQTT User: 
2022-03-15 12:39:02,046 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2022-03-15 12:39:02,989 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - I got the token!!  :-)
2022-03-15 12:39:02,990 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2022-03-15 12:39:03,354 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - last_data_time 2022-03-15T12:31:06+01:00
2022-03-15 12:39:03,357 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Starting MQTT

Cosik, 2022-03-15:
Are you sure? On my side is working. But I'm using not default user for mqqt. Did you tried restart addon? I think this should be rised to HA that something stop working. We didn't change anything.

Kwiatjk, 2022-03-15:
> My config what working before update
> HOYMILES_USER: xxx
> HOYMILES_PASSWORD: xxxx
> HOYMILES_PLANT_ID: 'xxxx'
> DEVELOPERS_MODE: false
> External_MQTT_Server: false
> MQTT_Host: core-mosquitto
> MQTT_User: xxxx
> MQTT_Pass: xxxx
> MQTT_TLS: true
> MQTT_TLS_PORT: '8883'
> Server was restarted
> 22-03-15 12:44:04,686 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway v.0.24.2d
> 2022-03-15 12:44:04,687 - HoymilesAdd-on - INFO - Starting up... 2022-03-15 12:44:04
> 2022-03-15 12:44:04,688 - HoymilesAdd-on - INFO - Using Internal MQTT Server:
> 2022-03-15 12:44:04,688 - HoymilesAdd-on - INFO - Using Internal MQTT User:
> 2022-03-15 12:44:04,688 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
> 2022-03-15 <https://global.hoymiles.com/platform/api/gateway/iam/auth_login%0D2022-03-15> 12:44:05,432 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - I got the token!!  :-)
> 2022-03-15 12:44:05,432 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
> 2022-03-15 <https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data%0D2022-03-15> 12:44:06,041 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - last_data_time 2022-03-15T12:31:06+01:00
> 2022-03-15 12:44:06,044 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Starting MQTT
> 2022-03-15 12:44:06,045 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Trying TLS: 8883

SOLVED start working after set External_MQTT to true

Cosik, 2022-03-15:
Probably they broke somehow getting data. We are collecting them as it is in documentation.

closed 2022-03-26 by Cosik

Kwiatjk, 2022-10-11:
My config what working before update
HOYMILES_USER: xxx
HOYMILES_PASSWORD: xxxx
HOYMILES_PLANT_ID: 'xxxx'
DEVELOPERS_MODE: false
External_MQTT_Server: false
MQTT_Host: core-mosquitto
MQTT_User: xxxx
MQTT_Pass: xxxx
MQTT_TLS: true
MQTT_TLS_PORT: '8883'

Server was restarted

22-03-15 12:44:04,686 - HoymilesAdd-on - INFO - **********
dmslabs&Cosik Hoymiles Gateway v.0.24.2d
2022-03-15 12:44:04,687 - HoymilesAdd-on - INFO - Starting up...
2022-03-15 12:44:04
2022-03-15 12:44:04,688 - HoymilesAdd-on - INFO - Using Internal MQTT Server:
2022-03-15 12:44:04,688 - HoymilesAdd-on - INFO - Using Internal MQTT User:
2022-03-15 12:44:04,688 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO -
Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2022-03-15 12:44:05,432 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO -
I got the token!!  :-)
2022-03-15 12:44:05,432 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO -
Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2022-03-15 12:44:06,041 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO -
last_data_time 2022-03-15T12:31:06+01:00
2022-03-15 12:44:06,044 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Starting MQTT
2022-03-15 12:44:06,045 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Trying TLS: 8883