dmslabsbr/hoymiles #73: Old Stable version uninstalling issue

Krzysztonek, 2022-03-18:
Dear @dmslabsbr ,

To install a brand new add-on version I tried to uninstall the obsolete old version. However, after executing this process I'm still able to see this add-on on a HASSIO dashboard:

[image]

When clicking this add-on it reports as UNINSTALLED (INSTALL prompt appears):

[image]

...but it still remains visible on a dashboard, even if it is not yet installed back.
Before installing a new version I need to completely remove the obsolete one, right?
Please kindly advise how to do it?

Cosik, 2022-03-18:
Did you try ctrl+F5?

Krzysztonek, 2022-03-18:
Hi @Cosik ,
Yes, I've just tried ctrl+F5 and... it helped! Thank you!
The obsolete version has been replaced with a new 1.0.4 version.
I have configured it with my credentials as per requirements and indicated the internal Mosquitto MQTT server.

[image]

Then, when trying to open a web IU of this newly installed add-on, the screen shows nothing more that that:

[image]

Here is my log:

[image]

What am I doing wrongly?

Cosik, 2022-03-18:
Web UI was disabled.

Krzysztonek, 2022-03-18:
So, how to enable it?
The button is accessible:
[image]

Cosik, 2022-03-18:
Was disabled in code. Button was not removed. Sorry.

Krzysztonek, 2022-03-18:
Now it's clear. Thank you! :)

closed 2022-03-26 by Cosik