dmslabsbr/hoymiles #76: Any chance to make this a HACS integration?

eduspina, 2022-05-13:
Hi there,

First of all, thanks for all your hard work on this add-on!!!

Is there any chance to make this add-on a HACS integration? For the ones, that like me, runs Home Assistant on a docker, it is impossible to use this add-on as we don't have the supervisor.

Thanks and sorry if this is not the right place to ask for it.

dmslabsbr, 2022-05-13:
It's a good idea, it's not a priority right now but we'll keep it noted for future use.

But, 

You can use this app in another docker container.
Just need to understand a little bit of python, docker and make some changes.

Create a new docker image with oldStable files., maybe you need to clone https://github.com/dmslabsbr/hoymiles/tree/master/oldStable,

 You need to install: https://github.com/eclipse/paho.mqtt.python and https://github.com/psf/requests in your new docker container.

You need to change .../oldStable/Dockerfile  and configure ../oldStable/secrets.ini with your mqtt server data.

There are some jobs and searches to do, but it will work.

eduspina, 2022-05-17:
Awesome!!!

Thanks Daniel, it worked... I had to tweak a few more things but it seems it is working!!!

[image]

Great job and keep up with the project!!! ;)

dmslabsbr, 2022-05-17:
Great!  
Thank you for your feedback.

closed 2022-05-17 by dmslabsbr