dmslabsbr/hoymiles #78: Add on not starting

brzozow, 2022-05-27:
It looks like an add on not starting.
In log there is one line:
s6-overlay-suexec: fatal: can only run as pid 1

I have the same error in edge and stable version.

Cosik, 2022-05-30:
There are not enough information from your side. What is your setup, which version did you tried, was it working before or not, etc.

brzozow, 2022-05-30:
Hi.
My system:
Home Assistant Core 2022.5.5
Home Assistant Supervisor 2022.05.3
Home Assistant OS 8.1
Kernel version 5.15.32-v8
Agent version 1.2.1

Hoymiles integration was working fine in HomeAssisstant 2022.4. After upgrade to 2022.5.5 Hoymiles does not work on my installation. All entities from Hoymiles integration are in "unavailable" state. After restarting integration I see "s6-overlay-suexec: fatal: can only run as pid 1" message in Logs. It looks that integration can't start properly. In supervisor logs I can see errors like this:

22-05-30 21:32:08 INFO (SyncWorker_1) [supervisor.docker.interface] Cleaning addon_44349f66_hoymiles_solardata_edge application
22-05-30 21:32:09 INFO (SyncWorker_1) [supervisor.docker.addon] Starting Docker add-on 44349f66/aarch64-addon-hoymiles_solardata_edge with version 1.0.8
22-05-30 21:33:27 ERROR (SyncWorker_7) [supervisor.docker.interface] Container addon_44349f66_hoymiles_solardata_edge is not running
22-05-30 21:33:39 WARNING (MainThread) [supervisor.misc.tasks] Watchdog found a problem with 44349f66_hoymiles_solardata_edge!

I found out that some other integrations has some similar problems:

Krzysztonek, 2022-05-30:
I faced a similar issue:

```
Logger: homeassistant.components.hassio
Source: components/hassio/__init__.py:963
Integration: Home Assistant Supervisor (documentation, issues)
First occurred: 30 maja 2022, 08:49:13 (193 occurrences)
Last logged: 00:54:02

Could not fetch stats for 44349f66_hoymiles_solardata: Container addon_44349f66_hoymiles_solardata is not running
Could not fetch stats for a0d7b954_vscode: 404 Client Error for http+docker://localhost/v1.41/containers/addon_a0d7b954_vscode/json: Not Found ("No such container: addon_a0d7b954_vscode")
```

Please, advise.

Cosik, 2022-05-31:
Ok, now we get more info. It seems that update crash something. I will try to setup some HA with supervisor.

Also did you try to reinstall and/or rebuild addon? It looks like docker image is not starting.

Krzysztonek, 2022-05-31:
Yes, I did try all the options (reinstall, rebuild, install from scratch), but it hasen't resolved the issue so far unfortunately.

Cosik, 2022-06-02:
I found information what is a problem, but still I don't know how to fix it. Still looking for solution

Cosik, 2022-06-02:
Hi, now should work, edge and stable. As usually HA break some backward compatibility and introduce new problems.

Please check

umrath, 2022-06-02:
The error message has changed.
After updating, I get the following now:

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Starting add-on container...
Thu Jun  2 12:51:51 CEST 2022
total 20
drwxr-xr-x    4 root     root          4096 Jun  2 12:51 .
drwxr-xr-x    1 root     root          4096 Jun  2 12:51 ..
drwxr-xr-x    2 root     root          4096 Jun  2 12:50 jsons
-rw-------    1 root     root           289 Jun  2 12:51 options.json
drwxr-xr-x    2 root     root          4096 Jun  2 12:50 templates
[12:51:51] INFO: /data/secrets.ini not exists!
Getting mqtt data...
[12:51:51] ERROR: Got unexpected response from the API: Service not enabled
[12:51:52] INFO: null
[12:51:52] INFO: {"supervisor":"2022.05.3","homeassistant":"2022.6.0","hassos":"8.1","docker":"20.10.14","hostname":"homeassistant","operating_system":"Home Assistant OS 8.1","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","haos","resolved"],"machine":"odroid-n2","arch":"aarch64","state":"running","supported_arch":["aarch64","armv7","armhf"],"supported":true,"channel":"stable","logging":"info","timezone":"Europe/Berlin"}
[12:51:52] INFO: {"supervisor":"2022.05.3","homeassistant":"2022.6.0","hassos":"8.1","docker":"20.10.14","hostname":"homeassistant","operating_system":"Home Assistant OS 8.1","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","haos","resolved"],"machine":"odroid-n2","arch":"aarch64","state":"running","supported_arch":["aarch64","armv7","armhf"],"supported":true,"channel":"stable","logging":"info","timezone":"Europe/Berlin"}
dmslabs - Home Assistant HoyMiles Solar Data Gateway Add-on
python3: can't open file '/data/hoymiles.py': [Errno 2] No such file or directory
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

Cosik, 2022-06-02:
Please use check for update in addon store and then rebuild addon. I don't know how to clean cache/storage in HA to get it running correctly. I fixed something without bumping version.

umrath, 2022-06-02:
I have manually rebuild the addon.
I do not see any update, even when manually checking for updates in the store.
The result remains the same.

llcoolmm, 2022-06-02:
I have exactly the same problem

ragman11, 2022-06-02:
Same Problem with Version 1.0.5

After Upgrading to Version 1.0.5, the Addon won't start anymore.

dmslabs - Home Assistant HoyMiles Solar Data Gateway Add-on
python3: can't open file '/data/hoymiles.py': [Errno 2] No such file or directory
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

What is the version of the add-on?
1.0.5

brzozow, 2022-06-02:
Hi,
I'm using HoyMiles Solar Gateway Edge Addon version 1.0.9. Addon is not working. Logs:

6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Starting add-on container...
Thu Jun  2 16:43:08 CEST 2022
total 20
drwxr-xr-x    4 root     root          4096 Jun  2 16:43 .
drwxr-xr-x    1 root     root          4096 Jun  2 16:43 ..
drwxr-xr-x    2 root     root          4096 Jun  2 16:42 jsons
-rw-------    1 root     root           532 Jun  2 16:43 options.json
drwxr-xr-x    2 root     root          4096 Jun  2 16:41 templates
[16:43:09] INFO: /data/secrets.ini not exists!
Getting mqtt data...
[16:43:11] INFO: null
[16:43:11] INFO: {"supervisor":"2022.05.3","homeassistant":"2022.5.5","hassos":"8.1","docker":"20.10.14","hostname":"homeassistant","operating_system":"Home Assistant OS 8.1","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","haos","resolved"],"machine":"raspberrypi4-64","arch":"aarch64","state":"running","supported_arch":["aarch64","armv7","armhf"],"supported":true,"channel":"stable","logging":"info","timezone":"Europe/Warsaw"}
[16:43:11] INFO: {"supervisor":"2022.05.3","homeassistant":"2022.5.5","hassos":"8.1","docker":"20.10.14","hostname":"homeassistant","operating_system":"Home Assistant OS 8.1","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","haos","resolved"],"machine":"raspberrypi4-64","arch":"aarch64","state":"running","supported_arch":["aarch64","armv7","armhf"],"supported":true,"channel":"stable","logging":"info","timezone":"Europe/Warsaw"}
dmslabs - Home Assistant HoyMiles Solar Data Gateway Add-on
python3: can't open file '/data/hoymiles.py': [Errno 2] No such file or directory
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

Cosik, 2022-06-02:
Ok I bumped version, now should work.

ragman11, 2022-06-02:
Just tested 1.0.6 and edge 1.0.10.  both still have the same error
 

only version that is running is 0.24.1d

Cosik, 2022-06-02:
If current changes will not help it will required redesign of docker file due to break changes in HA.

umrath, 2022-06-02:
Can confirm: 1.0.7 is working for me.

brzozow, 2022-06-02:
Edge version works fine. Thank you.

closed 2022-06-02 by brzozow