dmslabsbr/hoymiles #84: Stoped to get data after few days

Mangosteen-Yang, 2022-07-12:
Hi,

I met this issue many times. Every time restart can make it work for several days then this error shows.

[image]

Cosik, 2022-07-20:
Sorry for delay, About connection error we are not able to fix it. Also there is missing information what are you using and will be nice to get log in some other format that picture.

Mangosteen-Yang, 2022-07-20:
@Cosik Thanks for the response.

Maybe it can retry after some delay or restart automatically.

Please find raw log below:
```
2022-07-07 03:12:29,174 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/urllib3/connection.py", line 174, in _new_conn
    conn = connection.create_connection(
  File "/usr/local/lib/python3.9/site-packages/urllib3/util/connection.py", line 72, in create_connection
    for res in socket.getaddrinfo(host, port, family, socket.SOCK_STREAM):
  File "/usr/local/lib/python3.9/socket.py", line 954, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
socket.gaierror: [Errno -3] Try again
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 703, in urlopen
    httplib_response = self._make_request(
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 386, in _make_request
    self._validate_conn(conn)
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 1040, in _validate_conn
    conn.connect()
  File "/usr/local/lib/python3.9/site-packages/urllib3/connection.py", line 358, in connect
    self.sock = conn = self._new_conn()
  File "/usr/local/lib/python3.9/site-packages/urllib3/connection.py", line 186, in _new_conn
    raise NewConnectionError(
urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPSConnection object at 0x7f429ca03790>: Failed to establish a new connection: [Errno -3] Try again
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/requests/adapters.py", line 440, in send
    resp = conn.urlopen(
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 785, in urlopen
    retries = retries.increment(
  File "/usr/local/lib/python3.9/site-packages/urllib3/util/retry.py", line 592, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='global.hoymiles.com', port=443): Max retries exceeded with url: /platform/api/gateway/pvm/station_select_device_all (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f429ca03790>: Failed to establish a new connection: [Errno -3] Try again'))
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/hoymiles.py", line 274, in run
    self.execute(*self.args, **self.kwargs)
  File "/hoymiles.py", line 202, in publicate_data
    hoymiles_h.update_devices_status()
  File "/hoymilesapi.py", line 299, in update_devices_status
    status, hws_data = self.request_plant_hw()
  File "/hoymilesapi.py", line 337, in request_plant_hw
    retv = self.send_payload(GET_ALL_DEVICE_API, header, payload)
  File "/hoymilesapi.py", line 352, in send_payload
    retv, s_code = self.send_request(
  File "/hoymilesapi.py", line 180, in send_request
    response = sess.send(prepped)
  File "/usr/local/lib/python3.9/site-packages/requests/sessions.py", line 645, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/requests/adapters.py", line 519, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='global.hoymiles.com', port=443): Max retries exceeded with url: /platform/api/gateway/pvm/station_select_device_all (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f429ca03790>: Failed to establish a new connection: [Errno -3] Try again'))
```

Cosik, 2022-07-21:
I think it is a job for watchdog but to be honest I don't know how to implement it. Also did you test edge version? On Edge are some improvements which are still not in stable.

bridgewatermedia, 2022-07-21:
I have this same problem too, but I have bypassed the problem by creating an automation.  Trigger: State, solarH_xxxx_capacitor unavailable for 1 minute --> Action: Call Service, Home Assistant Supervisor: Restart add-on, HoyMiles Solar Gateway Addon

This has been working great for me - I had it set up to notify me when it stopped (usually after some update to some component) but now I've turned that off and it just works.

Mangosteen-Yang, 2022-07-21:
@Cosik not yet
Thanks, I will try the edge version.

Mangosteen-Yang, 2022-07-21:
@bridgewatermedia That is so clever!
I will try the edge version, if that doesn't work for me I will try this.

giacomitti, 2022-10-19:
I think I have the same issue here...

https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/urllib3/connection.py", line 174, in _new_conn
    conn = connection.create_connection(
  File "/usr/local/lib/python3.9/site-packages/urllib3/util/connection.py", line 72, in create_connection
    for res in socket.getaddrinfo(host, port, family, socket.SOCK_STREAM):
  File "/usr/local/lib/python3.9/socket.py", line 954, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
socket.gaierror: [Errno -3] Try again
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 703, in urlopen
    httplib_response = self._make_request(
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 386, in _make_request
    self._validate_conn(conn)
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 1042, in _validate_conn
    conn.connect()
  File "/usr/local/lib/python3.9/site-packages/urllib3/connection.py", line 358, in connect
    self.sock = conn = self._new_conn()
  File "/usr/local/lib/python3.9/site-packages/urllib3/connection.py", line 186, in _new_conn
    raise NewConnectionError(
urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPSConnection object at 0x7fec5ca06550>: Failed to establish a new connection: [Errno -3] Try again
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/requests/adapters.py", line 489, in send
    resp = conn.urlopen(
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 787, in urlopen
    retries = retries.increment(
  File "/usr/local/lib/python3.9/site-packages/urllib3/util/retry.py", line 592, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='global.hoymiles.com', port=443): Max retries exceeded with url: /platform/api/gateway/pvm/station_select_device_all (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7fec5ca06550>: Failed to establish a new connection: [Errno -3] Try again'))
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/hoymiles.py", line 274, in run
    self.execute(*self.args, **self.kwargs)
  File "/hoymiles.py", line 202, in publicate_data
    hoymiles_h.update_devices_status()
  File "/hoymilesapi.py", line 299, in update_devices_status
    status, hws_data = self.request_plant_hw()
  File "/hoymilesapi.py", line 337, in request_plant_hw
    retv = self.send_payload(GET_ALL_DEVICE_API, header, payload)
  File "/hoymilesapi.py", line 352, in send_payload
    retv, s_code = self.send_request(
  File "/hoymilesapi.py", line 180, in send_request
    response = sess.send(prepped)
  File "/usr/local/lib/python3.9/site-packages/requests/sessions.py", line 701, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/requests/adapters.py", line 565, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='global.hoymiles.com', port=443): Max retries exceeded with url: /platform/api/gateway/pvm/station_select_device_all (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7fec5ca06550>: Failed to establish a new connection: [Errno -3] Try again'))

Cosik, 2022-10-19:
Need full log and more details about what you used.

giacomitti, 2022-10-19:
> Need full log and more details about what you used.

Where is locate the full log? this happen after the last data publication

https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2022-10-18 17:31:59,266 - HoymilesAdd-on - INFO - Solar data publication...2022-10-18 17:31:59.266133
2022-10-18 17:31:59,268 - HoymilesAdd-on - INFO - DTU-W100_xxxxx data publication...2022-10-18 17:31:59.268396
2022-10-18 17:31:59,271 - HoymilesAdd-on - INFO - MI-1500_xxxxx data publication...2022-10-18 17:31:59.271100
2022-10-18 17:31:59,272 - HoymilesAdd-on - INFO - MI-1500_xxxxx data publication...2022-10-18 17:31:59.272747
2022-10-18 17:31:59,276 - HoymilesAdd-on - INFO - MI-1500_xxxxx data publication...2022-10-18 17:31:59.276095
2022-10-18 17:31:59,278 - HoymilesAdd-on - INFO - MI-1500_xxxxx data publication...2022-10-18 17:31:59.277898
2022-10-18 17:31:59,281 - HoymilesAdd-on - INFO - MI-1500_xxxxx data publication...2022-10-18 17:31:59.281260
2022-10-18 17:39:59,283 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: 

Running debian 11 (bullseye) x86, HA 2022.10.4, but this error is happening in old HA versions too.

Cosik, 2022-10-19:
Still don't know what you are using, which version of addon, how often it happen, etc.

giacomitti, 2022-10-19:
> Still don't know what you are using, which version of addon, how often it happen, etc.

Addon version 1.0.7, happens about every day morning, after restart the addon sometimes ocorrurs during the day.

Cosik, 2022-10-19:
So if it will appear every day please enable debug option and attach full log which is on a docker. Also please test one more time edge version. It looks like you have some problems with connection to HM page from your HA device. I will try to handle it somehow in the future,

closed 2023-10-28 by Cosik