dmslabsbr/hoymiles #85: Always needs a restart after restarting Home Assistant

umrath, 2022-07-20:
Whenever I restart Home Assistant for whichever reason, the Addon will stop to provide data to Home Assistant.

Workaround: This can be cured by restarting the Addon itself.

Cosik, 2022-07-20:
@umrath How long you wait? Are you aware that data is send every 15 min?

umrath, 2022-07-20:
It could very well be that I do not wait long enough.

Would it be an option to display cached data in case of a restart (until new data is available).
Alternatively, the addon could pull data on HA restart (as it does on addon restart anyway).

umrath, 2022-07-20:
Just tested it.
Restarted HA at 21:00.
Now it's 21:20 and I still see the entities as unavailable.

Cosik, 2022-07-21:
So how this should work? This base on mqtt messages, and don't have any response from HA that it was restarted. If we want to make some ping it will also increase needed resources. We reduce used resources a lot during last code refactor.

umrath, 2022-07-21:
How do others (zigbee2mqtt) to it?
Is there no way to listen for events, like a broadcast "I'm back!"?

I'm just a user and not into this topic on a technical level. I can only describe how the current way feels to me and how I would like it to act.

Cosik, 2022-07-22:
I'm not sure if zigbee2mqtt is good example. It works in different way. Please se #84 for some solution. Implementation will take some resources and knowledge which I don't have now.

ZaphodB72, 2024-02-16:
Having the same problem… isn‘t it possible for the addon to wait 1min before it fully starts? Maybe that would solve the problem?
I think maybe the addon starts before the mqtt server startet and that’s the problem?

Cosik, 2024-02-17:
@ZaphodB72 did you wait a 15 min? No one of you provide any log files. Addon is waiting for mqtt broker and try to connect to it in infinity loop which will be stop when connection is done. Data is send in 15min period due to refresh rate of Hoymiless cloud so it is out of sense to send it more often. So if addon will start earlier that homeassistant container start subscribe topics on mqtt you will need to wait 15min till addon will send new data.

I have no idea if this issue is different that behavior of your instances because no one of you was able to prepare good ticket and provide nought data to analyze it.

abpei, 2024-06-27:
Not sure if this helps but this is what happens on my end when the issue occurs

```
2024-06-27 06:53:12,505 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_of_tree
2024-06-27 06:53:22,731 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2024-06-27 06:53:29,080 - HoymilesAdd-on.hoymilesapi.Hoymiles - ERROR - Access error: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2024-06-27 06:53:29,080 - HoymilesAdd-on.hoymilesapi.Hoymiles - ERROR - Status code: 502Bad Gateway
Exception in thread publicate_data:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/hoymiles.py", line 341, in run
    self.execute(*self.args, **self.kwargs)
  File "/hoymiles.py", line 263, in publicate_data
    hoymiles_h.get_solar_data()
  File "/hoymilesapi.py", line 248, in get_solar_data
    status, self.solar_data = self.request_solar_data()
  File "/hoymilesapi.py", line 364, in request_solar_data
    if "status" in solar.keys():
```