dmslabsbr/hoymiles #89: 502: Bad Gateway

GileanEU, 2022-09-22:
I cannot get this integration to work. I have entered my Hoymiles user id and password, and plant id (I guess that is plant name?), but when I start the addon and try to go to web UI, I just get "502: Bad Gateway" to upper left corner.

Also this addon seems to close itself often and then I need to start it again. 

I got these errors on log:

Failed to to call /addons/44349f66_hoymiles_solardata/stats - Container addon_44349f66_hoymiles_solardata is not running
Failed to to call /addons/44349f66_hoymiles_solardata/stats - Can't read stats from addon_44349f66_hoymiles_solardata: Expecting value: line 1 column 1 (char 0)

Cosik, 2022-09-22:
Hi, you don't provided enough information to help you. About gateway please send #87

GileanEU, 2022-09-22:
Sorry, I am not sure what information I should provide? I am using HA addon version 1.0.7.
[image]
[image]

GileanEU, 2022-09-22:
Update: I now realized that logging in to Hoymiles S-Miles web app does not show that plant id. But mobile app shows that, so plant id is NOT the same thing as plant name. I corrected that, and I got new MQTT device, but this does not show but two sensors:
[image]