dmslabsbr/hoymiles #90: unable to start add-on on hassio : TypeError: strptime() -- tried all 3 versions.

srspereira, 2022-10-13:
hi.. I'm keeping getting this error message and add-on shutdowns. 
looks like add-on isn't properly hadling  zero value  or DTU is offline 
thanks.

[ **that's expected because DTU could be off line**
2022-10-13 11:46:24,466 - HoymilesAdd-on.hoymilesapi.Hoymiles - WARNING - real_power = 0   
Traceback (most recent call last):
  File "/hoymiles.py", line 385, in <module>
    sys.exit(main())  # next section explains the use of sys.exit
  File "/hoymiles.py", line 350, in main
    publicate_data(plant_list[id], mqtt_list[id])
  File "/hoymiles.py", line 203, in publicate_data
    hoymiles_h.get_solar_data()
  File "/hoymilesapi.py", line 237, in get_solar_data
    self.solar_data = self.adjust_solar_data(self.solar_data)
  File "/hoymilesapi.py", line 278, in adjust_solar_data
    last_data_time = datetime.strptime(last_data_time, '%Y-%m-%d %H:%M:%S')
TypeError: strptime() argument 1 must be str, not None
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

--- additional info regarding my HASSIO

{"supervisor":"2022.10.0","homeassistant":"2022.10.3","hassos":"9.2","docker":"20.10.17","hostname":"homeassistant","operating_system":"Home Assistant OS 9.2","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","haos","resolved"],"machine":"qemux86-64","arch":"amd64","state":"running","supported_arch":["amd64","i386"],"supported":true,"channel":"stable","logging":"info","timezone":"America/Sao_Paulo"}

srspereira, 2022-10-13:
DEVELOPERS MODE ON — STABLE VERSION 1.0.7

[hoymiles.log](https://github.com/dmslabsbr/hoymiles/files/9778415/hoymiles.log)

Cosik, 2022-10-13:
As usually please provide as many information as is possible. You didn't mention if it is the first time you run it or it was running previously. Is you dtu working and installation ready up and running on HM Web page, etc.

srspereira, 2022-10-13:
Yes 1st time running this add-on.
DTU is running, all data accessible from HM web page

as you can see in log .. add-on was able to get all data. but for some
reason couldn't handle :

last_data_time = datetime.strptime(last_data_time, '%Y-%m-%d %H:%M:%S')

TypeError: strptime() argument 1 must be str, not *None*

which means last_data_time was not initiated with valid date.

BTW what's the time issue in my installation?

Cosik, 2022-10-13:
Plugin is working as expected. Problem is with your setup, please look at this information:
`2022-10-13 12:39:36,680 - HoymilesAdd-on.hoymilesapi.Hoymiles - WARNING - real_power = 0`

Also request is returning nothing, empty and zero values:
`2022-10-13 12:39:36,680 - HoymilesAdd-on.hoymilesapi.Hoymiles - DEBUG - solar_data: {'today_eq': '0.0', 'month_eq': '0', 'year_eq': '0', 'total_eq': '0', 'real_power': '0', 'co2_emission_reduction': '0', 'plant_tree': '0', 'data_time': None, 'last_data_time': None, 'capacitor': '14.4', 'is_balance': 0, 'is_reflux': 0, 'reflux_station_data': None}`

From log I can see that no of your device is connected and never connected due to info from REST API.

When you create installation? Any proof of working farm? Print screen from HM cloud page? Maybe your DTU has no internet access and you are checking local page? I know that DTU-W100 support local page also.

srspereira, 2022-10-13:
thanks for your help. I'll check on farm premisses

Cosik, 2022-10-13:
Did you check it? Analysis was based only on my knowledge. Please let us know if you solved it and how.

srspereira, 2022-10-18:
> Did you check it? Analysis was based only on my knowledge. Please let us know if you solved it and how.

Hi cosik.. DTU was connected to internet BUT for some reason microinverters didn't send data, therefore db was empty. after installer issue a command for DTU reconnect to microinverters, db was initiated.