dmslabsbr/hoymiles #91: Energy dashboard value HA

Floris039, 2022-10-24:
First of all thanks for this addon. Which entity can I use for the solar panel yield in the energy dashboard? I can't find the right one because the values are very different every time I try a value.

Cosik, 2022-10-24:
Hi, I used _today_eq_ and for grid in my case daily _consumption_ and _production_ to grid, both are utility meter entities.

Floris039, 2022-10-24:
Thank you. Maybe the data isn't handled correctly because it's already in the afternoon here. So the sun is shining for some time now and the data is only added an hour ago or so.  For grid I also have a different device :)

Cosik, 2022-10-24:
On Dashboard data is refreshed every hour for statistics

Cosik, 2022-10-27:
@Floris039 could we close it?

closed 2022-10-27 by Floris039