dmslabsbr/hoymiles #99: 502: bad gateway

ricardomeleiro, 2023-02-05:
Hello.
I'm really interested to use this add-on on my home assistant. I already passed through the MQTT and also Hoymiles add-on configurations but I am just receiving the ''502: bad gateway'' error message in the Web UI. Someone could please help me to try to solve this problem?

Following log output:
```
2023-02-05 14:56:26,303 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway  v.1.0.13
2023-02-05 14:56:26,304 - HoymilesAdd-on - INFO - Starting up... 2023-02-05 14:56:26
2023-02-05 14:56:26,304 - HoymilesAdd-on - INFO - Using Internal MQTT Server: core-mosquitto
2023-02-05 14:56:26,304 - HoymilesAdd-on - INFO - Using Internal MQTT User: addons
2023-02-05 14:56:26,305 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2023-02-05 14:56:26,962 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - I got the token!!  :-)
2023-02-05 14:56:26,963 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_find
2023-02-05 14:56:27,508 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all
2023-02-05 14:56:28,047 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-02-05 14:56:28,508 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-02-05 14:56:28,960 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-02-05 14:56:29,418 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Starting MQTT core-mosquitto
2023-02-05 14:56:29,422 - HoymilesAdd-on.mqttapi.Mqtt - INFO - MQTT connected with result code 0
2023-02-05 14:56:29,423 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Connected to core-mosquitto
2023-02-05 14:56:30,448 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all
2023-02-05 14:56:30,982 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2023-02-05 14:56:31,439 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - last_data_time 2023-02-05T14:48:39-03:00
2023-02-05 14:56:31,439 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-02-05 14:56:31,892 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-02-05 14:56:32,346 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-02-05 14:56:32,803 - HoymilesAdd-on - INFO - Solar data publication...2023-02-05 14:56:32.803411
2023-02-05 14:56:32,804 - HoymilesAdd-on - INFO - DTU-Lite-S_516340 data publication...2023-02-05 14:56:32.804144
2023-02-05 14:56:32,810 - HoymilesAdd-on - INFO - HMS-1800-4T_2360700 data publication...2023-02-05 14:56:32.810859
2023-02-05 14:56:32,811 - HoymilesAdd-on - INFO - HMS-1800-4T_2360690 data publication...2023-02-05 14:56:32.811476
2023-02-05 14:56:32,811 - HoymilesAdd-on - INFO - HMS-1800-4T_2360710 data publication...2023-02-05 14:56:32.811945
2023-02-05 14:56:32,812 - HoymilesAdd-on - INFO - Main loop start!
```

As you can see the Add-on got the info of my DTU:
> 2023-02-05 14:56:32,804 - HoymilesAdd-on - INFO - DTU-Lite-S_516340 data publication...2023-02-05 14:56:32.804144

As also my micro inverters units:

> 2023-02-05 14:56:32,810 - HoymilesAdd-on - INFO - HMS-1800-4T_2360700 data publication...2023-02-05 14:56:32.810859
> 2023-02-05 14:56:32,811 - HoymilesAdd-on - INFO - HMS-1800-4T_2360690 data publication...2023-02-05 14:56:32.811476
> 2023-02-05 14:56:32,811 - HoymilesAdd-on - INFO - HMS-1800-4T_2360710 data publication...2023-02-05 14:56:32.811945

Thanks for your help!

Cosik, 2023-02-05:
Web ui is not implemented

ricardomeleiro, 2023-02-05:
Thanks for your feedback, @Cosik !

testustestmann, 2023-02-28:
Hello, i have the same problem:

log output:

dmslabs - Home Assistant HoyMiles Solar Data Gateway Add-on
2023-02-28 13:23:59,779 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway  v.1.0.7
2023-02-28 13:23:59,781 - HoymilesAdd-on - INFO - Starting up... 2023-02-28 13:23:59
2023-02-28 13:23:59,783 - HoymilesAdd-on - INFO - Using Internal MQTT Server: core-mosquitto
2023-02-28 13:23:59,783 - HoymilesAdd-on - INFO - Using Internal MQTT User: addons
2023-02-28 13:23:59,784 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2023-02-28 13:24:02,676 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - I got the token!!  :-)
2023-02-28 13:24:02,678 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_find
2023-02-28 13:24:03,162 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all
2023-02-28 13:24:03,581 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-02-28 13:24:03,954 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Starting MQTT core-mosquitto
2023-02-28 13:24:04,168 - HoymilesAdd-on.mqttapi.Mqtt - INFO - MQTT connected with result code 0
2023-02-28 13:24:04,168 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Connected to core-mosquitto
2023-02-28 13:24:06,345 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all
2023-02-28 13:24:06,809 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2023-02-28 13:24:07,213 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - last_data_time 2023-02-28T13:17:50+01:00
2023-02-28 13:24:07,214 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-02-28 13:24:07,578 - HoymilesAdd-on - INFO - Solar data publication...2023-02-28 13:24:07.578090
2023-02-28 13:24:07,586 - HoymilesAdd-on - INFO - DTU-Pro_792470 data publication...2023-02-28 13:24:07.586037
2023-02-28 13:24:07,592 - HoymilesAdd-on - INFO - HM-1500_3627800 data publication...2023-02-28 13:24:07.592671
2023-02-28 13:24:07,595 - HoymilesAdd-on - INFO - Main loop start!
2023-02-28 13:32:07,600 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all
2023-02-28 13:32:08,316 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2023-02-28 13:32:08,719 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - last_data_time 2023-02-28T13:17:50+01:00
2023-02-28 13:32:08,721 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-02-28 13:32:09,128 - HoymilesAdd-on - INFO - Solar data publication...2023-02-28 13:32:09.128082
2023-02-28 13:32:09,135 - HoymilesAdd-on - INFO - DTU-Pro_792470 data publication...2023-02-28 13:32:09.135379
2023-02-28 13:32:09,137 - HoymilesAdd-on - INFO - HM-1500_3627800 data publication...2023-02-28 13:32:09.137055
2023-02-28 13:40:09,141 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all
2023-02-28 13:40:09,832 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2023-02-28 13:40:10,241 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - last_data_time 2023-02-28T13:32:24+01:00
2023-02-28 13:40:10,243 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-02-28 13:40:10,636 - HoymilesAdd-on - INFO - Solar data publication...2023-02-28 13:40:10.636025
2023-02-28 13:40:10,639 - HoymilesAdd-on - INFO - DTU-Pro_792470 data publication...2023-02-28 13:40:10.639492
2023-02-28 13:40:10,643 - HoymilesAdd-on - INFO - HM-1500_3627800 data publication...2023-02-28 13:40:10.643425
2023-02-28 13:48:10,645 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all
2023-02-28 13:48:11,634 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2023-02-28 13:48:11,987 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - last_data_time 2023-02-28T13:47:52+01:00
2023-02-28 13:48:11,988 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-02-28 13:48:12,381 - HoymilesAdd-on - INFO - Solar data publication...2023-02-28 13:48:12.381133
2023-02-28 13:48:12,387 - HoymilesAdd-on - INFO - DTU-Pro_792470 data publication...2023-02-28 13:48:12.387338
2023-02-28 13:48:12,390 - HoymilesAdd-on - INFO - HM-1500_3627800 data publication...2023-02-28 13:48:12.390081

ricardomeleiro, 2023-02-28:
Hi @testustestmann, as @Cosik said the web ui is not implemented.  As I could check in your shared log the Add-on is communicating with your DTU (DTU-Pro_792470 d). So for now you just need to click on ''add a card'' > ''by entity'' and you should see the entity ''sensor.solar...''.
Hope it helps.

closed 2023-03-20 by Cosik