dmslabsbr/hoymiles #101: 502: Bad Gateway

tyllux, 2023-03-27:
Hi,

I'm constantly running into a 502 bad gateway error and I'm not able to resolve this issue.

(You can find the details to my Home Assistant configuration at 10:23:22 in the log at the end.)

Things I tried:

- Test with the stable-version
- Test with the edge-version
- Test with internal and external MQTT-Server
- Login to Home Assistant via http and https
- Several reboots and re-installations
- While checking the log, everything seems fine (connection to global.hoymiles.com is established) - except the INFO that the "secrets.ini" is missing

-> a 502 is often a password / auth issue - so I'm guessing the root of the problem lies in the missing "secrets.ini"
-> maybe there is a simple solution, but I need a hint ;-)

Excerpt of the log:

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Starting add-on container...
Mon Mar 27 10:23:22 CEST 2023
total 24
drwxr-xr-x    5 root     root          4096 Mar 27 10:23 .
drwxr-xr-x    1 root     root          4096 Mar 27 10:23 ..
drwxr-xr-x    2 root     root          4096 Mar 27 00:33 __pycache__
drwxr-xr-x    2 root     root          4096 Mar 27 10:21 jsons
-rw-------    1 root     root           575 Mar 27 10:23 options.json
drwxr-xr-x    2 root     root          4096 Mar 27 00:33 templates
[10:23:22] INFO: /data/secrets.ini not exists!
Getting mqtt data...
[10:23:22] INFO: null
[10:23:22] INFO: {"supervisor":"2023.03.2","homeassistant":"2023.3.6","hassos":"9.5","docker":"20.10.22","hostname":"homeassistant","operating_system":"Home Assistant OS 9.5","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","haos","resolved","journal","disk"],"machine":"generic-x86-64","arch":"amd64","state":"running","supported_arch":["amd64","i386"],"supported":true,"channel":"stable","logging":"info","timezone":"Europe/Berlin"}
[10:23:22] INFO: {"supervisor":"2023.03.2","homeassistant":"2023.3.6","hassos":"9.5","docker":"20.10.22","hostname":"homeassistant","operating_system":"Home Assistant OS 9.5","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","haos","resolved","journal","disk"],"machine":"generic-x86-64","arch":"amd64","state":"running","supported_arch":["amd64","i386"],"supported":true,"channel":"stable","logging":"info","timezone":"Europe/Berlin"}
dmslabs - Home Assistant HoyMiles Solar Data Gateway Add-on
2023-03-27 10:23:23,044 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway  v.1.0.13
2023-03-27 10:23:23,045 - HoymilesAdd-on - INFO - Starting up... 2023-03-27 10:23:23
2023-03-27 10:23:23,045 - HoymilesAdd-on - INFO - Using Internal MQTT Server: core-mosquitto
2023-03-27 10:23:23,045 - HoymilesAdd-on - INFO - Using Internal MQTT User: addons
2023-03-27 10:23:23,046 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2023-03-27 10:23:23,892 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - I got the token!!  :-)
2023-03-27 10:23:23,892 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_find
2023-03-27 10:23:24,424 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all
2023-03-27 10:23:24,900 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-03-27 10:23:25,487 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Starting MQTT core-mosquitto
2023-03-27 10:23:25,489 - HoymilesAdd-on.mqttapi.Mqtt - INFO - MQTT connected with result code 0
2023-03-27 10:23:25,489 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Connected to core-mosquitto
2023-03-27 10:23:26,521 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all
2023-03-27 10:23:26,913 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2023-03-27 10:23:27,244 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - last_data_time 2023-03-27T10:17:11+02:00
2023-03-27 10:23:27,245 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-03-27 10:23:27,590 - HoymilesAdd-on - INFO - Solar data publication...2023-03-27 10:23:27.590508
2023-03-27 10:23:27,591 - HoymilesAdd-on - INFO - DTU-Pro_123456 data publication...2023-03-27 10:23:27.591415
2023-03-27 10:23:27,592 - HoymilesAdd-on - INFO - HM-1500_1234567 data publication...2023-03-27 10:23:27.592011
2023-03-27 10:23:27,593 - HoymilesAdd-on - INFO - Main loop start!

Cosik, 2023-03-27:
#99 have you seen it?

tyllux, 2023-03-27:
Da**, I really missed that ...

Thanks a lot!

closed 2023-03-27 by tyllux