dmslabsbr/hoymiles #102: Data updates freeze on server 504

maakuth, 2023-04-03:
After running for a an hour or two, the edge version built from commit 044187 gets stuck due to server returning 504 Gateway timeout.
This has happened multiple times and it recovers correctly with `docker restart`.
I updated from some 2021 build just a few days ago, so I don't have long term idea if this is a persistent problem.
Even if it is rare, I would think getting a 504 should not stop the process completely, but perhaps retry after a minute or so.
I've just now restarted the thing, I'll report back here if it crashes again.

```
2023-04-02 13:34:28,748 - HoymilesAdd-on - INFO - Solar data publication...2023-04-02 13:34:28.748734
2023-04-02 13:34:28,749 - HoymilesAdd-on - INFO - DTU-W100_xxxxx data publication...2023-04-02 13:34:28.749192
2023-04-02 13:34:28,749 - HoymilesAdd-on - INFO - HM-1500_xxxxx data publication...2023-04-02 13:34:28.749736
2023-04-02 13:34:28,750 - HoymilesAdd-on - INFO - HM-1500_xxxxx data publication...2023-04-02 13:34:28.750023
2023-04-02 13:34:28,750 - HoymilesAdd-on - INFO - HM-1500_xxxxx data publication...2023-04-02 13:34:28.750378
2023-04-02 13:34:28,750 - HoymilesAdd-on - INFO - HM-1500_xxxxx data publication...2023-04-02 13:34:28.750685
2023-04-02 13:34:28,751 - HoymilesAdd-on - INFO - HM-1500_xxxxx data publication...2023-04-02 13:34:28.751050
2023-04-02 13:34:28,751 - HoymilesAdd-on - INFO - HM-1500_xxxxx data publication...2023-04-02 13:34:28.751421
2023-04-02 13:34:28,751 - HoymilesAdd-on - INFO - HM-1500_xxxxx data publication...2023-04-02 13:34:28.751810
2023-04-02 13:42:28,752 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all
2023-04-02 13:43:29,014 - HoymilesAdd-on.hoymilesapi.Hoymiles - ERROR - Access error: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all
2023-04-02 13:43:29,014 - HoymilesAdd-on.hoymilesapi.Hoymiles - ERROR - Status code: 504Gateway Timeout

maakuth, 2023-04-04:
Now a similar issue happened again, though with different error code. 
However, restarting the container fixes it.
I think adding an exception handler would help.

```
2023-04-04 08:13:58,573 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all
2023-04-04 08:13:59,540 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2023-04-04 08:14:00,284 - HoymilesAdd-on.hoymilesapi.Hoymiles - ERROR - Access error: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2023-04-04 08:14:00,284 - HoymilesAdd-on.hoymilesapi.Hoymiles - ERROR - Status code: 50310

juslex, 2023-04-04:
Same issue here

nineteen0815, 2023-04-04:
Same issue here

nineteen0815, 2023-04-04:
ERROR
The requested URL could not be retrieved

While trying to retrieve the URL:
https://global.hoymiles.com/platform/station/view
The following error was encountered:
Connection Failed
The system returned:

    [No Error]

The remote host or network may be down. Please try the request again.

Confused Website avaible/not avaible every hour in Browser
Hoymiles App works stable

Marcski, 2023-04-06:
Apparently the same issue i am having, different error code though (5031000, AttributeError: 'str' object has no attribute 'keys'*)

Are there any updates?

Cosik, 2023-04-07:
Please test #104 before merge

maakuth, 2023-04-07:
@Cosik I'm now running your branch, at least it started fine. I will report back if I see it crash again.

Cosik, 2023-04-08:
I forget to mention that change is on edge

maakuth, 2023-04-08:
No problem, I was on edge already and had a look at your patch beforehand.

maakuth, 2023-04-09:
Unfortunately the issue seems to persist. The last lines are very similar as to before.
```
2023-04-09 02:27:35,305 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all
2023-04-09 02:27:45,544 - HoymilesAdd-on.hoymilesapi.Hoymiles - ERROR - Access error: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all
2023-04-09 02:27:45,544 - HoymilesAdd-on.hoymilesapi.Hoymiles - ERROR - Status code: 5031000

Cosik, 2023-04-09:
@maakuth but wait, I didn't hide errors, I just fix code to not crash when they appear. Is it working correctly now when error appear?

maakuth, 2023-04-09:
It was the old build (edge from the master branch) I was running unfortunately, so no observations of your fixes reaction to the error as of yet. Now waiting for it to happen with your fix.

maakuth, 2023-04-10:
Hmm, at least it recovered from this error, though I don't know if it's quite the same as the issue we're talking about.
Perhaps it takes a moment to load stuff in the backend right after login?

```
2023-04-10 01:39:36,928 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all
2023-04-10 01:39:37,786 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2023-04-10 01:39:38,750 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - I got the token!!  :-)
2023-04-10 01:39:38,750 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2023-04-10 01:39:39,480 - HoymilesAdd-on.hoymilesapi.Hoymiles - ERROR - request_plant_hw dtu string indices must be integers
2023-04-10 01:39:39,480 - HoymilesAdd-on.hoymilesapi.Hoymiles - WARNING - Faild updating string indices must be integers
2023-04-10 01:39:39,480 - HoymilesAdd-on.hoymilesapi.Hoymiles - WARNING - Dump data for analysis
2023-04-10 01:39:39,480 - HoymilesAdd-on.hoymilesapi.Hoymiles - WARNING - {'today_eq': '0.0', 'month_eq': '471904', 'year_eq': '723801', 'total_eq': '15310900', 'real_power': '0', 'co2_emission_reduction': '15264967.3', 'plant_tree': '834', 'data_time': '2023-04-09 21:19:59', 'last_data_time': '2023-04-09 21:19:59', 'capacitor': '9.9', 'is_balance': 0, 'is_reflux': 0, 'reflux_station_data': None}
--- Last three lines repeating 13 times almost the same, except timestamps, then resuming normally:
2023-04-10 01:39:39,484 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2023-04-10 01:39:39,949 - HoymilesAdd-on.hoymilesapi.Hoymiles - WARNING - real_power = 0
2023-04-10 01:39:39,949 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - last_data_time 2023-04-09T21:19:59+00:00
2023-04-10 01:39:39,949 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-04-10 01:39:40,466 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-04-10 01:39:40,983 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-04-10 01:39:41,572 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-04-10 01:39:42,092 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-04-10 01:39:42,605 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-04-10 01:39:43,211 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-04-10 01:39:43,727 - HoymilesAdd-on - INFO - Solar data publication...2023-04-10 01:39:43.727015
```

Cosik, 2023-04-10:
Yes it could be. The main problem is that hoymiles don't provide API spec, all work based on reverse engineering. This could introduce some trouble.

closed 2023-04-11 by maakuth