dmslabsbr/hoymiles #103: Addon keeps crashing

Marcski, 2023-04-06:
2023-04-06 03:09:21,529 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all
2023-04-06 03:09:22,788 - HoymilesAdd-on.hoymilesapi.Hoymiles - ERROR - Access error: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all
2023-04-06 03:09:22,789 - HoymilesAdd-on.hoymilesapi.Hoymiles - ERROR - Status code: 5031000
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/hoymiles.py", line 307, in run
    self.execute(*self.args, **self.kwargs)
  File "/hoymiles.py", line 236, in publicate_data
    hoymiles_h.update_devices_status()
  File "/hoymilesapi.py", line 327, in update_devices_status
    status, hws_data = self.request_plant_hw()
  File "/hoymilesapi.py", line 367, in request_plant_hw
    if 'status' in retv.keys():
AttributeError: 'str' object has no attribute 'keys'****

Cosik, 2023-04-06:
#102

Kwiatjk, 2023-04-19:
Same 
https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 703, in urlopen
    httplib_response = self._make_request(
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 449, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 444, in _make_request
    httplib_response = conn.getresponse()
  File "/usr/local/lib/python3.9/http/client.py", line 1377, in getresponse
    response.begin()
  File "/usr/local/lib/python3.9/http/client.py", line 320, in begin
    version, status, reason = self._read_status()
  File "/usr/local/lib/python3.9/http/client.py", line 281, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "/usr/local/lib/python3.9/socket.py", line 704, in readinto
    return self._sock.recv_into(b)
  File "/usr/local/lib/python3.9/ssl.py", line 1241, in recv_into
    return self.read(nbytes, buffer)
  File "/usr/local/lib/python3.9/ssl.py", line 1099, in read
    return self._sslobj.read(len, buffer)
OSError: [Errno 22] Invalid argument

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/requests/adapters.py", line 489, in send
    resp = conn.urlopen(
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 787, in urlopen
    retries = retries.increment(
  File "/usr/local/lib/python3.9/site-packages/urllib3/util/retry.py", line 550, in increment
    raise six.reraise(type(error), error, _stacktrace)
  File "/usr/local/lib/python3.9/site-packages/urllib3/packages/six.py", line 769, in reraise
    raise value.with_traceback(tb)
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 703, in urlopen
    httplib_response = self._make_request(
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 449, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 444, in _make_request
    httplib_response = conn.getresponse()
  File "/usr/local/lib/python3.9/http/client.py", line 1377, in getresponse
    response.begin()
  File "/usr/local/lib/python3.9/http/client.py", line 320, in begin
    version, status, reason = self._read_status()
  File "/usr/local/lib/python3.9/http/client.py", line 281, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "/usr/local/lib/python3.9/socket.py", line 704, in readinto
    return self._sock.recv_into(b)
  File "/usr/local/lib/python3.9/ssl.py", line 1241, in recv_into
    return self.read(nbytes, buffer)
  File "/usr/local/lib/python3.9/ssl.py", line 1099, in read
    return self._sslobj.read(len, buffer)
urllib3.exceptions.ProtocolError: ('Connection aborted.', OSError(22, 'Invalid argument'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/hoymiles.py", line 274, in run
    self.execute(*self.args, **self.kwargs)
  File "/hoymiles.py", line 204, in publicate_data
    hoymiles_h.get_alarms()
  File "/hoymilesapi.py", line 402, in get_alarms
    retv = self.send_payload(DATA_FIND_DETAILS, header, payload)
  File "/hoymilesapi.py", line 352, in send_payload
    retv, s_code = self.send_request(
  File "/hoymilesapi.py", line 180, in send_request
    response = sess.send(prepped)
  File "/usr/local/lib/python3.9/site-packages/requests/sessions.py", line 701, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/requests/adapters.py", line 547, in send
    raise ConnectionError(err, request=request)
requests.exceptions.ConnectionError: ('Connection aborted.', OSError(22, 'Invalid argument'))

GileanEU, 2023-05-22:
I have experienced this every now and then earlier, but for a couple of days, maybe a week now, this happens now many times a day. Basically I notice it when apparently there is no solar output on Energy panel (although there actually is), I just stop the addon and then start it again. So, I get a big spike on next hour Energy panel update - and some times after that this plugin dies again. Watchdog does not help because I think it sees that addon is still running. This is what I see on logs.

2023-05-22 07:23:54,078 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all
2023-05-22 07:23:54,834 - HoymilesAdd-on.hoymilesapi.Hoymiles - ERROR - Access error: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all
2023-05-22 07:23:54,835 - HoymilesAdd-on.hoymilesapi.Hoymiles - ERROR - Status code: 5031000
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/hoymiles.py", line 274, in run
    self.execute(*self.args, **self.kwargs)
  File "/hoymiles.py", line 202, in publicate_data
    hoymiles_h.update_devices_status()
  File "/hoymilesapi.py", line 299, in update_devices_status
    status, hws_data = self.request_plant_hw()
  File "/hoymilesapi.py", line 337, in request_plant_hw
    retv = self.send_payload(GET_ALL_DEVICE_API, header, payload)
  File "/hoymilesapi.py", line 354, in send_payload
    retv = json.loads(retv)
  File "/usr/local/lib/python3.9/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "/usr/local/lib/python3.9/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/local/lib/python3.9/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

Floris039, 2023-05-22:
I also have this issue. Same error code. Happening a few times a day at this moment.

sirteemu, 2023-05-23:
Same problems here, for almost two weeks the addon has crashed several times a day, requires a manual restart for the add-on. I tried removing the add-on, trying different versions and the problem hasn't gone away.

Cosik, 2023-05-24:
Hi I will try too look at this today, but it looks like json parser problem or request don't provide correct response

Cosik, 2023-05-26:
Please check if #109 fix problems

rpanachi, 2023-08-03:
I had the same issue here.
Addon version: 1.0.8
Home Assistant: 2023.7.3

```
2023-08-03 10:25:17,136 - HoymilesAdd-on.hoymilesapi.Hoymiles - ERROR - Status code: 5991000
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/hoymiles.py", line 274, in run
    self.execute(*self.args, **self.kwargs)
  File "/hoymiles.py", line 204, in publicate_data
    hoymiles_h.get_alarms()
  File "/hoymilesapi.py", line 402, in get_alarms
    retv = self.send_payload(DATA_FIND_DETAILS, header, payload)
  File "/hoymilesapi.py", line 354, in send_payload
    retv = json.loads(retv)
  File "/usr/local/lib/python3.9/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "/usr/local/lib/python3.9/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/local/lib/python3.9/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
```

Because that, the sensor become unavailable and I need to restart the addon to work again. Any solution for this?

closed 2023-10-28 by Cosik