dmslabsbr/hoymiles #105: Hassio beginner ... i do not get entities for the addon...

chrisloff, 2023-04-08:
I have installed the addon.
Configured everything - the log shows no issues
Why do i not see any entity for the hoymiles addon?

Here my log:
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Starting add-on container...
Sat Apr  8 10:47:52 CEST 2023
total 24
drwxr-xr-x    5 root     root          4096 Apr  8 10:47 .
drwxr-xr-x    1 root     root          4096 Apr  8 10:47 ..
drwxr-xr-x    2 root     root          4096 Apr  8 07:26 __pycache__
drwxr-xr-x    2 root     root          4096 Apr  8 10:18 jsons
-rw-------    1 root     root           304 Apr  8 10:47 options.json
drwxr-xr-x    2 root     root          4096 Apr  8 07:26 templates
[10:47:52] INFO: /data/secrets.ini not exists!
Getting mqtt data...
[10:47:55] INFO: null
[10:47:56] INFO: {"supervisor":"2023.04.0","homeassistant":"2023.4.0","hassos":"9.5","docker":"20.10.22","hostname":"homeassistant","operating_system":"Home Assistant OS 9.5","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","haos","resolved","journal","disk"],"machine":"raspberrypi3-64","arch":"aarch64","state":"running","supported_arch":["aarch64","armv7","armhf"],"supported":true,"channel":"stable","logging":"info","timezone":"Europe/Berlin"}
[10:47:56] INFO: {"supervisor":"2023.04.0","homeassistant":"2023.4.0","hassos":"9.5","docker":"20.10.22","hostname":"homeassistant","operating_system":"Home Assistant OS 9.5","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","haos","resolved","journal","disk"],"machine":"raspberrypi3-64","arch":"aarch64","state":"running","supported_arch":["aarch64","armv7","armhf"],"supported":true,"channel":"stable","logging":"info","timezone":"Europe/Berlin"}
dmslabs - Home Assistant HoyMiles Solar Data Gateway Add-on
2023-04-08 10:47:58,008 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway  v.1.0.7
2023-04-08 10:47:58,008 - HoymilesAdd-on - INFO - Starting up... 2023-04-08 10:47:58
2023-04-08 10:47:58,011 - HoymilesAdd-on - INFO - Using Internal MQTT Server: core-mosquitto
2023-04-08 10:47:58,011 - HoymilesAdd-on - INFO - Using Internal MQTT User: addons
2023-04-08 10:47:58,013 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2023-04-08 10:48:01,208 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - I got the token!!  :-)
2023-04-08 10:48:01,209 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_find
2023-04-08 10:48:02,231 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all
2023-04-08 10:48:02,771 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-04-08 10:48:03,340 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Starting MQTT core-mosquitto
2023-04-08 10:48:04,426 - HoymilesAdd-on.mqttapi.Mqtt - INFO - MQTT connected with result code 0
2023-04-08 10:48:04,427 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Connected to core-mosquitto
2023-04-08 10:48:05,399 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all
2023-04-08 10:48:05,925 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2023-04-08 10:48:06,423 - HoymilesAdd-on.hoymilesapi.Hoymiles - WARNING - real_power = 0
2023-04-08 10:48:06,438 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - last_data_time 2023-04-08T09:35:05+02:00
2023-04-08 10:48:06,439 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-04-08 10:48:06,926 - HoymilesAdd-on - INFO - Solar data publication...2023-04-08 10:48:06.926447
2023-04-08 10:48:06,929 - HoymilesAdd-on - INFO - DTU-WLite_114876 data publication...2023-04-08 10:48:06.929053
2023-04-08 10:48:06,933 - HoymilesAdd-on - INFO - HM-1500_594396 data publication...2023-04-08 10:48:06.933118
2023-04-08 10:48:06,939 - HoymilesAdd-on - INFO - Main loop start!

Cosik, 2023-04-08:
I don't know why. Did you check if your HA is receiving mqtt messages?

dtimes, 2023-04-10:
I guess it´s caused by only installing the mosquitto broker addon  and not adding the core mosquitto integration to HA ( you need to complete both steps to get the entities) , had this issue some time ago and missed the step

Cosik, 2023-05-02:
no response from @chrisloff

closed 2023-05-02 by Cosik