dmslabsbr/hoymiles #107: Version 1.0.15: Crash on Start: mkdir: can't create directory '/data/templates': Permission denied

faken, 2023-05-04:
Latest available version of the AddOn is not working anymore getting the following log output: 
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Starting add-on container...
Thu May  4 08:50:01 CEST 2023
total 12
drwxr-xr-x    2 root     root          4096 May  4 08:50 .
drwxr-xr-x    1 root     root          4096 May  4 08:50 ..
-rw-------    1 root     root           372 May  4 08:50 options.json
mkdir: can't create directory '/data/templates': Permission denied
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

Cosik, 2023-05-04:
@maakuth I was thinking that it is stable.

maakuth, 2023-05-04:
Oh, sorry about that. It works fine in my setup. Perhaps Home Assistant OS runs it somehow differently than myself. Unfortunately I don't have time to setup a test system for that this week.

faken, 2023-05-04:
@Cosik Can you maybe publish a new version 1.0.16? 
My Hass.io setup still wants to install 1.0.15

faken, 2023-05-04:
Never mind, I've removed the repository and added it again, now it wants to install 1.0.14

faken, 2023-05-04:
Just to confirm: 1.0.14 is working fine on my HASS.io-instance, thanks for the quick update!

Cosik, 2023-05-04:
Ok, I just released dummy version 1.0.16

closed 2023-05-04 by Cosik