dmslabsbr/hoymiles #110: Addon crashes

Floris039, 2023-06-07:
I get the following error in my logs:

`2023-06-07 15:28:10,825 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - last_data_time 2023-06-07T15:17:22+02:00
2023-06-07 15:28:10,825 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-06-07 15:28:11,732 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-06-07 15:28:12,069 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-06-07 15:28:12,427 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-06-07 15:28:12,800 - HoymilesAdd-on - INFO - Solar data publication...2023-06-07 15:28:12.800382
2023-06-07 15:28:12,800 - HoymilesAdd-on - INFO - DTU-Pro-S_620210 data publication...2023-06-07 15:28:12.800878
2023-06-07 15:28:12,801 - HoymilesAdd-on - INFO - HMS-2000-4T_2832020 data publication...2023-06-07 15:28:12.801180
2023-06-07 15:28:12,801 - HoymilesAdd-on - INFO - HMS-2000-4T_2832040 data publication...2023-06-07 15:28:12.801438
2023-06-07 15:28:12,801 - HoymilesAdd-on - INFO - HMS-2000-4T_2832050 data publication...2023-06-07 15:28:12.801894
2023-06-07 15:28:12,802 - HoymilesAdd-on - INFO - HMS-2000-4T_2832030 data publication...2023-06-07 15:28:12.802481
2023-06-07 15:36:12,803 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/urllib3/connection.py", line 174, in _new_conn
    conn = connection.create_connection(
  File "/usr/local/lib/python3.9/site-packages/urllib3/util/connection.py", line 72, in create_connection
    for res in socket.getaddrinfo(host, port, family, socket.SOCK_STREAM):
  File "/usr/local/lib/python3.9/socket.py", line 954, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
socket.gaierror: [Errno -3] Try again
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 703, in urlopen
    httplib_response = self._make_request(
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 386, in _make_request
    self._validate_conn(conn)
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 1042, in _validate_conn
    conn.connect()
  File "/usr/local/lib/python3.9/site-packages/urllib3/connection.py", line 358, in connect
    self.sock = conn = self._new_conn()
  File "/usr/local/lib/python3.9/site-packages/urllib3/connection.py", line 186, in _new_conn
    raise NewConnectionError(
urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPSConnection object at 0x7f342bf72c70>: Failed to establish a new connection: [Errno -3] Try again
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/requests/adapters.py", line 489, in send
    resp = conn.urlopen(
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 787, in urlopen
    retries = retries.increment(
  File "/usr/local/lib/python3.9/site-packages/urllib3/util/retry.py", line 592, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='global.hoymiles.com', port=443): Max retries exceeded with url: /platform/api/gateway/pvm/station_select_device_all (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f342bf72c70>: Failed to establish a new connection: [Errno -3] Try again'))
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/hoymiles.py", line 274, in run
    self.execute(*self.args, **self.kwargs)
  File "/hoymiles.py", line 202, in publicate_data
    hoymiles_h.update_devices_status()
  File "/hoymilesapi.py", line 299, in update_devices_status
    status, hws_data = self.request_plant_hw()
  File "/hoymilesapi.py", line 337, in request_plant_hw
    retv = self.send_payload(GET_ALL_DEVICE_API, header, payload)
  File "/hoymilesapi.py", line 352, in send_payload
    retv, s_code = self.send_request(
  File "/hoymilesapi.py", line 180, in send_request
    response = sess.send(prepped)
  File "/usr/local/lib/python3.9/site-packages/requests/sessions.py", line 701, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/requests/adapters.py", line 565, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='global.hoymiles.com', port=443): Max retries exceeded with url: /platform/api/gateway/pvm/station_select_device_all (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f342bf72c70>: Failed to establish a new connection: [Errno -3] Try again'))`

Cosik, 2023-06-07:
Missing info about setup.

Floris039, 2023-06-07:
I use Home assistant
HoyMiles Solar Gateway Addon
Current version: 1.0.7
Solar panel setup: DTU-PRO

Cosik, 2023-06-08:
Please check edge version.

closed 2023-10-28 by Cosik