dmslabsbr/hoymiles #111: is possible to add consumption from Grid

go2thanawat, 2023-06-11:
i see hoymile cloud is electric consumption from grid data  
is possible to add consumption from Grid to this add-on

Cosik, 2023-07-09:
Yes, it is possible, but I have no access to such installation.

mmz06, 2023-07-18:
Hi @Cosik 
Shall I provide you with a user account on my plant so you can access this maybe ?
I'm not really familiar with this, how can I provide you the credentials, as I can't find any way to send it to you as personal ?

Cosik, 2023-07-25:
@mmz06 yes, probably I will need access to your plant. When I will have some spare time I will back to you.

Wind06, 2023-10-24:
Hi @Cosik did you get  access to such configuration to get Grid data ?
If not I can give you access to my plant, I am interested by getting such information.
Thanks,

Wind06, 2023-10-25:
Hi @Cosik, yes I know but if I give you access to my plant would it be possible for you to get such data with the Addon ?

Cosik, 2023-10-25:
Should be, I need to check responses of requests and adapt code if possible.

Wind06, 2023-10-25:
Can I send you the access on a private mailbox 😉

Cosik, 2023-10-25:
Yes, please.

Wind06, 2023-10-25:
hmm not very familiar with GitHub, but cannot find your email address. How can I proceed?

Cosik, 2023-10-27:
@Wind06 maybe reach me over discord.

Wind06, 2023-10-27:
Hi Cosik, I think that you send me an invite under discord but I lost it, can you please resend it. Thanks and sorry for that.

ebiancarelli, 2023-10-27:
My system returns this block of data
`{
  "today_eq": "0.17",
  "month_eq": "131.1",
  "year_eq": "1805.07",
  "total_eq": "1805.07",
  "real_power": "277.1",
  "co2_emission_reduction": "1.79966",
  "plant_tree": "98",
  "data_time": "2023-10-27 08:15:58",
  "last_data_time": "2023-10-27T08:15:58-06:00",
  "capacitor": "2000.0",
  "is_balance": 0,
  "is_reflux": 1,
  "reflux_station_data": {
    "start_date": "",
    "end_date": "",
    "pv_power": "277.1",
    "grid_power": "-180",
    "load_power": "457.1",
    "bms_power": "0",
    "bms_soc": "0",
    "inv_num": 0,
    "meter_location": 2,
    "pv_to_load_eq": "0",
    "load_from_pv_eq": "0",
    "meter_b_in_eq": "5317230.832999999",
    "meter_b_out_eq": "0",
    "bms_in_eq": "0",
    "bms_out_eq": "0",
    "self_eq": "1805071",
    "pv_eq_total": "0",
    "use_eq_total": "0",
    "flows": [
      {
        "out": 2,
        "in": 1
      },
      {
        "out": 4,
        "in": 1
      }
    ],
    "icon_pv": 0,
    "icon_grid": 1,
    "icon_load": 1,
    "icon_bms": 1,
    "icon_gen": 0,
    "icon_pvi": 0,
    "mb_in_eq": {
      "today_eq": "0",
      "month_eq": "0",
      "year_eq": "0",
      "total_eq": "0"
    },
    "mb_out_eq": {
      "today_eq": "0",
      "month_eq": "0",
      "year_eq": "0",
      "total_eq": "0"
    }
  },
  "real_power_measurement": "277.1",
  "real_power_total_increasing": "277.1",
  "power_ratio": "13.9",
  "capacitor_kW": "2.0",
  "today_eq_Wh": "167.0"
}`

The grid info is contained in an enclosed json struct here with `grid_power` and `load_power` being interesting data points to graph.
`"reflux_station_data": {
    "start_date": "",
    "end_date": "",
    "pv_power": "277.1",
    "grid_power": "-180",
    "load_power": "457.1",
    "bms_power": "0",
    "bms_soc": "0",
    "inv_num": 0,
    "meter_location": 2,
    "pv_to_load_eq": "0",
    "load_from_pv_eq": "0",
    "meter_b_in_eq": "5317230.832999999",
    "meter_b_out_eq": "0",
    "bms_in_eq": "0",
    "bms_out_eq": "0",
    "self_eq": "1805071",
    "pv_eq_total": "0",
    "use_eq_total": "0",
    "flows": [
      {
        "out": 2,
        "in": 1
      },
      {
        "out": 4,
        "in": 1
      }
    ],
    "icon_pv": 0,
    "icon_grid": 1,
    "icon_load": 1,
    "icon_bms": 1,
    "icon_gen": 0,
    "icon_pvi": 0,
    "mb_in_eq": {
      "today_eq": "0",
      "month_eq": "0",
      "year_eq": "0",
      "total_eq": "0"
    },
    "mb_out_eq": {
      "today_eq": "0",
      "month_eq": "0",
      "year_eq": "0",
      "total_eq": "0"
    }
  },`

Cosik, 2023-10-27:
@ebiancarelli but I need requests also to know how to implement it.

ebiancarelli, 2023-10-27:
The block of JSON shown is gotten from MQTT using the 'stable' branch of code. If I guess, I think the json to dictionary is only taking the top level of tags, and not recursing into tags with additional elements.

Cosik, 2023-10-27:
But need more info, current implementation based on `pvm/station_select_device_of_tree` which defined how installation looks like. On this list has to be meter also. Then I could add it as other stuff.

ebiancarelli, 2023-10-27:
I sent you a friend request on Discord.

ebiancarelli, 2023-10-27:
Does this help?

```
2023-10-27 12:14:28,888 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_of_tree

2023-10-27 12:14:28,890 - HoymilesAdd-on.hoymilesapi.Hoymiles - DEBUG - {'Content-Type': 'application/json;charset=UTF-8', 'Cache-Control': 'no-cache', 'Host': 'global.hoymiles.com', 'Connection': 'keep-alive', 'Accept': 'application/json, text/plain, */*', 'Accept-Encoding': 'gzip, deflate, br', 'User-Agent': 'Mozilla/5.0 (Macintosh; Intel Mac OS X 11_2_3) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/89.0.4389.90 Safari/537.36', 'Accept-Language': 'pt-BR,pt;q=0.9,it-IT;q=0.8,it;q=0.7,es-ES;q=0.6,es;q=0.5,en-US;q=0.4,en;q=0.3', 'Cookie': "'uid=fff9c382-389f-4a47-8dc9-c5486fc3d9f5; hm_token=1.wPHrxCR7XZ8DK7p2Lx7asgCp200MhDHPv9iBs2irrgL69YYUMj9url4yZ47MDvw9vwSuntubZ4JZi5CK3IkjVf; Path=/; Domain=.global.hoymiles.com;Expires=Sat, 30 Mar 2024 22:11:48 GMT;'", 'Content-Length': '69'}

2023-10-27 12:14:30,313 - HoymilesAdd-on.hoymilesapi.Hoymiles - DEBUG - content: b'{"status":"0","message":"success","data":[{"id":980570,"sn":"10FD81844117","vc":"","dtu_sn":"10FD81844117","type":1,"version":3,"replace_num":0,"model_no":"DTU-Pro-S","soft_ver":"V00.02.14","hard_ver":"H09.06.01","extend_data":{},"warn_data":{"_rw":"","connect":true,"warn":false},"children":[{"id":4457570,"sn":"116481038963","vc":"","dtu_sn":"10FD81844117","type":3,"version":3,"replace_num":0,"model_no":"HMS-1800-4T","soft_ver":"V01.00.14","hard_ver":"H00.04.00","extend_data":{"grid_name":"","port_array":[1,2,3,4],"grid_id":0,"grid_version":""},"warn_data":{"warn":false,"connect":true},"children":[]}]}],"systemNotice":null}'
```

Wind06, 2023-10-27:
> I sent you a friend request on Discord.

Hmm are you sure I did not get any request .... check wind0613 instead of wind06

ebiancarelli, 2023-10-27:
> > I sent you a friend request on Discord.
> 
> Hmm are you sure I did not get any request .... check wind0613 instead of wind06

I sent the request to @Cosik .

Wind06, 2023-10-27:
Ah ok :-)

Cosik, 2023-10-28:
@ebiancarelli no sorry it not help, I didn't see meter on this request. About discord, I realized that it changed user identyfication so I will be Cosik#6329

Cosik, 2023-10-28:
@ebiancarelli and @Wind06 I'm looking in to code, and what is send and I realized that data could be added by manual configuration of mqtt entities.

Cosik, 2023-10-28:
@ebiancarelli and @Wind06 please check #119 and test it.

Wind06, 2023-10-28:
@Cosik Many thanks for your help and quick update. I will test it tomorrow and give an update

Wind06, 2023-10-29:
Hello, all data seems to be reported correctly. 
A big THANKS to @Cosik for his quick help and new version of the Addon.

ebiancarelli, 2023-10-29:
How do I test? Do I need to run the 'edge' version of the addon?

Cosik, 2023-10-29:
it is on edge version

ebiancarelli, 2023-10-29:
I'm not seeing anything different after waiting for the sensor data to be updates (I can see MQTT messages). I can't seem to get the `edge` version to show me DEBUG logs either so I can't troubleshoot

Wind06, 2023-10-29:
Maybe need to reload the Sensors view, or delete it and recreate it.
check also the version of the Addon it must be 

HoyMiles Solar Gateway Edge Addon 
Current version: 1.2.1

Cosik, 2023-10-29:
@ebiancarelli did you check configuration?

ebiancarelli, 2023-10-29:
My addon version is 1.1.1 (edge)...

Wind06, 2023-10-29:
you need to update it :-)

ebiancarelli, 2023-10-29:
Yes... How do I force that to happen?

@Cosik , I have 
```
DEVELOPERS_MODE: true
LOG_LEVEL: DEBUG
```

Wind06, 2023-10-29:
When selecting the Addon on top of your screen you should be prompted for new version update. You can try selecting the "Additional modules shop" and select top right "Check for updates"

ebiancarelli, 2023-10-29:
don't see it when I run `ha addons info 44349f66_hoymiles_solardata_edge`
```
version: 1.1.1
version_latest: 1.1.1
```

Cosik, 2023-10-29:
Because you still using stable version...

ebiancarelli, 2023-10-29:
Let me make sure I'm adding the addon properly. This is what I see in the AddOn Store after adding the repo.

[image]

Then I select the 'Edge' version.

Is that correct?

Wind06, 2023-10-29:
Yes that's correct and you should see version  1.2.1

ebiancarelli, 2023-10-29:
Here are infos from the command line. and a selected snip -
```
update_available: false
url: https://github.com/dmslabsbr/hoymiles/tree/master/edge
usb: false
version: 1.1.1
version_latest: 1.1.1
```
---------------------------------------

➜  ~ ha addons upgrade  44349f66_hoymiles_solardata_edge   
Processing... Done.

Error: No update available for add-on 44349f66_hoymiles_solardata_edge
➜  ~ ha addons info  44349f66_hoymiles_solardata_edge 
advanced: false
apparmor: default
arch:
- armhf
- armv7
- aarch64
- amd64
- i386
audio: false
audio_input: null
audio_output: null
auth_api: false
auto_update: false
available: true
boot: auto
build: true
changelog: true
description: Read Hoymiles Gateway Solar Data Edge Version.
detached: false
devices: []
devicetree: false
discovery: []
dns:
- 44349f66-hoymiles-solardata-edge.local.hass.io
docker_api: false
documentation: true
full_access: false
gpio: false
hassio_api: false
hassio_role: default
homeassistant: null
homeassistant_api: false
host_dbus: false
host_ipc: false
host_network: true
host_pid: false
host_uts: false
hostname: 44349f66-hoymiles-solardata-edge
icon: true
ingress: true
ingress_entry: /api/hassio_ingress/nllDTecfXQ7HB4WooneV0iirxl9vGC5f9cCUrI9SKEk
ingress_panel: false
ingress_port: 5000
ingress_url: /api/hassio_ingress/nllDTecfXQ7HB4WooneV0iirxl9vGC5f9cCUrI9SKEk/
ip_address: 172.30.32.1
kernel_modules: false
logo: true
long_description: "# HoyMiles Solar Data Gateway Add-on Edge version\nDevelopment
  versions can be unstable.\n\nApplication to read Hoymiles Gateway Solar Data using
  unofficial API\n\nI developed this program to integrate my solar system data to
  [Home Assistant](https://www.home-assistant.io/) Application through an add-on.\n\nNow,
  [Cosik](https://github.com/Cosik)  is helping too.\n\n[![Open your Home Assistant
  instance and show the add add-on repository dialog with a specific repository URL
  pre-filled.](https://my.home-assistant.io/badges/supervisor_add_addon_repository.svg)](https://my.home-assistant.io/redirect/supervisor_add_addon_repository/?repository_url=https%3A%2F%2Fgithub.com%2Fdmslabsbr%2Fhoymiles)\n\nBut
  you can use the application without using the Home Assistant. You just need a machine
[... 127 lines cut ...]
      LOG_LEVEL:
        description: Must be one of DEBUG, INFO, WARNING, ERROR, or CRITICAL
        name: LOG_LEVEL
      LOG_TO_FILE:
        description: Enable logging to file.
        name: LOG_TO_FILE
      MQTT_Host:
        description: Required if using an external MQTT Server.
        name: MQTT_Host
      MQTT_Pass:
        description: Required if using an external MQTT Server.
        name: MQTT_Pass
      MQTT_User:
        description: Required if using an external MQTT Server.
        name: MQTT_User
uart: false
udev: false
**update_available: false
url: https://github.com/dmslabsbr/hoymiles/tree/master/edge
usb: false
version: 1.1.1
version_latest: 1.1.1**
video: false
watchdog: false
webui: null

Cosik, 2023-10-29:
@ebiancarelli please check again. I've found error in configuration files.

ebiancarelli, 2023-10-29:
I removed the repo and added it back.

I see that the stable addon is v1.0.8. Edge is v1.1.1.

Does HA cache addon info that I need to purge?

[image]

Cosik, 2023-10-29:
@ebiancarelli sorry there were much more things to fix. I wonder why I didn't noticed that oldStable disappear. Now by removing and adding repo again, you should see 1.2.1 in edge and oldStable version.
[image]

ebiancarelli, 2023-10-29:
That did it for me. I was able to install v1.2.1 and see the new meter sensors.

Thanks!!!

mmz06, 2023-11-06:
Works perfectly from my side as well. 
Thank you @Cosik :-)

closed 2024-02-12 by Cosik