dmslabsbr/hoymiles #112: WebUI Error 502 and no MQTT Entries

matze19999, 2023-06-27:
Hi,

I just tested your stable and edge addon version but both show no errors in the logs, but also no WebUI and MQTT Entries are available.

Here is the log:

`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Starting add-on container...
Tue Jun 27 14:56:02 CEST 2023
total 12
drwxr-xr-x    2 root     root          4096 Jun 27 14:56 .
drwxr-xr-x    1 root     root          4096 Jun 27 14:56 ..
-rw-------    1 root     root           377 Jun 27 14:56 options.json
[14:56:02] INFO: /data/secrets.ini not exists!
Getting mqtt data...
[14:56:03] INFO: null
[14:56:03] INFO: {"supervisor":"2023.06.2","homeassistant":"2023.6.3","hassos":"10.3","docker":"23.0.6","hostname":"homeassistant","operating_system":"Home Assistant OS 10.3","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","haos","resolved","journal","disk","mount"],"machine":"qemux86-64","arch":"amd64","state":"running","supported_arch":["amd64","i386"],"supported":true,"channel":"stable","logging":"info","timezone":"Europe/Berlin"}
[14:56:03] INFO: {"supervisor":"2023.06.2","homeassistant":"2023.6.3","hassos":"10.3","docker":"23.0.6","hostname":"homeassistant","operating_system":"Home Assistant OS 10.3","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","haos","resolved","journal","disk","mount"],"machine":"qemux86-64","arch":"amd64","state":"running","supported_arch":["amd64","i386"],"supported":true,"channel":"stable","logging":"info","timezone":"Europe/Berlin"}
dmslabs - Home Assistant HoyMiles Solar Data Gateway Add-on
2023-06-27 14:56:03,726 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway  v.1.0.17
2023-06-27 14:56:03,727 - HoymilesAdd-on - INFO - Starting up... 2023-06-27 14:56:03
2023-06-27 14:56:03,727 - HoymilesAdd-on - INFO - Using External MQTT Server: True
2023-06-27 14:56:03,727 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2023-06-27 14:56:06,366 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - I got the token!!  :-)
2023-06-27 14:56:06,367 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_find
2023-06-27 14:56:06,736 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all
2023-06-27 14:56:07,866 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-06-27 14:56:09,051 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Starting MQTT 192.168.178.180
2023-06-27 14:56:09,148 - HoymilesAdd-on.mqttapi.Mqtt - INFO - MQTT connected with result code 0
2023-06-27 14:56:09,148 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Connected to 192.168.178.180
2023-06-27 14:56:10,059 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all
2023-06-27 14:56:10,467 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2023-06-27 14:56:12,680 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - last_data_time 2023-06-27T14:48:38+02:00
2023-06-27 14:56:12,680 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-06-27 14:56:13,842 - HoymilesAdd-on - INFO - Solar data publication...2023-06-27 14:56:13.842289
2023-06-27 14:56:13,842 - HoymilesAdd-on - INFO - DTU-WLite_1371790 data publication...2023-06-27 14:56:13.842598
2023-06-27 14:56:13,843 - HoymilesAdd-on - INFO - HM-800_6640490 data publication...2023-06-27 14:56:13.843177
2023-06-27 14:56:13,844 - HoymilesAdd-on - INFO - Main loop start!`

Cosik, 2023-06-27:
Did you try to use MQTT Explorer and wait at least 30 to receive message?

matze19999, 2023-06-28:
> Did you try to use MQTT Explorer and wait at least 30 to receive message?

Yes, but no messages

Cosik, 2023-06-29:
Web UI is not implemented. Please enable debug option and send whole logs. In logs there is no problem visible, so it means that problem is on your side. Script is able to connect to mqtt broker.

Cosik, 2023-07-09:
no response from @matze19999

closed 2023-07-09 by Cosik