dmslabsbr/hoymiles #113: Stops at https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all

ragman11, 2023-07-05:
Hi,

since today the Adon stopped working, also the edge Version.  What can be the cause?
Tried already several reboots of HA and the addon.

023-07-05 12:10:52,406 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway  v.1.0.7
2023-07-05 12:10:52,406 - HoymilesAdd-on - INFO - Starting up... 2023-07-05 12:10:52
2023-07-05 12:10:52,407 - HoymilesAdd-on - INFO - Using Internal MQTT Server: core-mosquitto
2023-07-05 12:10:52,407 - HoymilesAdd-on - INFO - Using Internal MQTT User: addons
2023-07-05 12:10:52,408 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2023-07-05 12:10:53,784 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - I got the token!!  :-)
2023-07-05 12:10:53,785 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_find
2023-07-05 12:10:54,266 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all

jeff1975, 2023-07-06:
I have the same problem will probably be after Hoymiles update (inverter limit).

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Starting add-on container...
Thu Jul  6 07:20:11 CEST 2023
total 24
drwxr-xr-x    5 root     root          4096 Jul  6 07:20 .
drwxr-xr-x    1 root     root          4096 Jul  6 07:20 ..
drwxr-xr-x    2 root     root          4096 Oct  9  2022 __pycache__
drwxr-xr-x    2 root     root          4096 Jul  6 06:43 jsons
-rw-------    1 root     root           326 Jul  6 07:20 options.json
drwxr-xr-x    2 root     root          4096 Oct  9  2022 templates
[07:20:11] INFO: /data/secrets.ini not exists!
Getting mqtt data...
[07:20:13] INFO: null
[07:20:14] INFO: {"supervisor":"2023.07.1","homeassistant":"2023.6.3","hassos":"10.3","docker":"23.0.6","hostname":"Home-Assistant","operating_system":"Home Assistant OS 10.3","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","haos","resolved","journal","disk","mount"],"machine":"qemux86-64","arch":"amd64","state":"running","supported_arch":["amd64","i386"],"supported":true,"channel":"stable","logging":"info","timezone":"Europe/Vienna"}
[07:20:14] INFO: {"supervisor":"2023.07.1","homeassistant":"2023.6.3","hassos":"10.3","docker":"23.0.6","hostname":"Home-Assistant","operating_system":"Home Assistant OS 10.3","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","haos","resolved","journal","disk","mount"],"machine":"qemux86-64","arch":"amd64","state":"running","supported_arch":["amd64","i386"],"supported":true,"channel":"stable","logging":"info","timezone":"Europe/Vienna"}
dmslabs - Home Assistant HoyMiles Solar Data Gateway Add-on
2023-07-06 07:20:15,621 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway  v.1.0.7
2023-07-06 07:20:15,626 - HoymilesAdd-on - INFO - Starting up... 2023-07-06 07:20:15
2023-07-06 07:20:15,628 - HoymilesAdd-on - INFO - Using Internal MQTT Server: core-mosquitto
2023-07-06 07:20:15,629 - HoymilesAdd-on - INFO - Using Internal MQTT User: addons
2023-07-06 07:20:15,631 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: 
2023-07-06 07:20:16,459 - INFO - I got the token!!  :-)
2023-07-06 07:20:16,461 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: 
2023-07-06 07:20:16,916 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: 
2023-07-06 07:28:17,494 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: 
2023-07-06 07:36:18,967 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_all

kucinskimarek, 2023-07-06:
I confirm, the same situation at my place. Maybe some changes on the side of HM servers?

NikolasRupp, 2023-07-06:
Have the same Issue since yesterday morning

reesechris, 2023-07-06:
Same problem here. Started this morning

gerwai, 2023-07-06:
yes, same issue

xlucasfelix, 2023-07-06:
The same here!

eduspina, 2023-07-06:
Same since yesterday evening.

listiges-kaenguru, 2023-07-06:
Same problem here since yesterday 9pm german time.

diogenes-ti, 2023-07-06:
Mesmo problema. Ultima leitura em 05/07/2023 08:00h
Alguem sabe como resolver?

ragman11, 2023-07-07:
@dmslabsbr Is there a way to fix this soon?

Cosik, 2023-07-07:
Please check #114

ragman11, 2023-07-08:
ist the new Version already pushed?

Cosik, 2023-07-09:
New version is released, please check it.

ragman11, 2023-07-09:
Hi, with the edge Version all hoymiles data are back, thank you

closed 2023-07-09 by ragman11

listiges-kaenguru, 2023-07-09:
Hi, it is working with the edge-version. When will you apply the changes to stable? I want Hoymiles in my persistant data. The entities from edge are separate.

Cosik, 2023-07-09:
Edge version will be soon stable version so, I'm not sure if we should Fix it.

kucinskimarek, 2023-07-09:
> Edge version will be soon stable version so, I'm not sure if we should Fix it.

I don't quite understand your answer. So should I dump the stable version and install edge to make the app work or should I still wait for the fix/update of the stable version?

jeff1975, 2023-07-09:
Thanks, everything works
HoyMiles Solar Gateway Addon
HoyMiles Solar Gateway Edge Addon
one more question, can I enter two or three plants in Hoymiles_Plant ID?

reesechris, 2023-07-09:
@Cosik do you already know when edge will become stable? I also used to use stable and don‘t want to mix up variables now. Imho thats a hotfix which could also get applied to stable? And thx for fixing! ❤️

Cosik, 2023-07-09:
I already implement fix on stable 1.0.8.I will try to do it today evening CET time.

Cosik, 2023-07-09:
> Thanks, everything works
> HoyMiles Solar Gateway Addon
> HoyMiles Solar Gateway Edge Addon
> one more question, can I enter two or three plants in Hoymiles_Plant ID?

On Edge it should work, somewhere was a feature request, please separate ids with only comma.

kucinskimarek, 2023-07-09:
> I already implement fix on stable 1.0.8.I will try to do it today evening CET time.

Stable works perfectly. Thanks for taking up the subject quickly.

jeff1975, 2023-07-09:
thanks works

rbreiff, 2023-07-09:
Thank you for the update, it works (Stable).

reesechris, 2023-07-09:
Thanks a lot for the quick fix!

listiges-kaenguru, 2023-07-09:
Thank you so much! 💪

xlucasfelix, 2023-07-09:
Thanks for the update! ✌

PR #114 (2023-07-09): Fixing Issue 113 for Stable Version by changing API endpoint
Closes #113 by changing the API Endpont to "station_select_device_of_tree" instead of "station_select_device_all"