dmslabsbr/hoymiles #117: HoyMiles Solar Gateway Addon The add-on seems to not be ready, it might still be starting.

biasowicka, 2023-07-11:
I got problem on my HA with HoyMiles Solar Gateway Addon

BAD GATEWAY:

HoyMiles Solar Gateway Addon
The add-on seems to not be ready, it might still be starting. Do you want to try again?

Same Problem in other POST do NOT resolve problem:

my log:

total 24
drwxr-xr-x    5 root     root          4096 Jul 11 17:04 .
drwxr-xr-x    1 root     root          4096 Jul 11 17:04 ..
drwxr-xr-x    2 root     root          4096 Jul 11 14:46 __pycache__
drwxr-xr-x    2 root     root          4096 Jul 11 16:48 jsons
-rw-------    1 root     root           484 Jul 11 17:04 options.json
drwxr-xr-x    2 root     root          4096 Jul 11 14:46 templates
[17:04:03] INFO: /data/secrets.ini not exists!
Getting mqtt data...
[17:04:07] INFO: null
[17:04:08] INFO: {"supervisor":"2023.07.1","homeassistant":"2023.7.1","hassos":"10.3","docker":"23.0.6","hostname":"homeassistant","operating_system":"Home Assistant OS 10.3","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","haos","resolved","journal","disk","mount"],"machine":"raspberrypi3-64","arch":"aarch64","state":"running","supported_arch":["aarch64","armv7","armhf"],"supported":true,"channel":"stable","logging":"info","timezone":"Europe/Warsaw"}
[17:04:08] INFO: {"supervisor":"2023.07.1","homeassistant":"2023.7.1","hassos":"10.3","docker":"23.0.6","hostname":"homeassistant","operating_system":"Home Assistant OS 10.3","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","haos","resolved","journal","disk","mount"],"machine":"raspberrypi3-64","arch":"aarch64","state":"running","supported_arch":["aarch64","armv7","armhf"],"supported":true,"channel":"stable","logging":"info","timezone":"Europe/Warsaw"}
dmslabs - Home Assistant HoyMiles Solar Data Gateway Add-on
2023-07-11 17:04:10,745 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway  v.1.0.8
2023-07-11 17:04:10,746 - HoymilesAdd-on - INFO - Starting up... 2023-07-11 17:04:10
2023-07-11 17:04:10,765 - HoymilesAdd-on - INFO - Using Internal MQTT Server: core-mosquitto
2023-07-11 17:04:10,765 - HoymilesAdd-on - INFO - Using Internal MQTT User: addons
2023-07-11 17:04:10,768 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2023-07-11 17:04:10,773 - HoymilesAdd-on.hoymilesapi.Hoymiles - DEBUG - {'Content-Type': 'application/json;charset=UTF-8', 'Cookie': "'uid=fff9c382-389f-4a47-8dc9-c5486fc3d9f5; EGG_SESS=XHfAhiHWwU__OUVeKh0IiITBnmwA-IIXEzTCHgHgww6ZYYddOPntPSwVz4Gx7ISbfU0WrvzOLungThcL-9D2KxavrtyPk8Mr2YXLFzJwvM0usPvhzYdt2Y2S9Akt5sjP'", 'Content-Length': '242'}
2023-07-11 17:04:12,673 - HoymilesAdd-on.hoymilesapi.Hoymiles - DEBUG - content: b'{"status":"0","message":"success","data":{"token":"1.TRhWzTvamvRzlnI86lEOipKIJ4J7XSoscME8cBcjqyxuYdPyr8eOnOrBIwv9y7tpMVUI3gNfl5NPqInbb9C7pd"},"systemNotice":null}'
2023-07-11 17:04:12,689 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - I got the token!!  :-)
2023-07-11 17:04:12,690 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_find
2023-07-11 17:04:12,694 - HoymilesAdd-on.hoymilesapi.Hoymiles - DEBUG - {'Content-Type': 'application/json;charset=UTF-8', 'Cache-Control': 'no-cache', 'Host': 'global.hoymiles.com', 'Connection': 'keep-alive', 'Accept': 'application/json, text/plain, */*', 'Accept-Encoding': 'gzip, deflate, br', 'User-Agent': 'Mozilla/5.0 (Macintosh; Intel Mac OS X 11_2_3) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/89.0.4389.90 Safari/537.36', 'Accept-Language': 'pt-BR,pt;q=0.9,it-IT;q=0.8,it;q=0.7,es-ES;q=0.6,es;q=0.5,en-US;q=0.4,en;q=0.3', 'Cookie': "'uid=fff9c382-389f-4a47-8dc9-c5486fc3d9f5; hm_token=1.TRhWzTvamvRzlnI86lEOipKIJ4J7XSoscME8cBcjqyxuYdPyr8eOnOrBIwv9y7tpMVUI3gNfl5NPqInbb9C7pd; Path=/; Domain=.global.hoymiles.com;Expires=Sat, 30 Mar 2024 22:11:48 GMT;'", 'Content-Length': '68'}
2023-07-11 17:04:14,342 - HoymilesAdd-on.hoymilesapi.Hoymiles - DEBUG - content: b'{"status":"0","message":"success","data":{"id":994990,"gid":122206,"name":"Jankowice Biasowicka 1","type":1,"tz_id":41,"city_code":"PL12059000000000","status":40,"create_by":211753,"create_at":"2023-02-09 19:41:40","classify":1,"tz_name":"UTC+01","pic_path":"","capacitor":"2.76","address":"8C Biasowicka,Jankowice,Wojew\xc3\xb3dztwo \xc5\x9al\xc4\x85skie,Powiat rybnicki 44-264,Poland","layout_step":2,"is_balance":0,"is_reflux":0,"remarks":"","config":{"sun_spec_num":0,"power_limit":"","power_limit_pf":"","power_limit_re":"","module_max_power":0,"owner_is_show_layout":1,"owner_is_modify_dev":0,"third_party_user":[],"billing_type":0,"billing_start":"","billing_every":0,"client_type":300,"layout_show":1},"is_stars":1,"money_unit":"PLN","electricity_price":0.95,"in_price":0,"usd":"","nk_name":null,"city_id":2003,"weather_of_cid":0,"timezone":{"id":41,"dis_name":"(UTC+01:00) Sarajevo,Skopje,Warsaw,Zagreb","name":"Europe/Sarajevo","tz_name":"UTC+01","offset":7200000},"local_time":"2023-07-11 17:04:14","parent_city":[{"id":155,"pid":0,"code":"PL00000000000000","weather_of_cid":0,"city_name":"Poland","country_code":"PL","level":1},{"id":820,"pid":155,"code":"PL12000000000000","weather_of_cid":0,"city_name":"\xc5\x9al\xc4\x85skie","country_code":"PL","level":2},{"id":2003,"pid":820,"code":"PL12059000000000","weather_of_cid":0,"city_name":"Jankowice","country_code":"PL","level":3}],"latitude":"50.03791673629545","longitude":"18.565014973282814","meter_location":0,"owner_list":[{"uid":229836,"name":"Jarek","user_name":"Jarek_Jankowice"}],"group":{"id":122206,"name":"Future Soleil Energy Poland","pid":47028,"type":4,"tid":160,"contact":"","phone":"","area":"","icon":""},"money_data":{"code":"PLN","unit":"z\xc5\x82"},"warn_data":{"s_uoff":false,"s_ustable":false,"s_uid":false,"l3_warn":false,"g_warn":false,"me_warn":false},"bms_capacitor":"0","inv_mode":0},"systemNotice":null}'
2023-07-11 17:04:14,358 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_of_tree
2023-07-11 17:04:14,362 - HoymilesAdd-on.hoymilesapi.Hoymiles - DEBUG - {'Content-Type': 'application/json;charset=UTF-8', 'Cache-Control': 'no-cache', 'Host': 'global.hoymiles.com', 'Connection': 'keep-alive', 'Accept': 'application/json, text/plain, */*', 'Accept-Encoding': 'gzip, deflate, br', 'User-Agent': 'Mozilla/5.0 (Macintosh; Intel Mac OS X 11_2_3) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/89.0.4389.90 Safari/537.36', 'Accept-Language': 'pt-BR,pt;q=0.9,it-IT;q=0.8,it;q=0.7,es-ES;q=0.6,es;q=0.5,en-US;q=0.4,en;q=0.3', 'Cookie': "'uid=fff9c382-389f-4a47-8dc9-c5486fc3d9f5; hm_token=1.TRhWzTvamvRzlnI86lEOipKIJ4J7XSoscME8cBcjqyxuYdPyr8eOnOrBIwv9y7tpMVUI3gNfl5NPqInbb9C7pd; Path=/; Domain=.global.hoymiles.com;Expires=Sat, 30 Mar 2024 22:11:48 GMT;'", 'Content-Length': '68'}
2023-07-11 17:04:15,210 - HoymilesAdd-on.hoymilesapi.Hoymiles - DEBUG - content: b'{"status":"0","message":"success","data":[{"id":930290,"sn":"4121A001DBA0","vc":"","dtu_sn":"4121A001DBA0","type":1,"version":3,"replace_num":0,"model_no":"DTU-Pro-S","soft_ver":"V00.02.15","hard_ver":"H09.06.01","extend_data":{},"warn_data":{"_rw":"","connect":true,"warn":false},"children":[{"id":4556150,"sn":"138281661331","vc":"","dtu_sn":"4121A001DBA0","type":3,"version":3,"replace_num":0,"model_no":"HMT-2250-6T","soft_ver":"V01.00.10","hard_ver":"H00.04.00","extend_data":{"grid_name":"","port_array":[1,2,3,4,5,6],"grid_id":0,"grid_version":""},"warn_data":{"warn":false,"connect":true},"children":[]}]}],"systemNotice":null}'
2023-07-11 17:04:15,227 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-07-11 17:04:15,231 - HoymilesAdd-on.hoymilesapi.Hoymiles - DEBUG - {'Content-Type': 'application/json;charset=UTF-8', 'Cache-Control': 'no-cache', 'Host': 'global.hoymiles.com', 'Connection': 'keep-alive', 'Accept': 'application/json, text/plain, */*', 'Accept-Encoding': 'gzip, deflate, br', 'User-Agent': 'Mozilla/5.0 (Macintosh; Intel Mac OS X 11_2_3) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/89.0.4389.90 Safari/537.36', 'Accept-Language': 'pt-BR,pt;q=0.9,it-IT;q=0.8,it;q=0.7,es-ES;q=0.6,es;q=0.5,en-US;q=0.4,en;q=0.3', 'Cookie': "'uid=fff9c382-389f-4a47-8dc9-c5486fc3d9f5; hm_token=1.TRhWzTvamvRzlnI86lEOipKIJ4J7XSoscME8cBcjqyxuYdPyr8eOnOrBIwv9y7tpMVUI3gNfl5NPqInbb9C7pd; Path=/; Domain=.global.hoymiles.com;Expires=Sat, 30 Mar 2024 22:11:48 GMT;'", 'Content-Leng [... cut ...]
2023-07-11 17:04:20,341 - HoymilesAdd-on - INFO - Solar data publication...2023-07-11 17:04:20.340929
2023-07-11 17:04:20,344 - HoymilesAdd-on - INFO - DTU-Pro-S_930290 data publication...2023-07-11 17:04:20.343959
2023-07-11 17:04:20,347 - HoymilesAdd-on - INFO - HMT-2250-6T_4556150 data publication...2023-07-11 17:04:20.347848
2023-07-11 17:04:20,355 - HoymilesAdd-on - INFO - Main loop start!

Version:

Home Assistant 2023.7.1
Supervisor 2023.07.1
Operating System 10.3
Interfejs użytkownika: 20230705.1 - latest

Thanks for help

biasowicka, 2023-07-11:
Yes, I know it works, but I can't find the sensors anywhere... I have various integrations of different systems and I have no problem with them.

Please advise.

Thank you.

SYSTEM LOG

23-07-11 17:21:02 INFO (SyncWorker_3) [supervisor.docker.interface] Stopping addon_44349f66_hoymiles_solardata application
23-07-11 17:21:11 INFO (SyncWorker_3) [supervisor.docker.interface] Cleaning addon_44349f66_hoymiles_solardata application
23-07-11 17:21:14 INFO (SyncWorker_4) [supervisor.docker.addon] Starting Docker add-on 44349f66/aarch64-addon-hoymiles_solardata with version 1.0.8
23-07-11 17:26:59 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
23-07-11 17:26:59 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.IPV4_CONNECTION_PROBLEM/ContextType.SYSTEM
23-07-11 17:26:59 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
23-07-11 17:27:00 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
23-07-11 17:27:00 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.NO_CURRENT_BACKUP/ContextType.SYSTEM
23-07-11 17:27:00 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_IPV6_ERROR/ContextType.DNS_SERVER
23-07-11 17:27:00 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR
23-07-11 17:27:00 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DOCKER_CONFIG/ContextType.SYSTEM
23-07-11 17:27:00 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.MULTIPLE_DATA_DISKS/ContextType.SYSTEM
23-07-11 17:27:00 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_FAILED/ContextType.DNS_SERVER
23-07-11 17:27:00 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
23-07-11 17:27:00 INFO (MainThread) [supervisor.resolution.check] System checks complete
23-07-11 17:27:00 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
23-07-11 17:27:02 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
23-07-11 17:27:02 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
23-07-11 17:27:02 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
23-07-11 17:28:26 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token

biasowicka, 2023-07-11:
nie wiem, czy to nie jest problemem ale na MQTT mam integracje  z SUPLA... na zewnętrznym brokerze

Cosik, 2023-07-11:
It will not interfer, but I can see that you configured different broker in addon so it could be an issue

biasowicka, 2023-07-11:
MQTT core-mosquitto ??
 :)

Cosik, 2023-07-11:
You should use the same brocker as you configured in home assistant.

biasowicka, 2023-07-11:
Yes it is Mosquitto broker

Cosik, 2023-07-12:
And you checked using mqtt explorer that there is no messages from addon? Look for sensors started from solarH_994990

biasowicka, 2023-07-12:
Could not fetch stats for 44349f66_hoymiles_solardata: Container addon_44349f66_hoymiles_solardata is not running

:/

Cosik, 2023-07-12:
seems that there is a bug in topic names probably related to last changes in API, but messages are working and could be added manually

Cosik, 2023-07-12:
@biasowicka please try use edge version, I don't know why stable don't detect new 1.1.0 version I will try to fix it today

biasowicka, 2023-07-12:
ok, I'm installing Edge versions, I'll let you know
 

[image]
[image]

biasowicka, 2023-07-12:
I had to disable integration on MQQT with an external server that supports SUPLA, unfortunately now I have a choice of either Hoymiles or SUPLA (duzo measuring devices) unfortunately it is impossible to integrate two at once. unless you can help? I am not that proficient in Home Assistance, but I advise you reasonably well.

[image]

Cosik, 2023-07-12:
So it mean that you have problems with your MQTT configuration. Are you aware that you could manually configure mqtt topics/sensors which cause that homeassistant will subscribe them?

biasowicka, 2023-07-12:
Thanks for the example..

I've already made complicated sensors by hand based on others, but here I would rather need some inspiration

:)

Thank you
:)

biasowicka, 2023-07-16:
Hi...

I just fight with configuration, sensor are set but no data

sensor:
  - name: "Hoymiles Real Power"
    unit_of_measurement: kWh
    device_class: energy
    force_update: 1
    state_topic: "homeassistant/sensor/dmslabs/solarH_994990_real_power"
    #payload_available: "online"
    #payload_not_available: "offline"
  - name: "Hoymiles Dzienne"
    unit_of_measurement: kWh
    device_class: energy
    force_update: 1
    state_topic: "homeassistant/sensor/dmslabs/solarH_994990_today_eq"
    payload_available: "online"
    payload_not_available: "offline"
  - name: "Hoymiles Total Power"
    unit_of_measurement: kWh
    device_class: "energy"
    state_class: "total_increasing"
    force_update: 1
    state_topic: "homeassistant/sensor/dmslabs/solarH_994990_total_eq"
    payload_available: "online"
    payload_not_available: "offline"

:)

Cosik, 2023-07-16:
Data is refresh with the same rate as it is no web page. Did you checked if home assistant is receiving any message on your topic?

biasowicka, 2023-07-16:
UndefinedError: 'value_json' is undefined

:)

biasowicka, 2023-07-16:
192.168.8.3
homeassistant
status = offline
sensor
dmslabs
solarH_994990_array_size
solarH_994990_array_size_kW
solarH_994990_co2_emission_reduction
solarH_994990_is_balance
solarH_994990_is_reflux
solarH_994990_last_data_time
solarH_994990_month_eq
solarH_994990_plant_tree
solarH_994990_real_power
solarH_994990_real_power_kw
solarH_994990_real_power_measurement
solarH_994990_real_power_total_increasing
solarH_994990_reflux_station_data
solarH_994990_today_eq
solarH_994990_today_eq_Wh
solarH_994990_year_eq
solarH_994990_total_eq
solarH_994990_power_ratio
solarH_4556150_alarm_code
solarH_4556150_alarm_string

biasowicka, 2023-07-16:
to ja juz sie poddaje, gdziekolwiek sa tutoriale jak dodac przez MQQT, ale jak wczesniej ja na MQQT z Home Assistant mam zintegogorwana i polaczone urzadzenia supla na wysokie brokerze https://cloud.supla.org/

Niestety nie można uruchomić konfiguracji, albo mam działać czujniki SUPLA albo podgląd do DTU, ale chce jedno i drugie.

Co zrobić z tymi czujnikami to ja już wymiękam niby sa ale nie odbierają danych, nie wiem czy to włąsnie nie jest problem z brakiem połączenia do MQQT. pozdrawiam

closed 2023-10-28 by Cosik