dmslabsbr/hoymiles #123: Improvement on installation description

TiPunch69, 2023-12-05:
I tried to install you add-on, but the readme seems to be a little bare bone.
I followed it to the end and checked the protocol tab - all seems to be working, or at least there are no errors. But how do I get my solar system in the devices? The frontend tab says "Bad Gateway" but I am not sure if this is ok.

Could you extend the installation and usage instructions.

Thanks and Best Regards

therealpanse, 2023-12-07:
got the same error.
I guess it's something requiring additional configuration of MQTT?

`[19:48:30] INFO: /data/secrets.ini not exists!
Getting mqtt data...
[19:48:30] ERROR: Got unexpected response from the API: Service not enabled
[19:48:31] INFO: null
[19:48:31] INFO: {"supervisor":"2023.11.6","homeassistant":"2023.12.0","hassos":"11.2","docker":"24.0.7","hostname":"homeassistant","operating_system":"Home Assistant OS 11.2","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","haos","resolved","journal","disk","mount"],"machine":"generic-x86-64","arch":"amd64","state":"running","supported_arch":["amd64","i386"],"supported":true,"channel":"stable","logging":"info","timezone":"Europe/Berlin"}
[19:48:31] INFO: {"supervisor":"2023.11.6","homeassistant":"2023.12.0","hassos":"11.2","docker":"24.0.7","hostname":"homeassistant","operating_system":"Home Assistant OS 11.2","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","haos","resolved","journal","disk","mount"],"machine":"generic-x86-64","arch":"amd64","state":"running","supported_arch":["amd64","i386"],"supported":true,"channel":"stable","logging":"info","timezone":"Europe/Berlin"}
dmslabs - Home Assistant HoyMiles Solar Data Gateway Add-on
2023-12-07 19:48:31,296 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway  v.1.0.8
2023-12-07 19:48:31,296 - HoymilesAdd-on - INFO - Starting up... 2023-12-07 19:48:31
2023-12-07 19:48:31,296 - HoymilesAdd-on - INFO - Using Internal MQTT Server: 
2023-12-07 19:48:31,296 - HoymilesAdd-on - INFO - Using Internal MQTT User: 
2023-12-07 19:48:31,296 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2023-12-07 19:48:32,695 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - I got the token!!  :-)
2023-12-07 19:48:32,695 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_find
2023-12-07 19:48:32,980 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_of_tree
2023-12-07 19:48:33,846 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-12-07 19:48:34,743 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Starting MQTT 
2023-12-07 19:48:34,743 - HoymilesAdd-on.mqttapi.Mqtt - ERROR - Invalid host.`

Cosik, 2023-12-07:
Hi, what do you expect from instruction?

Cosik, 2023-12-07:
@therealpanse only you got `invalid host` and sorry, without debug logs and more details about your configuration I'm not able to do anything. Also topic is "Improvement on installation description". Maybe at weekend I will have some time to do it, because I don't have installed this addon

therealpanse, 2023-12-07:
could to provide at least the basics for configurations for any users not familiar? You tell anyone to insert some console commands, but that doesn't include any instructions if anything else has to be configured. I think that is the underlying issue that matches the issue description

NeonGreenT, 2023-12-11:
Hey there, I've installed everything and have it running but **no new entities appear**.
I checked the (mosquitto) MQTT broker via MQTT-Explorer and found all the hoymiles topics.
In an desperate act I added them as MQTT sensors to my configuration.yaml which kind of also works.
I figure this might be more of a mosquitto issue rather than an hoymiles addon issue, but apart from a 'connection success' I can't pull much information from its log. Have I missed something?

```
**Addon-Log:**
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Starting add-on container...
Mon Dec 11 10:23:07 CET 2023
total 24
drwxr-xr-x    5 root     root          4096 Dec 11 10:23 .
drwxr-xr-x    1 root     root          4096 Dec 11 10:23 ..
drwxr-xr-x    2 root     root          4096 Dec  3 19:40 __pycache__
drwxr-xr-x    2 root     root          4096 Dec 11 00:26 jsons
-rw-------    1 root     root           387 Dec 11 10:23 options.json
drwxr-xr-x    2 root     root          4096 Dec  3 19:40 templates
[10:23:07] INFO: /data/secrets.ini not exists!
Getting mqtt data...
[10:23:08] INFO: null
[10:23:09] INFO: {"supervisor":"2023.11.6","homeassistant":"2023.12.1","hassos":"11.2","docker":"24.0.7","hostname":"homeassistant","operating_system":"Home Assistant OS 11.2","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","haos","resolved","journal","disk","mount"],"machine":"qemux86-64","arch":"amd64","state":"running","supported_arch":["amd64","i386"],"supported":true,"channel":"stable","logging":"info","timezone":"Europe/Berlin"}
[10:23:09] INFO: {"supervisor":"2023.11.6","homeassistant":"2023.12.1","hassos":"11.2","docker":"24.0.7","hostname":"homeassistant","operating_system":"Home Assistant OS 11.2","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","haos","resolved","journal","disk","mount"],"machine":"qemux86-64","arch":"amd64","state":"running","supported_arch":["amd64","i386"],"supported":true,"channel":"stable","logging":"info","timezone":"Europe/Berlin"}
dmslabs - Home Assistant HoyMiles Solar Data Gateway Add-on
2023-12-11 10:23:10,440 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway  v.1.1.0
2023-12-11 10:23:10,441 - HoymilesAdd-on - INFO - Starting up... 2023-12-11 10:23:10
2023-12-11 10:23:10,442 - HoymilesAdd-on - INFO - Using External MQTT Server: True
2023-12-11 10:23:10,443 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2023-12-11 10:23:12,123 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - I got the token!!  :-)
2023-12-11 10:23:12,123 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_find
2023-12-11 10:23:12,494 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_of_tree
2023-12-11 10:23:12,884 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-12-11 10:23:13,305 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Starting MQTT 192.168.178.86
2023-12-11 10:23:13,329 - HoymilesAdd-on.mqttapi.Mqtt - INFO - MQTT connected with result code 0
2023-12-11 10:23:13,330 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Connected to 192.168.178.86
2023-12-11 10:23:14,341 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_of_tree
2023-12-11 10:23:14,697 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2023-12-11 10:23:15,680 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - last_data_time 2023-12-11T10:14:59+01:00
2023-12-11 10:23:15,681 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-12-11 10:23:16,078 - HoymilesAdd-on - INFO - Solar data publication...2023-12-11 10:23:16.078775
2023-12-11 10:23:16,080 - HoymilesAdd-on - INFO - _2147260 data publication...2023-12-11 10:23:16.080476
2023-12-11 10:23:16,081 - HoymilesAdd-on - INFO - HMS-800W-2T_8777290 data publication...2023-12-11 10:23:16.081678
2023-12-11 10:23:16,088 - HoymilesAdd-on - INFO - Main loop start!
2023-12-11 10:31:16,088 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_of_tree
2023-12-11 10:31:16,761 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2023-12-11 10:31:17,063 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - last_data_time 2023-12-11T10:29:58+01:00
2023-12-11 10:31:17,063 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2023-12-11 10:31:18,014 - HoymilesAdd-on - INFO - Solar data publication...2023-12-11 10:31:18.014580
[... cut ...]
      value_template: "{{ value_json.plant_tree }}"

    - name: "Solar Data Time"
      state_topic: "home/solar/json_2359680"
      value_template: "{{ value_json.data_time }}"

    - name: "Solar Array Size kW"
      state_topic: "home/solar/json_2359680"
      unit_of_measurement: "kW"
      value_template: "{{ value_json.array_size_kW }}"

    - name: "Solar Today Eq Wh"
      state_topic: "home/solar/json_2359680"
      unit_of_measurement: "Wh"
      value_template: "{{ value_json.today_eq_Wh }}"
```

Cosik, 2023-12-12:
@NeonGreenT adding them as separate sensors in config is not required. You should see new mqtt devices in HA. Depending on your solar installation it could be few devices. One should be solar installation as main device and inverters with statuses as separate. What is more each entity of device should have correct parameters which allow you to use them. Please check again and let me know.

bridgewatermedia, 2023-12-23:
Cosik fixed the "Bad Gateway"  error.   Thanks Cosik.

closed 2024-02-12 by Cosik