dmslabsbr/hoymiles #124: Addon sometimes reporting negative values

therealpanse, 2023-12-09:
hi,

after fiddling around a bit and waiting for some data, I noticed, that the addon reports negative values to the energy dashboard from time to time.
[image]

if I view the sensor by itself, the values are given correctly. 
[image]

I noticed, that this happens whenever the curve drops below the local maximum. I can provide more data if needed.

Cosik, 2023-12-09:
Hi @therealpanse you just prove that add-on is working correctly and provide correct data. You have probably wrongly configured energy dashboard. If you need help raise question on any HA forum or provide here more details about configuration and details about used entities in dashboard.

therealpanse, 2023-12-09:
well, thanks for nothing then 🙄

Cosik, 2023-12-09:
ok, no problem,

Cosik, 2023-12-09:
I'm guessing that you put wrong entity but just guessing.