dmslabsbr/hoymiles #125: Entity Unavailable - 2 units

bridgewatermedia, 2023-12-22:
I believe this is related to issue #84.  I have two solar Hoymiles units.  I am using the latest Edge version (version: 1.2.1) and have them both working, but one or the other will go unavailable quite frequently.  I currently have an automation to restart the addon and this fixes the issue temporarily, but I'm restarting 2 or 3 times per hour.  There must be a setting somewhere to either reduce the polling frequency or set a temp variable to display last known value until the addon restarts or obtains the latest values.  Usually it is one unit that fails.  Everything displays fine in the Hoymiles online interface.  

If you need access to a dual station to test, let me know and I can give you access to use my system as a guinea pig. 

Thanks!

juslex, 2023-12-22:
Some problem here

Cosik, 2023-12-22:
1. I need logs
2. More details about your configuration of addon, do you run two instances off addon or how it is working.
3. It is christmas time so I don't know when I will have a time.
4. In the last option will be access for installation with this configuration but lets check other stuff first.

bridgewatermedia, 2023-12-23:
I use only one copy of the Edge addon.  (Latest version 1.2.2 )  It works great except for the timeout "unavailable" issue.  Here is a screenshot of a couple of cards I have on the front end:

[frontend.pdf](https://github.com/dmslabsbr/hoymiles/files/13758882/frontend.pdf)

Here are the logs:
[logfilehoyles.log](https://github.com/dmslabsbr/hoymiles/files/13758875/logfilehoyles.log)

Two instances yesterday where the addon stopped.  No errors that I could see in the logs.  

It is Christmas, so that is why I finally have time to write about this - no hurry on my end.  I shortened my automation to restart the addon after 30secs of unavailable so it seems to have mitigated this for now.   

Merry Christmas :)

Cosik, 2023-12-23:
@bridgewatermedia please enable debug and collect logs, also please add full logs from begining of starting addon, those logs looks too short. What is more it looks like connection to mqtt broker was broken so it cause problems. I need to check how client id is generate in implementation, meanwhile you could try also create separate user for this add on and check behaviour.

Also could you send info about your installation? some print screens of diagrams from hoymiles cloud,

Cosik, 2023-12-23:
@bridgewatermedia also multiple installtions are supported by default but it was not tested, did you try to provide installation ids separated by comma?

Cosik, 2023-12-23:
this should fix it, please check edge from my fork https://github.com/Cosik/hoymiles

bridgewatermedia, 2023-12-24:
I Enabled debug mode.
I turned off restart addon automation so the addon would crash/stop/become unavailable.
I reversed the order of Plantids listed in Hoymiles config page to see if it was always the first, 
   or the second plant that was always stopping (it was plantID 96262 stopping when 96262,1768330)
   
Results:

It seems like the first plant listed is the one that "hangs" and times out as "unavailable".    Now plant 1768330 becomes unavailable after switching the order to 1768330,96262

The first run through, all values seems to work, but then approximately 20min after the first run, the logs stop and entities become unavailable.   The logs below are complete up until they hang.  

Screenshot of Hoymiles also included. 

[LatestLogs.log](https://github.com/dmslabsbr/hoymiles/files/13760625/LatestLogs.log)
[screenshots.pdf](https://github.com/dmslabsbr/hoymiles/files/13760626/screenshots.pdf)
[secondlogstart.log](https://github.com/dmslabsbr/hoymiles/files/13760627/secondlogstart.log)

Let me know if you need anything else.

Cosik, 2023-12-26:
@bridgewatermedia did you tried version from my fork? I've provided there a small fix if not please provide me access to this installation

donwolle, 2024-01-08:
Hi cosik, Edge Version does not fix the problem. Would you please try to fix the Problem :-)?

[image]

Cosik, 2024-01-08:
@donwolle I need access to such installation, without it I will be not able to fix it. What is more, did you check edge version from my fork?

donwolle, 2024-01-08:
Yes, i tried the Edge Version.

First plant looses connection after a few minutes. After restarting the addon i can see the correct Data.

Can i Check other logs for you?

donwolle, 2024-01-12:
Fresh installation of home assistant - 2nd plant still does not work :-)

Cosik, 2024-01-12:
I already wrote that without access to installation with two plants there is no chance to fix it. Spamming here will not help.

donwolle, 2024-01-12:
How can i send you my login details to my Test Home Assistant system? 

Seconds plant works after addon restart. After a few minutes Sensors disappear.

Cosik, 2024-01-13:
@donwolle  I will need access to hoymilles cloud not home assistant. You could try reach me over discord - .cosik/Cosik#6329

Yannikk-k, 2024-02-15:
@Cosik: I have the same problem as described in this issue. I would be happy to give you access to my installation. I have just sent you a request in Discord.

Cosik, 2024-02-15:
@Yannikk-k please tests new version of edge. Bug should be fixed.

hangingloose, 2024-02-16:
@Cosik 
If I would like to monitor a second plant, I only have to insert the plant ID at HOYMILES_PLANT_ID... 
Like 96524,1508550 or how should I do this?

with two plant-ID´s I get an error... :
...
2024-02-16 08:37:44,638 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway  v.1.0.10
2024-02-16 08:37:44,638 - HoymilesAdd-on - INFO - Starting up... 2024-02-16 08:37:44
2024-02-16 08:37:44,639 - HoymilesAdd-on - INFO - Using Internal MQTT Server: core-mosquitto
2024-02-16 08:37:44,639 - HoymilesAdd-on - INFO - Using Internal MQTT User: addons
Traceback (most recent call last):
  File "/hoymiles.py", line 404, in <module>
    sys.exit(main())  # next section explains the use of sys.exit
  File "/hoymiles.py", line 329, in main
    if int(config["HOYMILES_PLANT_ID"]) < 100:
ValueError: invalid literal for int() with base 10: '96524,1508550'

Cosik, 2024-02-16:
@hangingloose did you read my comment before?

hangingloose, 2024-02-16:
Sorry, I thougt, I have had thist version, but it wasn´t.
Thanks! Now it looks fine ;-)

Cosik, 2024-02-16:
@hangingloose and others, edge is a testing version. When I will get confirmation that it is working over few days I will introduce fixes for stable if it is affected also.

hangingloose, 2024-02-19:
thanks, it is working since 3 days ;-)

Cosik, 2024-02-20:
ok, so I'm closing topic.