dmslabsbr/hoymiles #126: Ingress - WebUI - remove button

bridgewatermedia, 2023-12-22:
I'm not really a developer, but I noticed there are a number of requests or support emails regarding the "WebUI" (ingress) not working.  It has never worked for this, as this doesn't need a UI, but in the config.json it lists:

"services": [
    "mqtt:need"
  ],
  "ingress": true,
  "ingress_port": 5000,
  "host_network": true,
  "panel_icon": "mdi:solar-power",
  "panel_title": "HoyMiles Solar Data Gateway Edge version.",

ingress:true,

Wouldn't it be an easy fix to remove the "WebUI" button by merely changing that to 

ingress:false,

Does this break anything specifically in the rest of the addon?

Cosik, 2023-12-23:
should be done now

bridgewatermedia, 2023-12-23:
Works perfect for me. Thanks Cosik.

closed 2023-12-23 by Cosik