dmslabsbr/hoymiles #128: login details

boege, 2023-12-26:
Hello,
Where do I have to enter my login details for the MQTT Broker and the Hoymiles Cloud?

Cosik, 2023-12-26:
In a proper fields in configuration tab on addon

boege, 2023-12-26:
Sorry, I'm not very experienced in this field
Is there a file path you can show me?

Cosik, 2023-12-26:
so if you installed addon and you click on it in home assistant there should be configuration option. Sorry I don't have any working instances of this addon. Mainly I'm using it as standalone. After installation somewhere here should be configuration possibility
[image]

boege, 2023-12-26:
oh, okay
So you don't know how to add the credentials on a Raspberry Pi, or have I misunderstood?
because I have the problem that the mqtt broker doesn't work because the login data from the cloud is not on the addon

Cosik, 2023-12-26:
Yes, you misunderstand. In general it is hard to help when people not adding enough info in a question. One of correct fitting answer for your first question is "in a correct place". So till you will not providing more details no one will be able to help you in a way that you will be satisfied.

Cosik, 2024-02-12:
no response

closed 2024-02-12 by Cosik