dmslabsbr/hoymiles #129: KeyError: 'LOG_TO_FILE': Addon crashes on startup

reesechris, 2024-01-03:
I am using v1.0.9 and since a while it seems that the addon is not starting up properly anymore. See logs attached. Is this a bug or can I fix this via config somehow?

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Starting add-on container...
Wed Jan  3 08:58:24 CET 2024
total 24
drwxr-xr-x    5 root     root          4096 Jan  3 08:58 .
drwxr-xr-x    1 root     root          4096 Jan  3 08:58 ..
drwxr-xr-x    2 root     root          4096 Dec 23 19:03 __pycache__
drwxr-xr-x    2 root     root          4096 Jan  3 08:57 jsons
-rw-------    1 root     root           490 Jan  3 08:58 options.json
drwxr-xr-x    2 root     root          4096 Jul  6 11:38 templates
[08:58:24] INFO: /data/secrets.ini not exists!
Getting mqtt data...
[08:58:25] INFO: null
[08:58:25] INFO: {"supervisor":"2023.12.0","homeassistant":"2023.7.1","hassos":"10.3","docker":"23.0.6","hostname":"homeassistant","operating_system":"Home Assistant OS 10.3","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","haos","resolved","journal","disk","mount"],"machine":"generic-x86-64","arch":"amd64","state":"running","supported_arch":["amd64","i386"],"supported":true,"channel":"stable","logging":"info","timezone":"Europe/Berlin"}
[08:58:25] INFO: {"supervisor":"2023.12.0","homeassistant":"2023.7.1","hassos":"10.3","docker":"23.0.6","hostname":"homeassistant","operating_system":"Home Assistant OS 10.3","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","haos","resolved","journal","disk","mount"],"machine":"generic-x86-64","arch":"amd64","state":"running","supported_arch":["amd64","i386"],"supported":true,"channel":"stable","logging":"info","timezone":"Europe/Berlin"}
dmslabs - Home Assistant HoyMiles Solar Data Gateway Add-on
2024-01-03 08:58:25,303 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway  v.1.1.2
2024-01-03 08:58:25,303 - HoymilesAdd-on - INFO - Starting up... 2024-01-03 08:58:25
Traceback (most recent call last):
  File "/hoymiles.py", line 429, in <module>
    sys.exit(main())  # next section explains the use of sys.exit
  File "/hoymiles.py", line 323, in main
    if config["LOG_TO_FILE"]:
KeyError: 'LOG_TO_FILE'
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

Cosik, 2024-01-03:
@reesechris regarding to logs you are not using version 1.0.9 but 1.1.2 which mean that you have mess, I suggest fully reinstall addon with proper branch - oldstable, stable or edge

reesechris, 2024-01-03:
@Cosik wow thx for the super quick answer. That actually solved the problem, thx!

closed 2024-01-03 by reesechris