dmslabsbr/hoymiles #131: missing 1 required positional argument: / AttributeError: 'Client' object has no attribute '_sock'

rikthefrog, 2024-02-11:
I have the following configuration.

**HOYMILES_USER: XXX
HOYMILES_PASSWORD: XXXX
HOYMILES_PLANT_ID: "XXXX"
Read_meter_data: true
External_MQTT_Server: true
MQTT_Host: 192.168.X.X
MQTT_User: XX
MQTT_Pass: XXX
MQTT_TLS: false
MQTT_TLS_PORT: "8883"
DEVELOPERS_MODE: false
LOG_LEVEL: INFO
LOG_TO_FILE: false
FILE_PATH: hoymiles.log**

The add on startup and it gets data from the API, but it looks then there is an issue with the mqtt.

I can connect with the credentials via mqtt client. Any idea what can be wrong here?

2024-02-11 22:28:42,610 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway  v.1.2.2
2024-02-11 22:28:42,610 - HoymilesAdd-on - INFO - Starting up... 2024-02-11 22:28:42
2024-02-11 22:28:42,611 - HoymilesAdd-on - INFO - Using External MQTT Server: True
2024-02-11 22:28:42,611 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2024-02-11 22:28:43,491 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - I got the token!!  :-)
2024-02-11 22:28:43,491 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_find
2024-02-11 22:28:43,783 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_of_tree
2024-02-11 22:28:44,661 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2024-02-11 22:28:45,550 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2024-02-11 22:28:45,840 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
Exception ignored in: <function Client.__del__ at 0xffff80b93550>
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 874, in __del__
    self._reset_sockets()
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 1133, in _reset_sockets
    self._sock_close()
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 1119, in _sock_close
    if not self._sock:
AttributeError: 'Client' object has no attribute '_sock'
Traceback (most recent call last):
  File "/hoymiles.py", line 435, in <module>
    sys.exit(main())  # next section explains the use of sys.exit
  File "/hoymiles.py", line 381, in main
    mqtt_list[id].start()
  File "/mqttapi.py", line 64, in start
    self._client = mqtt.Client(client_id='', clean_session=True, userdata=None,
TypeError: __init__() missing 1 required positional argument: 'callback_api_version'
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

skuijper, 2024-02-12:
Paho MQTT released version 2.0.0 2 days ago with breaking changes:
https://github.com/eclipse/paho.mqtt.python/releases/tag/v2.0.0

Cosik, 2024-02-12:
#133 is ready for testing

tombenetin, 2024-02-13:
version: 1.2.3 works like a charm. Thanks so much 🙏

rikthefrog, 2024-02-20:
Thank you for fixing it, works again.

closed 2024-05-07 by Cosik