dmslabsbr/hoymiles #132: Addon errors

oxynaz, 2024-02-11:
hi, i tested the addon with the 3 versions of the program(edge, stable, old) and every time the same error, can you help ?

Starting add-on container...
Mon Feb 12 00:13:21 CET 2024
total 12
drwxr-xr-x    2 root     root          4096 Feb 12 00:13 .
drwxr-xr-x    1 root     root          4096 Feb 12 00:13 ..
-rw-------    1 root     root           600 Feb 12 00:13 options.json
[00:13:21] INFO: /data/secrets.ini not exists!
Getting mqtt data...
[00:13:21] INFO: null
[00:13:21] INFO: {"supervisor":"2024.01.1","homeassistant":"2023.10.3","hassos":null,"docker":"24.0.6","hostname":"genou","operating_system":"Debian GNU/Linux 12 (bookworm)","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","resolved","journal","disk","mount"],"machine":"qemux86-64","arch":"amd64","state":"running","supported_arch":["amd64","i386"],"supported":true,"channel":"stable","logging":"info","timezone":"Europe/Paris"}
[00:13:21] INFO: {"supervisor":"2024.01.1","homeassistant":"2023.10.3","hassos":null,"docker":"24.0.6","hostname":"genou","operating_system":"Debian GNU/Linux 12 (bookworm)","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","resolved","journal","disk","mount"],"machine":"qemux86-64","arch":"amd64","state":"running","supported_arch":["amd64","i386"],"supported":true,"channel":"stable","logging":"info","timezone":"Europe/Paris"}
dmslabs - Home Assistant HoyMiles Solar Data Gateway Add-on
2024-02-12 00:13:21,861 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway  v.1.2.2
2024-02-12 00:13:21,861 - HoymilesAdd-on - INFO - Starting up... 2024-02-12 00:13:21
2024-02-12 00:13:21,862 - HoymilesAdd-on - INFO - Using Internal MQTT Server: core-mosquitto
2024-02-12 00:13:21,862 - HoymilesAdd-on - INFO - Using Internal MQTT User: addons
2024-02-12 00:13:21,862 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2024-02-12 00:13:23,084 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - I got the token!!  :-)
2024-02-12 00:13:23,084 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_find
2024-02-12 00:13:24,079 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_of_tree
2024-02-12 00:13:24,513 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
Exception ignored in: <function Client.__del__ at 0x7f12fc9d74c0>
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 874, in __del__
    self._reset_sockets()
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 1133, in _reset_sockets
    self._sock_close()
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 1119, in _sock_close
    if not self._sock:
AttributeError: 'Client' object has no attribute '_sock'
Traceback (most recent call last):
  File "/hoymiles.py", line 435, in <module>
    sys.exit(main())  # next section explains the use of sys.exit
  File "/hoymiles.py", line 381, in main
    mqtt_list[id].start()
  File "/mqttapi.py", line 64, in start
    self._client = mqtt.Client(client_id='', clean_session=True, userdata=None,
TypeError: __init__() missing 1 required positional argument: 'callback_api_version'
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

m-orciani, 2024-02-12:
Me too. It appears that Miles Cloud has changed something regarding their API

Krzysztonek, 2024-02-12:
Same issue here.
```
...
Exception ignored in: <function Client.__del__ at 0x7f58255be4c0>
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 874, in __del__
    self._reset_sockets()
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 1133, in _reset_sockets
    self._sock_close()
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 1119, in _sock_close
    if not self._sock:
AttributeError: 'Client' object has no attribute '_sock'
Traceback (most recent call last):
  File "/hoymiles.py", line 366, in <module>
    sys.exit(main())  # next section explains the use of sys.exit
  File "/hoymiles.py", line 329, in main
    mqtt.start()
  File "/mqttapi.py", line 64, in start
    self._client = mqtt.Client(client_id='', clean_session=True, userdata=None,
TypeError: __init__() missing 1 required positional argument: 'callback_api_version'
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

tombenetin, 2024-02-12:
same here: 

2024-02-12 12:17:12,654 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway  v.1.1.2
2024-02-12 12:17:12,654 - HoymilesAdd-on - INFO - Starting up... 2024-02-12 12:17:12
2024-02-12 12:17:12,654 - HoymilesAdd-on - INFO - Using Internal MQTT Server: core-mosquitto
2024-02-12 12:17:12,654 - HoymilesAdd-on - INFO - Using Internal MQTT User: addons
2024-02-12 12:17:12,654 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2024-02-12 12:17:12,959 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - I got the token!!  :-)
2024-02-12 12:17:12,959 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_find
2024-02-12 12:17:14,094 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_of_tree
2024-02-12 12:17:15,117 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2024-02-12 12:17:15,464 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2024-02-12 12:17:16,458 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
Exception ignored in: <function Client.__del__ at 0x7f324920b430>
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 874, in __del__
    self._reset_sockets()
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 1133, in _reset_sockets
    self._sock_close()
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 1119, in _sock_close
    if not self._sock:
AttributeError: 'Client' object has no attribute '_sock'
Traceback (most recent call last):
  File "/hoymiles.py", line 429, in <module>
    sys.exit(main())  # next section explains the use of sys.exit
  File "/hoymiles.py", line 375, in main
    mqtt_list[id].start()
  File "/mqttapi.py", line 64, in start
    self._client = mqtt.Client(client_id='', clean_session=True, userdata=None,
TypeError: __init__() missing 1 required positional argument: 'callback_api_version'
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

closed 2024-02-12 by Cosik